coolsnowwolf/lede #2621: 木马Patch好像有问题

stock169, 2019-12-29:
. /home/pchome/lede/include/shell.sh; gzip -dc /home/pchome/lede/dl/trojan-1.14.0-18763b7529963efef8dcb01f50389a6f49213ae1.tar.gz | tar -C /home/pchome/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/trojan-1.14.0/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /home/pchome/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/trojan-1.14.0

Applying ./patches/001-force-openssl-version.patch using plaintext: 
patching file CMakeLists.txt
Hunk #1 FAILED at 43.
1 out of 1 hunk FAILED -- saving rejects to file CMakeLists.txt.rej
Patch failed!  Please fix ./patches/001-force-openssl-version.patch!
Makefile:66: recipe for target '/home/pchome/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/trojan-1.14.0/.prepared_8ecfaea1d57cf94f2009d16b20960d53_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/pchome/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/trojan-1.14.0/.prepared_8ecfaea1d57cf94f2009d16b20960d53_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/pchome/lede/package/lean/trojan'
time: package/lean/trojan/compile#0.14#0.08#0.20
package/Makefile:107: recipe for target 'package/lean/trojan/compile' failed
make[2]: *** [package/lean/trojan/compile] Error 2
make[2]: Leaving directory '/home/pchome/lede'

EstFoxs, 2019-12-29:
确实，编译出错

stock169, 2019-12-29:
自己改成 1.1.0 应该可以过

lqhorochi, 2019-12-29:
确实有问题,大雕快修复好

stock169, 2019-12-29:
改 001-force-openssl-version.patch 文件成下面样子就好了

--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -43,7 +43,7 @@ if(MSVC)
     add_definitions(-DBOOST_DATE_TIME_NO_LIB)
 endif()
 
-find_package(OpenSSL 1.1.0 REQUIRED)
+find_package(OpenSSL 1.1.1 REQUIRED)
 include_directories(${OPENSSL_INCLUDE_DIR})
 target_link_libraries(trojan ${OPENSSL_LIBRARIES})
 if(OPENSSL_VERSION VERSION_GREATER_EQUAL 1.1.1)

zcluo, 2019-12-29:
同样问题

获取 Outlook for iOS<https://aka.ms/o0ukef>

ihuaer, 2019-12-29:
> 改 001-force-openssl-version.patch 文件成下面样子就好了
>
> --- a/CMakeLists.txt
> +++ b/CMakeLists.txt
> @@ -43,7 +43,7 @@ if(MSVC)
> add_definitions(-DBOOST_DATE_TIME_NO_LIB)
> endif()
>
> -find_package(OpenSSL 1.1.0 [...]

感谢，马上编译试试效果

closed 2019-12-29 by coolsnowwolf