coolsnowwolf/lede #2632: rpi-4b编译64位固件开不了机

EnnawYang, 2019-12-29:
RT，使用最新源码编译刷入无反应，连接hdmi显示彩色开机画面，无法进入系统，刷回前几天编译的固件能正常开机

SharerMax, 2019-12-29:
同样的问题

zhaianan, 2019-12-30:
今天最新的源码同样啊

closed 2019-12-30 by coolsnowwolf

SharerMax, 2019-12-31:
猜测是内核更新的问题，手都删除`target/linux/brcm2708/patches-4.19`下的patch文件替换为[openwrt项目的patch文件](https://github.com/openwrt/openwrt/tree/master/target/linux/brcm2708/patches-4.19)，可编译，但是也没跑起来。。

最终稿revert到 [87f94af11323cd9d5333b5265214e617d784c231](https://github.com/coolsnowwolf/lede/tree/87f94af11323cd9d5333b5265214e617d784c231) 编译后成功运行

注意revert的时候删除`.config` 文件，重新使用 `make menuconfig` 生成，[最新的代码](https://github.com/coolsnowwolf/lede/tree/64fb649bc80be2564364dba1c98bab82ce8bf0d6)生成的config文件好像没用无线网络

SharerMax, 2020-01-01:
找到问题了，系统没有按64位运行，涉及到的修改在 [这里](https://github.com/coolsnowwolf/lede/commit/a5f1ebf45ba0a1eb907b8b11614f02d27706efcb#diff-dd50d5733cf6569124def77a40fd996aL11)
可以手动修改boot分区里面的config文件，添加一行`arm_64bit=1`，强制系统64位启动，具体顺序可参考删除的代码
但是好像有个问题，我是 删除 `.config` 文件后再执行 `make clean`，然后重新编译，生成的镜像没有无线功能，有点尴尬
@zhaianan @EnnawYang

SharerMax, 2020-01-01:
没有无线的问题可能是无线驱动更换的问题，涉及到的修改在[这里](https://github.com/coolsnowwolf/lede/commit/a5f1ebf45ba0a1eb907b8b11614f02d27706efcb#diff-ead510ffb2ca139d1f1897da68e8a6acL151)，驱动由 `brcmfmac-firmware-43455-sdio brcmfmac-firmware-43455-clm_blob brcmfmac-firmware-43455-sdio-rpi-4b` 更换为 `cypress-firmware-43455-sdio cypress-nvram-43455-sdio-rpi-4b`, 但是我没找到更换后驱动的相关文件，可以尝试在 `make menuconfig` 中的 `Firmare`子项中手动选择`brcmfmac-firmware-43455-sdio brcmfmac-firmware-43455-clm_blob brcmfmac-firmware-43455-sdio-rpi-4b`

EnnawYang, 2020-01-02:
> 没有无线的问题可能是无线驱动更换的问题，涉及到的修改在[这里](https://github.com/coolsnowwolf/lede/commit/a5f1ebf45ba0a1eb907b8b11614f02d27706efcb#diff-ead510ffb2ca139d1f1897da68e8a6acL151)，驱动由 `brcmfmac-firmware-43455-sdio [...]

pi4b的无线对我来说可有可无，1连接速率慢，2经常会崩溃，官方的openwrt也是如此，我用来也只是做个单臂旁路由，或者PT下载机，前段时间更换了x86设备所以用不到了，树莓派只是用来折腾系统的，因为开不了机所以提供不了日志，我有空再试下吧

SharerMax, 2020-01-02:
> pi4b的无线对我来说可有可无，1连接速率慢，2经常会崩溃，官方的openwrt也是如此，我用来也只是做个单臂旁路由，或者PT下载机，前段时间更换了x86设备所以用不到了，树莓派只是用来折腾系统的，因为开不了机所以提供不了日志，我有空再试下吧

嗯，我已经测试过了，做好镜像后，在boot分区中的`config`文件中添加一行 `arm_64bit=1`, 成功开机，我已经用上了

FattyboyN, 2020-01-16:
请问无线的问题修复了吗？昨天编译完遇到同样的问题，接口菜单里没有无线，make menuconfig已经手动选了brcmfmac-firmware-43455所有相关的

SharerMax, 2020-01-16:
> 请问无线的问题修复了吗？昨天编译完遇到同样的问题，接口菜单里没有无线，make menuconfig已经手动选了brcmfmac-firmware-43455所有相关的

已经修复 相关修复为 #2745 
你可以尝试拉取最新代码后，删除 ```.config```文件，重新用 ```make menuconfig```来勾选配置
不需要再自己勾选相关驱动，新驱动已经默认选中了