coolsnowwolf/lede #2647: SSR+ Trojan无法运行，日志提示如下，请问如何解决？

EthanChoy, 2019-12-30:
Error loading shared library libboost_system.so.1.71.0: No such file or directory (needed by /usr/sbin/trojan) Started!

closed 2019-12-30 by coolsnowwolf

txdywy, 2020-01-13:
请问如何解决的？

我看我openwrt上/usr/lib里是libboost_system.so.1.72.0

EthanChoy, 2020-01-14:
@txdywy 你也是单独编译的apk吧。我最后编译了整个固件。就没有这个问题了。我没有解决。

txdywy, 2020-01-14:
是的，刚刚去找到了
~/lede/bin/packages/x86_64/packages/

下有boost_1.71.0-4_x86_64.ipk

晚点安装一下试试能不能解决

txdywy, 2020-01-15:
安装提示不能downgrade 到1.71.0

看起来是我的openwrt里libboost是1.72.0
这个lede目前是1.71.0,所以编译出来的trojan也是这个要求

看了下trojan自己up到1.72.0,所以可能下载原版的openwrt编译一下或许能满足.尝试中

txdywy, 2020-01-15:
> @txdywy 你也是单独编译的apk吧。我最后编译了整个固件。就没有这个问题了。我没有解决。

解决了。单独编译的环境里面依赖的boost版本低于实际安装系统。用openwrt原版最新代码（boost是1.72.0）编译trojan就行了

chalau, 2020-04-25:
> 用openwrt原版最新代码（boost是1.72.0）编译trojan就行了

具体怎么操作？

txdywy, 2020-04-25:
下载openwrt原版代码在里面编译trojan

chalau, 2020-04-25:
> […](#)

谢谢啊，这么快回复：就是把ubuntu里现在的boost_1.71.0-4_x86_64.ipk替换成原版里下载的boost_1.72.0-4_x86_64.ipk？

txdywy, 2020-04-25:
可能不行，我后来是下载了一套openwrt最新代码，里面环境是boost1.72.0，在里面编译trojan的ipk来安装

chalau, 2020-04-25:
> […](#)
骨灰级菜鸟终于看明白了：）

chalau, 2020-04-27:
> 可能不行，我后来是下载了一套openwrt最新代码，里面环境是boost1.72.0，在里面编译trojan的ipk来安装
> 
> 
> > 下载openwrt原版代码在里面编译trojan chalau notifications@github.com 于 2020年4月25日周六
> > … <#m_8144071900189919133_>
> > 用openwrt原版最新代码（boost是1.72.0）编译trojan就行了 具体怎么操作？ — You are receiving this
> > <#2647 (comment)
> > <https://github.com/coolsnowwolf/lede/issues/2647#issuecomment-619312302>>,
> >
> >
> > 谢谢啊，这么快回复：就是把ubuntu里现在的boost_1.71.0-4_x86_64.ipk替换成原版里下载的boost_1.72.0-4_x86_64.ipk？

感谢、搞定：1.72