coolsnowwolf/lede #2648: 终归到底，这个又是一个失败的项目而已

coolsnowwolf, 2019-12-30:
我一直想象的那种人人为我，我为人人，大家一起来维护一个项目的景象始终没有出现，也没有出现的迹象。

维护这个项目的过程中，主动分享的同学并不多。很多来汇报问题的人是以一种小白求大大免费解决问题，解决完就走人的方式来的。

然而既不愿提供足够的信息，截图日志什么都没有，也不愿写一些自己尝试的过程供后人参考。

issues 那边，一堆随随便便的一句话问题，却想着别人认认真真的去帮他分析解决的，比比皆是。

退一步来说，就算有些 issue 是很简单的问题，5000 多个 fork ，却也没有多少人愿意回复下，就好像这个完全就是作者一个人服务所有人的项目一样。

更别提pulls 区那边的一堆垃圾 yml 了，关都关不过来。

互帮互助的气氛就是搞不起来。对比下国外的社区差好远。

国内的开源环境到底是什么样呢？他们使用现成的开源的成果，却很少有人遵守开源协议。

更有甚者，毫无贡献、拿来就用，然后宣传开源后就属于全人类的了，然后说是自己的成果（即使不是公开说出来，也是暗示性的影响。为数不多的人遵守 GPL ，更别提哪怕在 Credit 注明一下来源的人），甚至反骂开源的作者。

这个项目已经让我觉得越来越无趣，罢了吧。

zb479519891, 2019-12-30:
这个项目是我自己愿意动手尝试的第一个项目，一直以来看到编译就头大，后来看了油管上的分享特意过来看了下，最后决定亲自动手，自己的两个路由都是用这个项目做的

dmfootman, 2019-12-30:
我第一次编译的体验就是这个带来的, 过程没有很顺利, 但也算个有意思的起始. 我对这次体验很满意, 对现在的路由也很满意. 感谢你的付出, 祝未来顺利

lqhorochi, 2019-12-30:
支持大雕,大雕加油
平时都有反馈bug的

dunkeng, 2019-12-30:
这个项目其实很伟大的，我第一次编译就是在这里，无奈水平太差无法参与代码上的贡献。
项目目前什么都不懂的小白太多，说明项目确实是很好的。
问题只是没成长的小白太多而且，这不是一个项目的问题，是整体问题。
国内各个论坛，无论是包括OPENWRT，各手机固件，基本上的套路都是跪求，催更，拿到了新版本就骂各种问题，提问水平也全是小白。
我之前闲鱼卖了一个小主机，卖家就是个小白，完全不懂LINUX，进个BIOS还要我一项项给他翻译说明，问了的全是软件问题，要不是没确认收货，完全不想鸟他，最后还给我差评，说我东西有问题，什么ubuntu下shutdown -h关机不断电，我还得教他命令，我让他退货还不退，他妈的没办法。目前国内总体素质确实低下，品行不端的人太多了。

lysssssss, 2019-12-30:
这是我第一个在开源社区里提交pr的项目，虽然没解决什么大问题，但在其中也学到了不少，感谢大雕的付出。同时也希望国内的开源风气慢慢变好。也鄙视那些用了别人代码还理直气壮的人。
也没什么好说的，那就祝大雕新年大吉把

elmagnificogi, 2019-12-30:
我作为一个吃瓜观众，客观的来说，其实造成这种情况的，一部分是大雕你自己搞得。

首先本身这个lede也好，openwrt也好，软/硬路由也好，本身就是一个专业的东西。
但是大雕你降低了这个专业准入的门槛，那么好了，各种小白，各种水货都加入了。
你看看你的群，除了伸手党，一键党，还有啥，问问题没人解，发红包抢的比谁都快。
色图一个发的比一个快，车倒是开起来了，但是项目呢？

讲道理有多少专业的，愿意继续这样下去，反观openwrt或者lede他们本身的社区，
哪里有那么多小白，能用或者会用的人，多少都是专业的，而不是这种啥都不会就会个一键编译的废物。

其次对于这项目而言，相关的文档也好，教程也好，你大屌没时间搞，小白也没时间啊，
他们刷完了就走了，谁还没事写个教程做个分享？伸手的太多了。

想要继续维护下去我的建议就是，小白搞走，专业的留下，不能提供任何帮助的小白请滚开。
至少要有一个规划或者什么其他的，组织一下，让小白有点用吧，填填文档写写教程百科总是可以的吧

要培养起来一个专业的氛围确实很难，定期分享，互相交流学习才能走得更远吧
更多的小白除了能被割韭菜还能干啥？

fzner, 2019-12-30:
作为一个业外人士，除了默默推广，帮更小白的人解答傻瓜问题以外，真的没法提供技术上的贡献，唯有感谢。

asushugo, 2019-12-30:
作为一名后端程序员，在大雕从分享刷机包，到开源，一直使用到现在，这其中带给我了许多的提升，让我更熟悉地了解Linux，每一个commit都是可以学习的。我也尽我所能，如果有问题我可以解决，我也会去pull requests，确实我觉得这才是开源精神，支持“人人为我，我为人人”的态度。不止在github，在Q群也可以看到很多的解答，很感谢大雕的这种奉献精神，让我们学到了这么多，也希望如此好的项目能持续下去！如果没有了这个项目，我已经不知道可以去哪里找到这么好用的固件了。

dolphinpaopao, 2019-12-30:
前两月才接触了解到L大的这个开源项目，也是首个按照教程自我编译成功的固件。感谢L大的辛勤付出。目前工作中接触到的基层人士绝大多数同样缺乏自我学习精神，习惯衣来张手饭来张口，不理解不在乎学习是需要成本的。对于轻易获得的东西不珍惜。其实L大可以尝试稍微提升一下使用门槛，比如固件使用全英文设置，一点点小小的转变却能过滤掉一大波不值得理会的连搜索也不愿意尝试的伸手党。再次感谢L大的付出。

esirplayground, 2019-12-30:
环境恶劣，小人猖狂，无论你做出什么决定，我都无条件支持你！

dolphinpaopao, 2019-12-30:
> 环境恶劣，小人猖狂，无论你做出什么决定，我都无条件支持你！

eSir您也来了，目前国内现状就是这么无奈，心怀感激的还是少数，无怪乎好东西都是小圈子流传。少数受到多数的拖累。

GDZmax, 2019-12-30:
非常感谢大佬的辛苦，只是限制与个人水平，不能做出贡献。

alxe007, 2019-12-30:
非常理解L大的既无奈又无力解决的心情，马上2020年了祝L大事事顺心

klever1988, 2019-12-30:
有一个很现实的原因，真正有能力参与到开源项目的中国普通程序员，其实他们工作要比国外繁重得多。国外的开源社区能繁荣，相当一部分原因是他们有闲暇时间关注自己的兴趣爱好。开源社区的繁荣依赖的是大量中端程序员的热情参与，单靠金字塔顶的几位大神也是带不火一个项目的。但是代表最高生产力的中端程序员们，都在为房奋斗的996工作制里失去了对代码最初的兴趣。我认识几位程序员朋友，这是我自己的感想。

Ryuchen, 2019-12-30:
L大，我是一周前开始接触这个项目的，本人特别想参与到这个项目中来，但是一直觉得自己水平不够，想看懂部分代码之后再参与进来，不过看了L大这么苦恼，感觉不一定需要完全看懂代码，琐碎的小事也可以帮忙，接下来参与到这个项目中来，帮助L大解决一些问题。

lqhorochi, 2019-12-31:
固件一切是迈克怼大雕后，大雕干脆不干了。用大雕的源码做收费固件，还怼大雕！真是啥人都有。
无论如何，最多贡献还是大雕的源码

killDevils, 2019-12-31:
Dear Coolsnowwolf,

I really wanna show all my respect to you.

I was the guy you mentioned as “小白” 6 months ago. If it weren’t for your great LEDE project, I would be “小白” forever!

So, thank you so much!

借助您的成果与智慧，我给公司总交换机架上梯子，每个人都看到了一个全新的互联网世界。连前台小美女都改刷油管的吃播了。

您付出太多了，您歇歇。

我希望
现在能给这个项目赞助Money，
未来能给这个项目贡献技术。

vlanbin, 2019-12-31:
环境如此，尊重你的决定。
从这个项目入门，受益良多。感谢！
技术不止步。

hcym, 2019-12-31:
感谢大雕，无论如何选择都可以

AlanLang, 2019-12-31:
感谢您的付出，谢谢

azgh, 2019-12-31:
每次看到好项目走到这种境地，心里都不舒服，感谢 @coolsnowwolf 。
几个月前，一直学习并尝试编译固件，一直失败，对小白来说，这个确实有一定难度，提过问题，也自己思考问题，并尝试解决。上周，我人生第一次编译固件成功，很兴奋，确实学习到了很多东西。
对于小白来说，有时编译成功了，都不知道自己是运气还是确实了解原理了，所以除了脸皮厚敢问问题，真的不敢帮人解决问题，很惭愧，自己技术不精，没能起到帮助作用。
很多生活在墙内的人，基本都在GITHUB里寻找希望，是LEAN、SS和V2RAY这样的项目带来了曙光，感谢为此付出的人，当然也尊重你的决定，只是希望还会有人愿意为墙里哪些渴望自由的人们提供帮助，如果有什么我们这些小白能做的，我愿意尝试。

xinpenghan, 2019-12-31:
我本身是做运营的，曾经负责管理日活跃2000w的网站，具体网站保密。对于这个问题我有几个看法，希望可以帮到你！也同时谢谢你的项目。给我帮助很多！

一、先抛开迈克那sb不说。这种开源的项目在国内是很容易被复制的，其它的也都是你抄我，我抄他。国内市场就这样，咱们也没办法。就算注册了版权也没用，改了名字，随便加点自己的东西就成自己的了。既然避免不了，咱们只能心态好一些了。

二、实用性建议：
**虽然开源了，但是可以参考运营法则，应该适当做一些用户留存和用户的引导。**

1）先说用户留存，要使你的固件更有效传播和分享。
     可以在固件加入你固件的名字。有代表性。并在固件首页加入api接口，断网显示的内容和拨号联网后显示api接口内容

2）api接口内容：
     可以体现你的个人想法还有仓库链接，还要加入issues的入口链接。因为issues的链接很多人找不到入口，我第一次使用github的时候，我反正不知道有issues可以使用。甚至很多人没使用github的人就更不知道了，小白真的很傻

3）api加入actions链接，因为这样可以简单的发布你的固件，实现一个固件生态循环。

4）虽然这样固件不纯净了，但是可以引入更多的人，然后从中简单的引导，使大家知道这个仓库的地址还有作者的辛苦劳动。传播速度会快很多。

5）如果安装上面的做了后，仓库地址流量肯定会变多，但是人就是这样，不管肯定不行。

6）人多了，懂技术的人也会慢慢变多，可以简单的制度化。可以简单的做一些人流分类

7）可以定期发布一个简单的招聘，比如招一些日常维护社区的人和招懂技术的，一同维护项目，当然没有工资哈。社区人流量多了，自然会有人愿意干，你要做的就是画出路线，让他们走就行了。不需要你花费多长时间。

8）给这些人分开建立qq群，你可以制定简单的规矩，发布到群里，无规矩不成方圆。国家有法律，公司有公司制度。用制度管理人，是最简单的方法。

9）定期发布招聘，时间可以自己定。可以更新不活跃的人，然后剔除他们，实现人员更新。

三、现在谈谈迈克，这个sb是真的渣，偷别人的东西声称是自己的。如果有上面这些东西，大家就会更加懂的，谁的代码才是正版，慢慢这些渣人也会少很多。

                                                                                            马上要跨年了，在这里祝大家新年快来！
                                                                                          **上面是我的个人看法，其它人不喜勿喷！**
                                                                                                                                         2019-12-31

xinpenghan, 2019-12-31:
我本身是做运营的，曾经负责管理日活跃2000w的网站，具体网站保密。对于这个问题我有几个看法，希望可以帮到你！也同时谢谢你的项目。给我帮助很多！

一、先抛开迈克那sb不说。这种开源的项目在国内是很容易被复制的，其它的也都是你抄我，我抄他。国内市场就这样，咱们也没办法。就算注册了版权也没用，改了名字，随便加点自己的东西就成自己的了。既然避免不了，咱们只能心态好一些了。

二、实用性建议：
**虽然开源了，但是可以参考运营法则，应该适当做一些用户留存和用户的引导。**

1）先说用户留存，要使你的固件更有效传播和分享。
     可以在固件加入你固件的名字。有代表性。并在固件首页加入api接口，断网显示的内容和拨号联网后显示api接口内容

2）api接口内容：
     可以体现你的个人想法还有仓库链接，还要加入issues的入口链接。因为issues的链接很多人找不到入口，我第一次使用github的时候，我反正不知道有issues可以使用。甚至很多人没使用github的人就更不知道了，小白真的很傻

3）api加入actions链接，因为这样可以简单的发布你的固件，实现一个固件生态循环。

4）虽然这样固件不纯净了，但是可以引入更多的人，然后从中简单的引导，使大家知道这个仓库的地址还有作者的辛苦劳动。传播速度会快很多。

5）如果安装上面的做了后，仓库地址流量肯定会变多，但是人就是这样，不管肯定不行。

6）人多了，懂技术的人也会慢慢变多，可以简单的制度化。可以简单的做一些人流分类

7）可以定期发布一个简单的招聘，比如招一些日常维护社区的人和招懂技术的，一同维护项目，当然没有工资哈。社区人流量多了，自然会有人愿意干，你要做的就是画出路线，让他们走就行了。不需要你花费多长时间。

8）给这些人分开建立qq群，你可以制定简单的规矩，发布到群里，无规矩不成方圆。国家有法律，公司有公司制度。用制度管理人，是最简单的方法。

9）定期发布招聘，时间可以自己定。可以更新不活跃的人，然后剔除他们，实现人员更新。

三、现在谈谈迈克，这个sb是真的渣，偷别人的东西声称是自己的。如果有上面这些东西，大家就会更加懂的，谁的代码才是正版，慢慢这些渣人也会少很多。

                                                                                            马上要跨年了，在这里祝大家新年快来！
                                                                                          上面是我的个人看法，其它人不喜勿喷！
                                                                                                                                         2019-12-31

CHDman, 2019-12-31:
OP群里小白真的太多了，伸手党也是一堆。真心喜欢折腾的专门搞一个群来，反馈各种问题一起解决问题，个人认为这样比较好。

Her0R0cket, 2019-12-31:
有些人为了名利什么都干的出来,国内抄作业的现象现在也没什么好转,合作的氛围还是不高,都喜欢拿过来盖自己的小楼,而不喜欢给别人的房子添砖加瓦.而且伸手党靠嘴已经劝退一堆各行业大神了,所以说不要在意那些伸手党的话,他们没有贡献也只会添乱.

luweiji, 2019-12-31:
谢谢大雕，帮不上什么忙，就提前说一声：春节快乐吧！

ForgotFun, 2019-12-31:
参加<跟着佐大学OpenWrt开发>培训班,让更多人的参与到开发中来吧。
http://forgotfun.org/2018/04/openwrt-training-2018.html

ForgotFun, 2019-12-31:
> 国内的开源环境到底是什么样呢？他们使用现成的开源的成果，却很少有人遵守开源协议。

所以现在做培训班，授人以如不如授人以渔。

smsource, 2019-12-31:
Coolsnowwolf加油

aoxiangfly, 2019-12-31:
Coolsnowwolf加油

jiangyunxing0, 2019-12-31:
哎！让大雕伤心了。

lbbboy, 2019-12-31:
fork的人90%都是小白吧，有能力开发的并不多。

a1768623020, 2019-12-31:
总的来说还是这个仓库拉低了编译固件的门槛，但是这里面Coolsnowwolf大大的贡献还是最大的，只不过名气以来之后，进来的小白太多了，我毕竟不是程序员，只能帮这解决一下故障，没什么代码贡献，不过还是要感谢Coolsnowwolf带给我们这么好用的固件。

chquse, 2019-12-31:
感谢大雕辛苦的贡献，祈望新年大雕顺心顺利！

lzd755, 2019-12-31:
让大雕伤心了，但我默默支持你，虽不能贡献代码，可以传僠你的精神，让更多的人走出去了解世界，了解什么是真正的开源精神，个别SB就别放心上，拉低档次。
感谢大雕辛苦的贡献，祈望新年大雕顺心顺利！

Myoko, 2019-12-31:
这个项目我刚接触不久！很多都不懂，但是我会继续学习搞下去。在这种情况下不管大大做什么决定我都支持

leaus, 2019-12-31:
我个人不懂得代码只会用，没能带来帮助，感谢大雕辛苦的贡献，祈望新年大雕顺心顺利！

dhxsy1994, 2019-12-31:
虽说自己是一个计软专业的学生，但是如果要说有贡献的话充其量就是能将这个项目推广给同样有动手能力的人。提到项目难度，可能对大神来说门槛很低，我还需要努力。想到这里，我觉得lean不必气馁，国内圈子风气不正，lean做出一个有节操的榜样，未来会好的。支持大雕，支持开源精神，遵守开源协议，从我做起！！

Hybrid-Force, 2019-12-31:
感谢 @coolsnowwolf 的贡献。

我不认为这是一个失败的项目，可能只是没有维护好。开源项目的维护往往不像大家想象的那么理想化，尤其是当国内用户（软性习惯未养成）和非技术型受众（硬性能力有限）占比很高的时候，就更难了。这个期望中的有序机制是需要相当的资源投入去实现和维护的，可能不是 @coolsnowwolf 一个人能够做到了，也不一定能够自发形成。

对于很多技术从业者来说，从代码到项目、社区和技术生态体系，需要的技能不同，往往也是从个人开发者到技术领袖的成长历程。

再次感谢 @coolsnowwolf，新年快乐。

yspcn, 2019-12-31:
> 我本身是做运营的，曾经负责管理日活跃2000w的网站，具体网站保密。对于这个问题我有几个看法，希望可以帮到你！也同时谢谢你的项目。给我帮助很多！
> 
> 一、先抛开迈克那sb不说。这种开源的项目在国内是很容易被复制的，其它的也都是你抄我，我抄他。国内市场就这样，咱们也没办法。就算注册了版权也没用，改了名字，随便加点自己的东西就成自己的了。既然避免不了，咱们只能心态好一些了。
> 
> 二、实用性建议：
> **虽然开源了，但是可以参考运营法则，应该适当做一些用户留存和用户的引导。**
> 
> 1）先说用户留存，要使你的固件更有效传播和分享。
> 可以在固件加入你固件的名字。有代表性。并在固件首页加入api接口，断网显示的内容和拨号联网后显示api接口内容
> 
> 2）api接口内容：
> 可以体现你的个人想法还有仓库链接，还要加入issues的入口链接。因为issues的链接很多人找不到入口，我第一次使用github的时候，我反正不知道有issues可以使用。甚至很多人没使用github的人就更不知道了，小白真的很傻
> 
> 3）api加入actions链接，因为这样可以简单的发布你的固件，实现一个固件生态循环。
> 
> 4）虽然这样固件不纯净了，但是可以引入更多的人，然后从中简单的引导，使大家知道这个仓库的地址还有作者的辛苦劳动。传播速度会快很多。
> 
> 5）如果安装上面的做了后，仓库地址流量肯定会变多，但是人就是这样，不管肯定不行。
> 
> 6）人多了，懂技术的人也会慢慢变多，可以简单的制度化。可以简单的做一些人流分类
> 
> 7）可以定期发布一个简单的招聘，比如招一些日常维护社区的人和招懂技术的，一同维护项目，当然没有工资哈。社区人流量多了，自然会有人愿意干，你要做的就是画出路线，让他们走就行了。不需要你花费多长时间。
> 
> 8）给这些人分开建立qq群，你可以制定简单的规矩，发布到群里，无规矩不成方圆。国家有法律，公司有公司制度。用制度管理人，是最简单的方法。
> 
> 9）定期发布招聘，时间可以自己定。可以更新不活跃的人，然后剔除他们，实现人员更新。
> 
> 三、现在谈谈迈克，这个sb是真的渣，偷别人的东西声称是自己的。如果有上面这些东西，大家就会更加懂的，谁的代码才是正版。
> 
> 马上新年了，祝大家新年快来！
> 以上是我的个人想法，其它人不喜勿喷！
> 2019-12-31

我感觉说的不错，赞一个，我是一个完全不懂代码的小白，现在编译成功了几个固件，自豪感满满，但是到哪里都说的是L大的源码，再次感谢L大的项目吧，
再说下pulls吧，我感觉应该是很多小白不会用，本来是要同步更新到自己仓库的，结果搞反了，我估计我第一次就是o(╯□╰)o

whxciotw, 2019-12-31:
首先感谢大雕的奉献和付出！

本人小白，没有编程经验，1年前仅有重装Windows的能力。
近1年来，随着入坑HomeAssistant，逐渐扩展了硬件、软件方面的一些能力，目前可以自行编译OP。但仍无法看懂代码。

感谢L大的辛勤付出，没有您的长期投入，是无法将OP普及至我这层级的小白的。

最后，支持您的任何决定。

Mokee-nmy, 2019-12-31:
莫愁前路无知己，支持Lean！

Hybrid-Force, 2019-12-31:
> 我一直想象的那种人人为我，我为人人，大家一起来维护一个项目的景象始终没有出现，也没有出现的迹象。
>
> 维护这个项目的过程中，主动分享的同学并不多。很多来汇报问题的人是以一种小白求大大免费解决问题，解决完就走人的方式来的。
>
> 然而既不愿提供足够的信息，截图日志什么都没有，也不愿写一些自己尝试的过程供后人参考。
>
> issues [...]

话说这些文字看起来怎么和 [@clowwindy 当年说的那么像](https://github.com/shadowsocks/shadowsocks-windows/issues/293#issuecomment-132253168)。。。

onepow, 2019-12-31:
感谢L大的 贡献，第一次LEDE系统编译是在 ESIRE的 YOUTUBE视频中介绍你的这个开源项目中完成的， 非常棒的项目，我内心非常支持和敬仰开源项目的，尤其是优秀的开源项目， 向你致敬，
另外像 那些使用你源代码稍做改动后 又不遵守开源协议并宣称是自已成果的人渣表示深深鄙视。

doumeki, 2019-12-31:
mike的事基本上都是小白在带节奏，那句从来没有我抄别人的，只有别人抄我的也是很久之前发的了，并没有针对代码本生。果然印证了那句一粉赛十黑，以后大家都没得玩。

pudding1111, 2019-12-31:
感谢付出。

ch1993128, 2019-12-31:
唉，一直在考试，打算寒假去研究下大雕的project。

lean老大拿来便用的op项目很多人是没有任何基础的，这样也就很常见了，很多人都只是用户，不是程序员。并不能作出啥贡献，可能也包括我在内

openwrt.org上面的开发指南很详细和系统，可是一直没时间去看。大雕可以在项目主页置顶一下，让更多人知道一下。

不过也可能没啥用，拿来就可以使用的东西，谁又愿意为其付出更多精力去系统的学一学呢。

但是不能说项目是失败的，我们崇尚白嫖，鼓励捷径，以成果为荣，却不看重基础行业和其他无人问津又必不可少行业的发展。这是文化大环境决定的，并不能为个人左右。

但若是没有改变，大概下一代也就这样了。

lean老大的项目是方便了我们的，一直也想码一点东西为lean老大项目做点事，可是最终还是事与愿违了。忙着毕业找工作，毕业的又永远有公司的要求比这个重要。

找了那么多借口也就为自己懒散得光明正大一点罢了。

最后说一句，我们专业学it的，不教vim,不讲gnu，不说make files很多学生到毕业不知道linus是谁，只知道gates是世界首富。

我们到底怎么了？

ch1993128, 2019-12-31:
谢谢lean老大

stock169, 2019-12-31:
这绝不是一个失败的项目，救活了多少矿难产品能让大众用上便宜功能丰富的设备，毕竟openwrt官方不会去支持这些矿难产品的，至于某些小人完全可以无视他们，无论是否关闭本项目都感谢L大无私的付出。

ch1993128, 2019-12-31:
> 这绝不是一个失败的项目，救活了多少矿难产品能让大众用上便宜功能丰富的设备，毕竟openwrt官方不会去支持这些矿难产品的，至于某些小人完全可以无视他们，无论是否关闭本项目都感谢L大无私的付出。

并没有阿，lean老大的项目主体就来源于op官方，lean在官方基础上作了很多优化和添加了很多软件包。并没有谁优谁劣。

官方op也并不少设备支持，只不过你们根本不上openwrt.org看看罢了，整个op项目的结构和开发者指南都在上面了。包括添加设备和添加删除软件包。国外op论坛活跃度非常高，开发者非常多。很多都是业余时间做的

ferre-han, 2019-12-31:
感谢 @coolsnowwolf 的贡献, 入手网件7800路由器,搜索发现你的代码,摸索研究,编译上千次 刷了上千次,提过2次问.自己关闭了,继续摸索,解决问题,我觉得来这里80%都是小白,环境问题没有其他渠道,小白只能来这里寻求帮助.......无奈......
再次感谢你

s3pc, 2019-12-31:
我大概就是你口中的伸手党，但我不想白伸手，如果接受btc支付，就给个地址。人到中年的我琐事缠身，在技术上暂时帮不到你，但我愿意为原创买单。也许你做这些不是为了经济利益，但我真心希望以我的方式回报你的付出。同时我也会继续学习，提高自己，尽量在技术上也帮到你。

lllrrr, 2019-12-31:
我是一小白，要感谢大雕带我入门，只因为自己不会写代码，没法出一份力。

ch1993128, 2019-12-31:
> 固件一切是迈克怼大雕后，大雕干脆不干了。用大雕的源码做收费固件，还怼大雕！真是啥人都有。
> 无论如何，最多贡献还是大雕的源码

这啥哦，收费固件啥情况，开源不可以收费的阿，做相关服务可以，我也挂闲鱼给人解决麻烦，编译东西收点10块8块小钱。申请版权什么情况

com388488, 2019-12-31:
希望可以组建一个开发组，把一些爱好的雕神的固件的玩家组建起来。一起开发，一些繁琐小事给开发组解决。且不是大神一个人撑起整个项目。再次，谴责比尔迈克那个傻逼，这种人走不远。其实也有很多人编译大雕固件传播大雕固件.会自觉注明出处.....国内环镜造就小白就会做伸手党。很多人只是想单纯科学上网。那还不如让他们直接去找那此编译大神固件的追随者，云编译我也建议删。直接让他们找那些已写好的github项目。

说这些是大神如果还继续开发下的一些建议，真停更，相信某傻逼就更狂了。ps:大神停更我也支持。

zuosc, 2019-12-31:
个人觉得需要把SS ssr V2Ray  完全隔离在项目外。

太多人因为这个入坑，天天讨论这个， 一点营养都没有。这也吸引了一大帮垃圾小白。

因为这些，也容易被监管。

stl88083365, 2019-12-31:
个人能力是硬伤，如果能有机会付出一些，我会毫不犹豫。值得说明一下就是因为国人心态问题，导致国内的一切开源项目都没有前景。毕竟都喜欢把饭喂到嘴边的感觉。仅仅是个人观点，感慨一下罢了！

garypang13, 2019-12-31:
支持停更. 本来就没什么存在的必要,小白就不应该去自己编译,有点动手能力的就自己去整官方的openwrt了,问题就出在这项目把初次编译的门槛降太低了,如果没有解决问题的能力的话,实际后面还有N道坎在等着,

ermiao, 2019-12-31:
> 个人觉得需要把SS ssr V2Ray 完全隔离在项目外。
> 
> 太多人因为这个入坑，一群天天讨论这个， 一点营养都没有。这也吸引了一大帮垃圾小白。
> 
> 还有关乎免流、去广告这些，最好也分离出去。这些容易使得项目被监管。
> 
> 简而言之就是区分用户群，专业的东西让专业的人来维护。简单的 东西，有利益相关的东西分离出去。
> 
> 这样更纯粹。
> 
> 技术是技术，其他是其他

这话说的真是没毛病……

dgulan, 2019-12-31:
首先感谢Lean 大大的分享。你的付出，给我这样的小白带来了方便。
本来不想发言的：
     在早期上网的时候，注册了霏凡论坛，默默的在那个论坛刷了好多年，也认识了很多的朋友……后来霏凡关闭了。自认算是个老网民吧。
    我是跟着Lean大的教程，从开虚拟机、装ubuntu，一步一步学会编译固件的。相当于从0开始。
（中间有许多小问题，都是靠着百度解决了一部分，也有一些仍然还不会）
自认为是个喜欢折腾的人，也有热心，愿意将自己的经验分享。
希望项目不要像霏凡一样，到最后只能走到尽头。

[177 more replies]

closed 2020-02-19 by coolsnowwolf