coolsnowwolf/lede #2658: SSR PLUS+ V2RAY 不支持 Let's Encrypt 证书

Kaiyuan, 2020-01-01:
WebSocket+TLS+Web ，证书用的 Let's Encrypt，能显示启动但是访问不了，然后我换成 AlphaSSL 的就可以正常...

ggsjj, 2020-01-01:
我就是用 Let's Encrypt都正常使用，你V2ray配置问题，还有提问题，日志都贴上来。让别人怎么查。 日志！日志！  不发错误日志上来还不如不提

SharerMax, 2020-01-01:
切换到 Let's Encrypt的证书后，看看web服务正常不正常，如果web服务不正常，说明你的证书有问题或者证书配置有问题

Kaiyuan, 2020-01-01:
> 切换到Let's Encrypt的证书后，看看web服务正常不正常，如果web服务不正常，说明你的证书有问题或者证书配置有问题

Web 正常，多个浏览器都正常，Windows 和 Mac 火狐、Chrome、Safari

SharerMax, 2020-01-01:
试试其它客户端好使不（比如Windows平台的v2rayN），好尴尬，没有WebSocket+TLS+Web的节点，我这也试不了

Kaiyuan, 2020-01-01:
> 试试其它客户端好使不（比如Windows平台的v2rayN），好尴尬，没有WebSocket+TLS+Web的节点，我这也试不了

手机，电脑都可以正常，就是 SSR PLUS+ 不能。

SharerMax, 2020-01-01:
我这好使，我刚刚找别人要了个节点试了下。方便发下你的配置界面看看么，发的话注意隐藏ip等信息

Kaiyuan, 2020-01-01:
证书生成：

```
acme.sh --installcert --issue --dns dns_cf -d *.abc.com -d abc.com \
--key-file       /var/www/.ssl/abc.com/privkey.pem    \
--fullchain-file /var/www/.ssl/abc.com/fullchain.pem    \
--reloadcmd     "systemctl restart nginx"
```

nginx:
```
server {
    listen 443 ssl http2;
    listen [::]:443 ssl http2;

    server_name 111.abc.com;
    root /var/www/abc.com/;

    access_log /var/log/nginx/access_abc.com.log;
    error_log /var/log/nginx/error_abc.com.log warn;

    # SSL
    ssl_certificate /var/www/.ssl/abc.com/privkey.pem;
    ssl_certificate_key /var/www/.ssl/abc.com/fullchain.pem;

    # index.php
    index index.php index.html;

    location / {
        
    }

    location /download/ {
        proxy_redirect off;
        proxy_pass http://127.0.0.1:161;
        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";
        proxy_set_header Host $http_host;

        # Show realip in v2ray access.log
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_intercept_errors on;
    }

}

# HTTP redirect
server {
    listen 80;
    listen [::]:80;

    server_name .l4h.win;

    return 301 https://111.abc.com/$request_uri;
}
```

v2ray:
```json
{
    "log": {
        "access": "/var/log/v2ray/access.log",
        "error": "/var/log/v2ray/error.log",
        "loglevel": "warning"
    },
    "inbounds": [
    {
        "sniffing": {
        "enabled": true,
        "destOverride": [
          "http",
          "tls"
        ]
      },
      "port": 161,
      "listen":"127.0.0.1",
      "protocol": "vmess",
      "settings": {
        "clients": [
          {
            "id": "asdasdasdasdasdasdasdasd",
            "alterId": 64,
            "security": "chacha20-poly1305"
          }
        ]
      },
      "streamSettings": {
        "network": "ws",
        "wsSettings": {
        "path": "/download/"
        }
      }
    }
  ],
  "inboundDetour": [
        {
            "protocol": "shadowsocks",
            "port": 110,
            "settings": {
                "method": "aes-256-cfb",
                "password": "dqdsfqwrddasdqq",
                "udp": true,
                "level": 1
            }
        }
  ],
  "outbounds": [
    {
      "protocol": "freedom",
      "settings": {}
    },
    {
      "protocol": "blackhole",
      "settings": {},
      "tag": "block"
    }
  ],
  "routing": {
    "domainStrategy": "AsIs",
    "rules": [
      {
        "type": "field",
        "outboundTag": "block",
        "protocol": [
          "bittorrent"
        ]
      }
    ]
  }
}
```

Kaiyuan, 2020-01-01:
> 我这好使，我刚刚找别人要了个节点试了下。方便发下你的配置界面看看么，发的话注意隐藏ip等信息
> 
> 我用的是最新的代码 R9.12.31

证书生成：

```
acme.sh --installcert --issue --dns dns_cf -d *.abc.com -d abc.com \
--key-file       /var/www/.ssl/abc.com/privkey.pem    \
--fullchain-file /var/www/.ssl/abc.com/fullchain.pem    \
--reloadcmd     "systemctl restart nginx"
```

nginx:
```
server {
    listen 443 ssl http2;
    listen [::]:443 ssl http2;

    server_name 111.abc.com;
    root /var/www/abc.com/;

    access_log /var/log/nginx/access_abc.com.log;
    error_log /var/log/nginx/error_abc.com.log warn;

    # SSL
    ssl_certificate /var/www/.ssl/abc.com/privkey.pem;
    ssl_certificate_key /var/www/.ssl/abc.com/fullchain.pem;

    # index.php
    index index.php index.html;

    location / {
        
    }

    location /download/ {
        proxy_redirect off;
        proxy_pass http://127.0.0.1:161;
        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";
        proxy_set_header Host $http_host;

        # Show realip in v2ray access.log
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_intercept_errors on;
    }

}

# HTTP redirect
server {
    listen 80;
    listen [::]:80;

    server_name .abc.com;

    return 301 https://111.abc.com/$request_uri;
}
```

v2ray:
```json
{
    "log": {
        "access": "/var/log/v2ray/access.log",
        "error": "/var/log/v2ray/error.log",
        "loglevel": "warning"
    },
    "inbounds": [
    {
        "sniffing": {
        "enabled": true,
        "destOverride": [
          "http",
          "tls"
        ]
      },
      "port": 161,
      "listen":"127.0.0.1",
      "protocol": "vmess",
      "settings": {
        "clients": [
          {
            "id": "asdasdasdasdasdasdasdasd",
            "alterId": 64,
            "security": "chacha20-poly1305"
          }
        ]
      },
      "streamSettings": {
        "network": "ws",
        "wsSettings": {
        "path": "/download/"
        }
      }
    }
  ],
  "inboundDetour": [
        {
            "protocol": "shadowsocks",
            "port": 110,
            "settings": {
                "method": "aes-256-cfb",
                "password": "dqdsfqwrddasdqq",
                "udp": true,
                "level": 1
            }
        }
  ],
  "outbounds": [
    {
      "protocol": "freedom",
      "settings": {}
    },
    {
      "protocol": "blackhole",
      "settings": {},
      "tag": "block"
    }
  ],
  "routing": {
    "domainStrategy": "AsIs",
    "rules": [
      {
        "type": "field",
        "outboundTag": "block",
        "protocol": [
          "bittorrent"
        ]
      }
    ]
  }
}
```
[image]

SharerMax, 2020-01-01:
服务端地址填域名，不是路径

0x5826, 2020-01-01:
服务器的证书要用fullchain那个

closed 2020-01-01 by Kaiyuan