coolsnowwolf/lede #2659: 编译出来的固件不支持无线桥接。

hong26, 2020-01-01:
机器型号是斐讯k2，要设置同网段无线桥接，在l大您在恩山发布的固件里面可以成功设置，但我自己编译出来的固件却无法设置无线桥接，显示无线未开启或未关联。请问是否是我编译的时候没有选中什么相关插件造成这个原因的呢？
[image]
[image]

SharerMax, 2020-01-01:
我之前好像遇到过，但不是K3,是xiaomi的，我选择的是重启大法，重启路由器，重启wifi接口，恢复

hong26, 2020-01-01:
> 我之前好像遇到过，但不是K2,是xiaomi的，我选择的是重启大法，重启路由器，重启wifi接口，恢复

我试过重启，无效！停用wifi再开启也试过，同样没用。请问你重启WiFi接口是通过什么方法重启的呢？

SharerMax, 2020-01-01:
`/etc/init.d/network reload`
我建议你直接移除，重新添加再试试

hong26, 2020-01-01:
> `/etc/init.d/network reload`
> 我建议你直接移除，重新添加再试试

谢谢你的建议，试过了，还是不行。估计是固件的驱动问题。

ch1993128, 2020-01-04:
裴讯k2要专用mt驱动，git上面有，找一下。你的2.4g频段都没显示，没装好驱动

hong26, 2020-01-09:
> 裴讯k2要专用mt驱动，git上面有，找一下。你的2.4g频段都没显示，没装好驱动

您的意思是l大这个源码里面的k2无线驱动有问题吗？需要怎么替换呢？可否详细指导一下。

closed 2020-02-19 by coolsnowwolf