coolsnowwolf/lede #2660: 关于openssl-1.0.2q下100-Configure-afalg-support.patch无法patch的问题

zanjie1999, 2020-01-01:
代码基于2020年1月1日最新代码

给一个ar9331的古董路由器TS-D084编译固件

看了一下,通过```package/libs/openssl/Makefile```,自动下载了1.0.2q版本的openssl

由于比较菜,没能看懂```100-Configure-afalg-support.patch```中的修改规则无力处理

编译输出内容:

```......
touch /home/sparkle/lede/staging_dir/target-mips_24kc_musl/root-ar71xx/stamp/.kmod-cryptodev_installed
echo "kmod-cryptodev" >> /home/sparkle/lede/staging_dir/target-mips_24kc_musl/pkginfo/cryptodev-linux.default.install
make[3]: Leaving directory '/home/sparkle/lede/package/kernel/cryptodev-linux'
time: package/kernel/cryptodev-linux/compile#0.13#0.03#0.15
make[3]: Entering directory '/home/sparkle/lede/package/libs/openssl'
touch /home/sparkle/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.prepared_7dba65fee6fc527f73054d3c1c9c9f97_6664517399ebbbc92a37c5bb081b5c53_check
. /home/sparkle/lede/include/shell.sh; gzip -dc /home/sparkle/lede/dl/openssl-1.0.2q.tar.gz | tar -C /home/sparkle/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /home/sparkle/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q

Applying ./patches/100-Configure-afalg-support.patch using plaintext: 
patching file Configure
Hunk #1 FAILED at 1532.
1 out of 1 hunk FAILED -- saving rejects to file Configure.rej
Patch failed!  Please fix ./patches/100-Configure-afalg-support.patch!
Makefile:347: recipe for target '/home/sparkle/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.prepared_7dba65fee6fc527f73054d3c1c9c9f97_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/sparkle/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.prepared_7dba65fee6fc527f73054d3c1c9c9f97_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/sparkle/lede/package/libs/openssl'
time: package/libs/openssl/compile#0.22#0.06#0.22
package/Makefile:107: recipe for target 'package/libs/openssl/compile' failed
make[2]: *** [package/libs/openssl/compile] Error 2
make[2]: Leaving directory '/home/sparkle/lede'
package/Makefile:103: recipe for target '/home/sparkle/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/sparkle/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sparkle/lede'
/home/sparkle/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2```

SharerMax, 2020-01-01:
如果是我，我可能会这样尝试
先执行 `make clean`, 然后删除 `.config` 重新执行下面的过程
`./scripts/feeds update -a `
`./scripts/feeds install -a`
`make menuconfig` 选择配置（使用默认配置，只选架构，机器型号）
然后 `make -j1 V=s` 重新编译

如果能编译通过，我再选择添加其它的组件后重新编译

如果不能我就把那个出错的patch删了，再试试

最后还是不好使，问问Google & 其它人 或者 弃疗 ~~

zanjie1999, 2020-01-02:
经过测试，只选择路由器型号，按默认的配置进行编译，也会有同样的问题，关于删除此path，正在测试中。。。

jonsonjordan, 2020-01-02:
我猜是你本地网络环境的锅，删掉lede目录，全局科学重新git后用默认配置编译一次试试

SharerMax, 2020-01-02:
> 我猜是你本地网络环境的锅，删掉lede目录，全局科学重新git后用默认配置编译一次试试

你这么说，有可能，我记我之前编译xiaomi的时候好像也是这个openssl，没下载全，然后挂全局就好使了

zanjie1999, 2020-01-02:
> > 我猜是你本地网络环境的锅，删掉lede目录，全局科学重新git后用默认配置编译一次试试
> 
> 你这么说，有可能，我记我之前编译xiaomi的时候好像也是这个openssl，没下载全，然后挂全局就好使了

 我是拿GitHub的Actions编译的，自己这边全局hk和全局jp也有试过，这大概不是网络问题

zanjie1999, 2020-01-02:
目前删除了

SharerMax, 2020-01-02:
> 目前删除了
> 100-Configure-afalg-support.patch
> 120-strip-cflags-from-binary.patch
> 正在重新编译中

删patch虽然是下下策，死马当活马医吧。生命不息，折腾不止~~

zanjie1999, 2020-01-03:
删除openssl的所有path后编译失败
```2020-01-02T08:31:19.6047145Z NOTE: If in doubt, on Unix-ish systems use './config'.
2020-01-02T08:31:19.6047486Z make[3]: *** [/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.configured_bb0f5b41ba156c868ab99b436e931024_95ecd1b761ddfb86c5f1f1f6e80da36f] Error 1
2020-01-02T08:31:19.6047595Z time: package/libs/openssl/compile#0.29#0.14#0.34
2020-01-02T08:31:19.6047683Z make[2]: *** [package/libs/openssl/compile] Error 2
2020-01-02T08:31:19.6048129Z make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
2020-01-02T08:31:19.6048229Z make: *** [world] Error 2
2020-01-02T08:31:19.6048572Z touch /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.prepared_d901b24b56aabaa2a6bf4c211c902a4b_6664517399ebbbc92a37c5bb081b5c53_check
2020-01-02T08:31:19.6049628Z . /home/runner/work/lede/lede/include/shell.sh; gzip -dc /home/runner/work/lede/lede/dl/openssl-1.0.2q.tar.gz | tar -C /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.. -xf - 
2020-01-02T08:31:19.6050023Z [ ! -d ./src/ ] || cp -fpR ./src/. /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q
2020-01-02T08:31:19.6050471Z touch /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.prepared_d901b24b56aabaa2a6bf4c211c902a4b_6664517399ebbbc92a37c5bb081b5c53
2020-01-02T08:31:19.6051026Z rm -f /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.configured_*
2020-01-02T08:31:19.6051436Z rm -f /home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/stamp/.openssl_installed
2020-01-02T08:31:19.6052195Z [ -f /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.configured_bb0f5b41ba156c868ab99b436e931024_95ecd1b761ddfb86c5f1f1f6e80da36f ] || { rm -f /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/*.so.* /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/*.a; find /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q -name \*.o | xargs rm -f; }
2020-01-02T08:31:19.6053501Z (cd /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q; ./Configure linux-mips-openwrt --prefix=/usr --libdir=lib --openssldir=/etc/ssl -I/home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/usr/include -I/home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/include -I/home/runner/work/lede/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/include -I/home/runner/work/lede/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/runner/work/lede/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include -L/home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/lib -L/home/runner/work/lede/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/lib -L/home/runner/work/lede/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib -znow -zrelro -Wl,--gc-sections shared no-heartbeats no-sha0 no-ssl2-method no-ssl3-method no-ec2m no-err no-camellia no-idea no-seed no-mdc2 no-whirlpool no-hw-padlock no-gost no-dtls no-dtls1 no-comp )
2020-01-02T08:31:19.6053967Z Configuring for linux-mips-openwrt
2020-01-02T08:31:19.6054311Z Makefile:347: recipe for target '/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/openssl-1.0.2q/.configured_bb0f5b41ba156c868ab99b436e931024_95ecd1b761ddfb86c5f1f1f6e80da36f' failed
2020-01-02T08:31:19.6054588Z make[3]: Leaving directory '/home/runner/work/lede/lede/package/libs/openssl'
2020-01-02T08:31:19.6054839Z package/Makefile:107: recipe for target 'package/libs/openssl/compile' failed
2020-01-02T08:31:19.6055068Z make[2]: Leaving directory '/home/runner/work/lede/lede'
2020-01-02T08:31:19.6055366Z package/Makefile:103: recipe for target '/home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
2020-01-02T08:31:19.6055596Z make[1]: Leaving directory '/home/runner/work/lede/lede'
2020-01-02T08:31:19.6055845Z /home/runner/work/lede/lede/include/toplevel.mk:216: recipe for target 'world' failed
2020-01-02T08:31:19.6057981Z ##[error]Process completed with exit code 2.
2020-01-02T08:31:19.6079745Z Cleaning up orphan processes
```

allenzt, 2020-01-03:
我看openssl package的提交日志，openssl的版本应该是1.1.1d呀

openssl package git commit:
2019-12-09 21:08 CN_SZTL      o OpenSSL: bump to v1.1.1d (#2332)
2019-11-09 02:16 LEAN-ESX     o qBittorrent: rollback qt5 and openssl
2019-10-25 06:51 LEAN-ESX     o openssl: OPENSSL_WITH_ASM default
2019-10-25 06:47 LEAN-ESX     o openssl: bump to v1.1.1d
2019-10-14 03:18 LEAN-ESX     o openssl: optimization speed for x64 （Compile with SSE2 and optimized assembly code)
2019-02-13 23:15 LEAN-ESX     o openssl: revert 1.0.2                                                       
2019-02-13 23:03 LEAN-ESX     o openssl: optimizations based on ARCH/small flash
2019-01-03 19:29 LEAN-ESX     o Merge branch master of https://github.com/coolsnowwolf/lede                 
2018-10-22 15:18 LEAN-ESX     o openssl:enable openssl multiple threads by default
2018-10-14 19:22 LEAN-ESX     o Enable optimization for openssl speed instead of size                       
2018-08-23 17:40 coolsnowwolf o sync with OpenWrt v18.06.1 stable new R8.1 version

所以，你是否更新代码到最新了

第二，你可以手动打patch
patch -p1 < 100-Configure-afalg-support.patch

zanjie1999, 2020-01-03:
> 我看openssl package的提交日志，openssl的版本应该是1.1.1d呀
>
> openssl package git commit:
> 2019-12-09 21:08 CN_SZTL o OpenSSL: bump to v1.1.1d (#2332)
> 2019-11-09 02:16 LEAN-ESX o qBittorrent: rollback qt5 and openssl [...]

Actions有个自动更新代码的任务，每天编译前会更新一次，先排除掉代码老的问题

关于openssl 1.1.1的话我这里在编译菜单中可以看到有一个openssl和一个openssl1.1的选项，翻编译的package可以得知openssl那一个也就是1.0.2q那个版本是默认引用的

至于为什么会用这一个也不排除是不是这个CPU比较老

closed 2020-02-19 by coolsnowwolf