coolsnowwolf/lede #2662: 编译gl-inet 的gl-ar750s报错，错误日志如下，请问是什么编译出错呢？谢谢！

PilotSnail, 2020-01-02:
错误日志如下：
scripts/Makefile.build:303: recipe for target 'arch/mips/kernel/module.o' failed
make[7]: *** [arch/mips/kernel/module.o] Error 1
make[6]: *** [arch/mips/kernel] Error 2
make[5]: *** [arch/mips] Error 2
scripts/Makefile.build:544: recipe for target 'arch/mips/kernel' failed
make[4]: *** [/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-4.19.91/.image] Error 2
Makefile:1052: recipe for target 'arch/mips' failed
make[3]: *** [install] Error 2
make[5]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-4.19.91'
time: target/linux/install#1.37#0.43#1.81
Makefile:19: recipe for target '/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-4.19.91/.image' failed
make[2]: *** [target/linux/install] Error 2
make[4]: Leaving directory '/home/runner/work/lede/lede/target/linux/ath79'
Makefile:13: recipe for target 'install' failed
make[3]: Leaving directory '/home/runner/work/lede/lede/target/linux'
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/stamp/.target_install] Error 2
target/Makefile:19: recipe for target '/home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/stamp/.target_install' failed
make[1]: Leaving directory '/home/runner/work/lede/lede'
/home/runner/work/lede/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
##[error]Process completed with exit code 2.

gsblay, 2020-01-02:
这个路由器我选 ATH79 来编译就没有成功过。要想 4.19 的固件编译出来，好像需要在 luci 中把 pptp-server 选中， 但这个编译出来的固件刷入路由器以后，5G 始终无法驱动起来，只有 2.4G.
最后，老老实实地选 ar71xx 编译，内核选 4.9 (4.14无法通过编译)，无线全部正常。

PilotSnail, 2020-01-02:
5G没信号似乎是Firmware中默认使用了ath10k-firmware-qca988x-ct固件，5G芯片未被驱动。取消选择ath10k-firmware-qca988x-ct，选择ath10k-firmware-qca9887-ct。编译，ath10k-firmware-qca988x-ct没有正常驱动QCA9887芯片。我现在编译的就没有5G信号，我回家试试看看前面这个解决方案。

gsblay, 2020-01-02:
我这边在 ATH79 下试过选不带 ct 的 fireware 编译过，同时将 Kernel modules 中的 Wireless Drivers 也选为对应的没有 ct 的选项，编译出来的固件依然无法驱动 5G.

PilotSnail, 2020-01-02:
按照你的建议下午编译报错是这样的
Collected errors:
 * check_data_file_clashes: Package bridge wants to install file /home/drcare/lede/build_dir/target-mips_24kc_musl/root-ath79/usr/sbin/brctl
        But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package bridge.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/drcare/lede'
package/Makefile:104: recipe for target '/home/drcare/lede/staging_dir/target-mips_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/drcare/lede/staging_dir/target-mips_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/drcare/lede'
/home/drcare/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

gsblay, 2020-01-02:
4.19.88可以按上面我说的通过编译，4.19.91不行。
我也是小白，不知原因！

Mokee-nmy, 2020-01-02:
我之前编译p2w-619ac一次成功，后来换了一个就不行，通过尝试以下方式排除问题，最终成功编译，目前没遇到问题了。
可以尝试进行make download -j1 V=s，看看是否都下载成功；
包下载都正常，可以尝试编译内核，make target/linux/compile V=s；
如果都正常了，可以尝试sudo apt-get update，更新软件源；
软件源都是最新的话可以尝试换软件源，然后在sudo apt-get update；

PilotSnail, 2020-01-02:
看了国外的一些社区讨论，貌似用ath79还真没有成功的哦，谢谢

gsblay, 2020-01-03:
Gl.inet的不少路由器，我自己碰到的 MT-300-V2, N300, Ar750S,B1300都这样,自己编译的固件都需要采用 uboot 刷入才行。

closed 2020-01-16 by PilotSnail