coolsnowwolf/lede #2666: v2ray直接下载makefile

fuqiang03, 2020-01-02:
不会上传代码 只会他在这里
方便升级
include $(TOPDIR)/rules.mk

PKG_NAME:=v2ray
PKG_VERSION:=4.22.0
PKG_MAINTAINER:=xxxx

ifeq ($(ARCH),mipsel)
	PKG_ARCH:=mipsle
endif

ifeq ($(ARCH),mips)
	PKG_ARCH:=mips
endif

ifeq ($(ARCH),x86_64)
	PKG_ARCH:=64
endif

ifeq ($(ARCH),arm)
	PKG_ARCH:=armv7
endif

ifeq ($(ARCH),aarch64)
	PKG_ARCH:=armv8
endif

PKG_SOURCE:=v2ray-linux-$(PKG_ARCH).zip
PKG_SOURCE_URL:=https://github.com/v2ray/v2ray-core/releases/download/v$(PKG_VERSION)/
PKG_BUILD_DIR:=$(BUILD_DIR)/$(PKG_NAME)-$(PKG_VERSION)
UNZIP_DIR:=$(BUILD_DIR)/$(PKG_NAME)-$(PKG_VERSION)/$(PKG_NAME)-unzip
PKG_HASH:=skip

include $(INCLUDE_DIR)/package.mk

define Package/$(PKG_NAME)
	SECTION:=net
	CATEGORY:=Network
	TITLE:=V2Ray is a cross-platform proxy software
	DEPENDS:=+ca-certificates 
	URL:=https://github.com/v2ray/v2ray-core
endef

define Package/$(PKG_NAME)/description
	V2Ray is a cross-platform proxy software
endef

define Build/Prepare
	mkdir -vp $(UNZIP_DIR)
	unzip -od $(UNZIP_DIR) $(DL_DIR)/$(PKG_SOURCE)
endef

define Build/Compile
endef

define Package/$(PKG_NAME)/install
	$(INSTALL_DIR) $(1)/usr/bin
	$(INSTALL_BIN) $(UNZIP_DIR)/v2ray $(1)/usr/bin/v2ray
	$(INSTALL_BIN) $(UNZIP_DIR)/v2ctl $(1)/usr/bin/v2ctl
	$(INSTALL_BIN) $(UNZIP_DIR)/geosite.dat $(1)/usr/bin/geosite.dat
	$(INSTALL_BIN) $(UNZIP_DIR)/geoip.dat $(1)/usr/bin/geoip.dat
	chmod 755 $(1)/usr/bin/*
endef

$(eval $(call BuildPackage,$(PKG_NAME)))

stephenchou888, 2020-01-02:
请问是直接替换LUCI下的MAKE，直接编译IPK吗？

fuqiang03, 2020-01-02:
不是  这个是下载执行程序用的文件

stephenchou888, 2020-01-03:
谢谢您。

lp123sun, 2020-01-04:
怎么更新到4.22.1，发现编译的一直是4.21.3呢

fuqiang03, 2020-01-04:
PKG_VERSION:=4.22.1

lp123sun, 2020-01-04:
> PKG_VERSION:=4.22.1

试过了，编译的文件名是4.22.1，通过/usr/bin/v2ray/v2ray -version看的，还是4.21.3

fuqiang03, 2020-01-04:
原来dl下面的要删除

lp123sun, 2020-01-04:
> 原来dl下面的要删除
> 要不还是以前的版本

原来dl下面没看到相关的包呀，倒是/lean/package/lean/v2ray/files/下面有各个系统已经编译好的包

fuqiang03, 2020-01-04:
不能存在两个同名的文件的

Mattraks, 2020-01-04:
[v2ray.zip](https://github.com/coolsnowwolf/lede/files/4021914/v2ray.zip)
经过修改测试可用

lp123sun, 2020-01-04:
> 不能存在两个同名的文件的
> 
> 自己可以下载对应的文件直接替换页可以的

去v2ray-core下载的v2ray-linux-64.zip文件，解压提取的文件直接放进编译好的lede里面替换v2ray，启动不了SSR Plus+

fuqiang03, 2020-01-05:
给权限了吗

zcluo, 2020-01-05:
给下日志

获取 Outlook for iOS<https://aka.ms/o0ukef>

lp123sun, 2020-01-05:
> 给权限了吗

给了，755权限

lp123sun, 2020-01-05:
> 给下日志 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

日志文件默认在什么路径，软件界面的日子很简单，就一个started，但是软件的主界面显示未运行。

fuqiang03, 2020-01-05:
配置好后 点击保存 可以手动执行下脚本文件 看看出现什么错误

lp123sun, 2020-01-05:
> 配置好后 点击保存 可以手动执行下脚本文件 看看出现什么错误

root@OpenWrt:~# /usr/bin/v2ray/v2ray -version
V2Ray 4.22.1 (V2Fly, a community-driven edition of V2Ray.) Custom (go1.13.5 linux/amd64)
A unified platform for anti-censorship.
root@OpenWrt:~# /usr/bin/v2ray/v2ray -config /var/etc/v2-ssr-retcp.json
V2Ray 4.22.1 (V2Fly, a community-driven edition of V2Ray.) Custom (go1.13.5 linux/amd64)
A unified platform for anti-censorship.
main: failed to read config file: /var/etc/v2-ssr-retcp.json > v2ray.com/core/main/json: failed to execute v2ctl to convert config file. > v2ray.com/core/common/platform/ctlcmd: v2ctl doesn't exist > stat /usr/bin/v2ray/v2ctl: no such file or directory
root@OpenWrt:~# /usr/bin/v2ray/v2ray -config /var/etc/v2-ssr-retcp.json
V2Ray 4.22.1 (V2Fly, a community-driven edition of V2Ray.) Custom (go1.13.5 linux/amd64)
A unified platform for anti-censorship.
2020/01/05 07:02:42 [Warning] v2ray.com/core: V2Ray 4.22.1 started
Killed
root@OpenWrt:~# /usr/bin/v2ray/v2ray -config /var/etc/v2-ssr-retcp.json
V2Ray 4.22.1 (V2Fly, a community-driven edition of V2Ray.) Custom (go1.13.5 linux/amd64)
A unified platform for anti-censorship.
Failed to start v2ray.com/core/app/proxyman/inbound: failed to listen TCP on 1234 > v2ray.com/core/transport/internet: failed to listen on address: 0.0.0.0:1234 > listen tcp 0.0.0.0:1234: bind: address already in use

开始就上传v2ray一个文件，提示缺少v2ctl，v2ctl上传后运行，过了一会提示killed，好像是个自动守护的弄得。
这样貌似可以了，估计是缺少了v2ctl。

fuqiang03, 2020-01-05:
要是能扶墙的话尽量go语言编译,体积会小很多
我就是因为没有出国能力 只能直接下客户端了

fuqiang03, 2020-01-06:
export GOPROXY=https://goproxy.io
如果有使用go语言编译的 需要输入上面的命令
参考资料
https://github.com/kuoruan/openwrt-v2ray/blob/master/README.md

closed 2020-01-06 by fuqiang03