coolsnowwolf/lede #2674: DDNS 显示已注册的 IP 地址bug

inlolo, 2020-01-03:
测试了一下google的ddns，发现已注册的 IP 地址是已经过时的ip
[image]
后台日志显示已经更新了ip
[image]
到google后台也显示更新了最新ip
[image]

inlolo, 2020-01-03:
实际功能使用没问题

coolsnowwolf, 2020-01-04:
DNS 缓存问题。这里看到是路由器本身的缓存，有时间差

inlolo, 2020-01-08:
> DNS 缓存问题。这里看到是路由器本身的缓存，有时间差

是本身缓存问题

jiuub, 2020-01-12:
> DNS 缓存问题。这里看到是路由器本身的缓存，有时间差

我也碰到这问题了，我发现其实是Turbo ACC 网络加速下的DNS加速导致的, 关闭该功能就正常获取IP了。我想问问能否nslookup时，指定DNS服务器，解决这个缓存问题

closed 2020-02-19 by coolsnowwolf