coolsnowwolf/lede #2677: 20200104编译报错, 求助,谢谢交流分享

TerryWang159, 2020-01-03:
OpenWrt-libtool: link: x86_64-openwrt-linux-musl-gcc-ranlib .libs/libracoon.a
OpenWrt-libtool: link: ( cd ".libs" && rm -f "libracoon.la" && ln -s "../libracoon.la" "libracoon.la" )
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../.. -I./../libipsec   -I/home/terrywang/lede/staging_dir/target-x86_64_m                      usl/usr/include -D_GNU_SOURCE -include ./src/include-glibc/glibc-bugs.h -I./src/include-glibc -I./src/include-glibc  -I/home/t                      errywang/lede/staging_dir/target-x86_64_musl/usr/include -I/home/terrywang/lede/staging_dir/target-x86_64_musl/include -I/home                      /terrywang/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/terrywang/lede/staging_dir/toolchain-x86_64_gc                      c-7.5.0_musl/include/fortify -I/home/terrywang/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -I../../src/racoon/mis                      sing -D_GNU_SOURCE -include ../../src/include-glibc/glibc-bugs.h -I../../src/include-glibc -I../../src/include-glibc -DSYSCONF                      DIR=\"/etc\" -DADMINPORTDIR=\"/var/racoon\" -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-vari                      able -Wno-error=unused-result -iremap/home/terrywang/lede/build_dir/target-x86_64_musl/linux-x86_64/ipsec-tools-0.8.2:ipsec-to                      ols-0.8.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/terrywang/le                      de/staging_dir/target-x86_64_musl/usr/include -I/home/terrywang/lede/staging_dir/target-x86_64_musl/include -I/home/terrywang/                      lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/terrywang/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_mus                      l/include/fortify -I/home/terrywang/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include  -MT eaytest.o -MD -MP -MF .deps/                      eaytest.Tpo -c -o eaytest.o eaytest.c
eaytest.c: In function 'rsa_verify_with_pubkey':
eaytest.c:106:41: error: dereferencing pointer to incomplete type 'EVP_PKEY {aka struct evp_pkey_st}'
  error = eay_check_rsasign(src, sig, evp->pkey.rsa);
                                         ^~
Makefile:900: recipe for target 'eaytest.o' failed
make[8]: *** [eaytest.o] Error 1
make[8]: Leaving directory '/home/terrywang/lede/build_dir/target-x86_64_musl/linux-x86_64/ipsec-tools-0.8.2/src/racoon'
Makefile:682: recipe for target 'all' failed
make[7]: *** [all] Error 2
make[7]: Leaving directory '/home/terrywang/lede/build_dir/target-x86_64_musl/linux-x86_64/ipsec-tools-0.8.2/src/racoon'
Makefile:363: recipe for target 'all-recursive' failed
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory '/home/terrywang/lede/build_dir/target-x86_64_musl/linux-x86_64/ipsec-tools-0.8.2/src'
Makefile:422: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/terrywang/lede/build_dir/target-x86_64_musl/linux-x86_64/ipsec-tools-0.8.2'
Makefile:348: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/terrywang/lede/build_dir/target-x86_64_musl/linux-x86_64/ipsec-tools-0.8.2'
Makefile:103: recipe for target '/home/terrywang/lede/build_dir/target-x86_64_musl/linux-x86_64/ipsec-tools-0.8.2/.built' fail                      ed
make[3]: *** [/home/terrywang/lede/build_dir/target-x86_64_musl/linux-x86_64/ipsec-tools-0.8.2/.built] Error 2
make[3]: Leaving directory '/home/terrywang/lede/feeds/packages/net/ipsec-tools'
time: package/feeds/packages/ipsec-tools/compile#11.89#1.82#16.63
package/Makefile:107: recipe for target 'package/feeds/packages/ipsec-tools/compile' failed
make[2]: *** [package/feeds/packages/ipsec-tools/compile] Error 2
make[2]: Leaving directory '/home/terrywang/lede'
package/Makefile:103: recipe for target '/home/terrywang/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/terrywang/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/terrywang/lede'
/home/terrywang/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

求助大神,谢谢~

stl88083365, 2020-01-04:
工具链都没编译通过。。请严格按照大雕首页的readme，执行

ferre-han, 2020-01-04:
删了 重新整啊

zcluo, 2020-01-04:
ubuntu 14.04没问题

获取 Outlook for iOS<https://aka.ms/o0ukef>

CYaNu, 2020-01-04:
还是那句话 第一次编译-j1 如果不是那么可以重新clone后尝试

closed 2020-01-10 by TerryWang159