coolsnowwolf/lede #2680: 编译报错的看一下

ch1993128, 2020-01-04:
建议严格按编译步骤

命令行输入 sudo apt-get update ，然后输入
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler
安装linux依赖库

git clone https://github.com/coolsnowwolf/lede   克隆lean老大项目地址

cd lede 挂载项目目录

./scripts/feeds update -a
./scripts/feeds install -a  更新feeds

make menuconfig   定制编译机型，软件包和驱动

make download -j8 V=s  下载所有相关包（需要保证富强环境）

make tools/install -j8 V=s
make toolchain/install -j8 V=s  编译工具和工具链

make -j8 V=s  编译openwrt镜像

建议每一步检查一下报错，软件包不冲突基本就不会有编译问题

ch1993128, 2020-01-04:
希望新入手lean项目的能少走点弯路，也希望能有更多人维护这个项目

CYaNu, 2020-01-04:
建议第一次编译-j1为好  容易出现莫名其妙的报错

Her0R0cket, 2020-01-05:
> 
> 
> 建议第一次编译-j1为好 容易出现莫名其妙的报错

对的-j8的话你永远不知道正确的报错信息,还有就是尽量不要乱选自己用不到的软件包,因为有些包真的是有问题的,不是因为编译者问题.

wz2869058, 2020-01-05:
建议一开始不要用-j8，就按L大的-j1，编译完一次再改

VictoriousRaptor, 2020-01-05:
建议在路由器上做好科学再编译
如果在本机做的话的话参考：
```
# curl
export http_proxy=http://localhost:1080
export https_proxy=https://localhost:1080
# or socks5
export http_proxy=socks5h://localhost:1080
export https_proxy=socks5h://localhost:1080

# git
git config --global http.proxy http://localhost:1080
git config --global https.proxy https://localhost:1080
```
[ref](https://gist.github.com/laispace/666dd7b27e9116faece6)

jzhou404, 2020-01-14:
必须在科学上网的环境下， 才可能编译成功。很多人， 都是失败在这里的。

另外，第一次， 建议按照L大的-j1单核编译。

make -j1 V=s （-j1 后面是线程数。第一次编译推荐用单线程，国内请尽量全局科学上网）

KevinMX, 2020-01-14:
``make -j8``其实就好。出问题再``make -j1 V=s``看报错就可以了。

closed 2020-02-19 by coolsnowwolf