coolsnowwolf/lede #2684: ssr-plus采用trojan节点，google play app无法连接网络！

jackwonglstc, 2020-01-04:
故障现象：youtube app 显示没有网络
根本原因：google play app无法连接网络！
 浏览器访问youtube一切正常，google网站访问也一切正常！

服务器采用的是编译的x64 lede固件，什么都没动，就是默认配置。安装在软路由上面，主要是为了用ssr-plus!
服务器三种模式都测试过，故障依旧，不管是全局还是绕过大陆模式都一样。
可能的分析：应该是app使用了特别的方式进行验证，而这种方式路由器并没有转发。或许是udp?
排除trojan服务器的问题，因为使用手机版的igniter，刷新一下youtube就可以了。而且后面播放视频也正常，显然是过了账号验证之后，要过一段时间才会再次验证。
当出现这种现象之后，使用其他翻墙软件刷新一下就可以继续使用了。过一段时间又会出现，估计是过一段时间会账号验证一次，只要APP验证账号，目前的路由器就会出问题。
也常识过把service.googleapis.cn这样地址加入强制走代理的清单，但是故障依旧。
不知道有没有其他人遇到过这样的问题？

SharerMax, 2020-01-04:
1. 在状态标签页，更新规则
2. 重启路由器后尝试

如果有使用其它DNS之类的服务请关闭

CYaNu, 2020-01-04:
trojan的ssl配置看看 h2去掉试试呢 两端

jackwonglstc, 2020-01-05:
应该和trojan配置无关，因为采用ss节点也一样出现这样的问题

jackwonglstc, 2020-01-05:
> 1. 在状态标签页，更新规则
> 2. 重启路由器后尝试
> 
> 如果有使用其它DNS之类的服务请关闭
用的默认lean的配置编译，没有动过其他也没有加装过其他任何服务
重启和刷新规则已经做过多次，故障依旧。
感觉是分流和转发出了问题，因为在手机端运行的任何翻墙软件均不会出现这个问题。
但是像twitter和电报之类的app均没有问题 就google的app有问题

ferre-han, 2020-01-05:
> > 1. 在状态标签页，更新规则
> > 2. 重启路由器后尝试
> > 
> > 如果有使用其它DNS之类的服务请关闭
> > 用的默认lean的配置编译，没有动过其他也没有加装过其他任何服务
> > 重启和刷新规则已经做过多次，故障依旧。
> > 感觉是分流和转发出了问题，因为在手机端运行的任何翻墙软件均不会出现这个问题。
> > 但是像twitter和电报之类的app均没有问题 就google的app有问题

那就是google的app有问题啦 跟雕神源码木有关系

CYaNu, 2020-01-05:
那么我怀疑可能是dns解析问题 用其他软件刷一下可以的话 会在缓存中留下解析 一段时间缓存失效那么问题会重现

CYaNu, 2020-01-05:
临时办法可以延长缓存生效时间 增大下缓存大小

jackwonglstc, 2020-01-05:
> 那么我怀疑可能是dns解析问题 用其他软件刷一下可以的话 会在缓存中留下解析 一段时间缓存失效那么问题会重现

有可能，有可能google app账号验证是通过APP直接发起请求，而不是通过HTTPS之类的协议，而这个服务器路由器没有正确的代理，导致误认为没有联网。
用其他VPN软件刷一下，就能正确验证是因为那个服务器正确联通了。因为他是定时验证的，只要不验证，就能一直使用，但是一旦验证，就会因为不能正确通信而导致报：没有联网错误。
但是如果是DNS解析问题，那么全局模式应该没有问题才对，但是全局模式一样有问题，而本地手机的代理软件就算设置了绕过大陆网站的分流模式也一样没有问题。
因此我还是怀疑是路由器上的SSR-PLUS的配置问题或者规则问题导致的。

CYaNu, 2020-01-05:
> 有可能，有可能google app账号验证是通过APP直接发起请求，而不是通过HTTPS之类的协议，而这个服务器路由器没有正确的代理，导致误认为没有联网。
> 用其他VPN软件刷一下，就能正确验证是因为那个服务器正确联通了。因为他是定时验证的，只要不验证，就能一直使用，但是一旦验证，就会因为不能正确通信而导致报：没有联网错误。 [...]

有可能

zhangdp, 2020-01-06:
国内google play域名是googleapis.cn，估计当成国内域名没走代理了，把这个域名设为强制走代理然后清空下dns手机断开重连下wifi试试

keivenliao, 2020-01-06:
我也遇到过一毛一样的问题，手机浏览器内访问google、youtube一切正常，但是各个Google相关app都无法访问，后来更换了smartdns后，到目前为止没有发生过。之前用的dnsforward 方案。推测应该与DNS有关。

jackwonglstc, 2020-01-08:
> 我也遇到过一毛一样的问题，手机浏览器内访问google、youtube一切正常，但是各个Google相关app都无法访问，后来更换了smartdns后，到目前为止没有发生过。之前用的dnsforward 方案。推测应该与DNS有关。

多谢提供的思路。确实我也用了dns forward,还有使用ssr-plus也是一样，还有使用clash也是一样。如果dns问题，全局也是有问题！让人疑惑。
现在我终于清楚是google账号管理和验证导致的问题。不知道google进行账号验证使用了什么独特的域名?现在还没搞清楚。
多谢你的思路，smartdns我去试试，看看效果如何？

hcym, 2020-01-08:
别乱搞什么dns，大雕的源码去掉所有无关插件只保留ss，肯定可以的，用官方源码也行，插件可能要自己找其他的git源

jackwonglstc, 2020-01-08:
最后的原因可能你想都不想不到。难怪我这几天折腾了几乎所有代理软件，几乎结果都一样！
其实代理全部没有问题，因为我用的是小米手机！
小米手机的miui系统对google 账号的认证进行了禁止，不允许他后台运行，好像是为了省电！但是google账号认证需要定期后台启动检查。于是悲剧了！
解决办法也简单，就是把智能省电模式对googleplay 应用进行关闭就一切正常了！

KevinZonda, 2020-01-09:
So now you can close the issue.

ffffjj, 2020-01-10:
我用的其他固件的SSR也出现了这样的问题，就一个安卓手机这个问题，应该是DNS的问题，我换节点可以上GOOGLE PLAY，但是YouTube就是视频加载不出来，现在也不知道怎么搞

coolsnowwolf, 2020-01-11:
某些手机系统内置的 DNS 问题，例如 MIUI和华为，刷为原生安卓即可解决

hcym, 2020-01-12:
难怪，一直用杂牌手机，谷歌app很少见到问题

Bnosora, 2020-01-12:
> 最后的原因可能你想都不想不到。难怪我这几天折腾了几乎所有代理软件，几乎结果都一样！
> 其实代理全部没有问题，因为我用的是小米手机！
> 小米手机的miui系统对google 账号的认证进行了禁止，不允许他后台运行，好像是为了省电！但是google账号认证需要定期后台启动检查。于是悲剧了！
> 解决办法也简单，就是把智能省电模式对googleplay 应用进行关闭就一切正常了！

这都能发现，厉害。之前miui10的时候我也遇到了这个问题，更到11就自己消失了。

Michaol, 2020-01-17:
我貌似发现了更深源的原因。如果是通过MIUI自己安装的框架，内置的Google服务配置文件，使用的是services.googleapis.cn。而通过apk mirror之类安装的框架，则是用的services.googleapis.com。这样导致了路由的GFW列表识别不出而不走梯子。我试了一下在SSP P+的强制走代理加入services.googleapis.cn，同时把加州google的216.58.197.195加入强制走代理WAN IP（216.58.197.0/24），手机上的google play、相册、油土鳖什么的app，都能顺利通过登录和连接了。

keivenliao, 2020-01-17:
> 我貌似发现了更深源的原因。如果是通过MIUI自己安装的框架，内置的Google服务配置文件，使用的是services.googleapis.cn。而通过apk mirror之类安装的框架，则是用的services.googleapis.com。这样导致了路由的GFW列表识别不出而不走梯子。我试了一下在SSP P+的强制走代理加入services.googleapis.cn，同时把加州google的216.58.197.195加入强制走代理WAN IP（216.58.197.0/24），手机上的google play、相册、油土鳖什么的app，都能顺利通过登录和连接了。

你这个信息蛮重要的。

Michaol, 2020-01-17:
补充，我遇到的问题是苹果顺利通过路由出去，但是小米的CC9上安装的google全家桶都不能出去，手机浏览器打开google完全没有问题，说明路由梯子没有问题。

Michaol, 2020-01-17:
测试更新，域名强制走代理加入services.googleapis.cn即可，不用强制加IP。各位不妨测试一下。

userxiaozhu, 2020-01-18:
> 测试更新，域名强制走代理加入services.googleapis.cn即可，不用强制加IP。各位不妨测试一下。

不行啊，我强制走代理然后dns解析到 216.58.200.3 ，然后根本ping不通，打开谷歌商店依旧显示未连接互联网

badcrazy, 2020-01-20:
> 我貌似发现了更深源的原因。如果是通过MIUI自己安装的框架，内置的Google服务配置文件，使用的是services.googleapis.cn。而通过apk mirror之类安装的框架，则是用的services.googleapis.com。这样导致了路由的GFW列表识别不出而不走梯子。我试了一下在SSP P+的强制走代理加入services.googleapis.cn，同时把加州google的216.58.197.195加入强制走代理WAN IP（216.58.197.0/24），手机上的google play、相册、油土鳖什么的app，都能顺利通过登录和连接了。
然而我也是小米手机，从miui9到11，从安卓6到10，从来没发现过这个问题，只是有些机场可以出国但是依然无法谷歌罢了

Michaol, 2020-01-21:
> 我貌似发现了更深源的原因。如果是通过MIUI自己安装的框架，内置的Google服务配置文件，使用的是services.googleapis.cn。而通过apk mirror之类安装的框架，则是用的services.googleapis.com。这样导致了路由的GFW列表识别不出而不走梯子。我试了一下在SSP [...]

我的小米6，最早就是通过apk mirror装框架，升级到现在的内测版依然无障碍。但是我老婆新买的CC9和我老妈的MAX2就有这个问题。我现在就是通过强制services.googleapis.cn走代理解决了。

Michaol, 2020-01-21:
> > 测试更新，域名强制走代理加入services.googleapis.cn即可，不用强制加IP。各位不妨测试一下。
> 
> 不行啊，我强制走代理然后dns解析到 216.58.200.3 ，然后根本ping不通，打开谷歌商店依旧显示未连接互联网

只需要域名services.googleapis.cn走代理，不需要强制解析DNS。另外打得开跟ping得通没有绝对关系，TCP跟ICMP，协议都不一样。

jackwonglstc, 2020-02-06:
> 我貌似发现了更深源的原因。如果是通过MIUI自己安装的框架，内置的Google服务配置文件，使用的是services.googleapis.cn。而通过apk mirror之类安装的框架，则是用的services.googleapis.com。这样导致了路由的GFW列表识别不出而不走梯子。我试了一下在SSP P+的强制走代理加入services.googleapis.cn，同时把加州google的216.58.197.195加入强制走代理WAN IP（216.58.197.0/24），手机上的google play、相册、油土鳖什么的app，都能顺利通过登录和连接了。

你的方法我早就试过了，事实证明不行。只要把google play服务的省电策略的智能省电关闭，就一切ok。目前miui已经到了11最新版本，貌似google play服务已经不再有省电策略的选项。这个故障已经不会再有了，小米已经意识到这个问题，做了修正。

jackwonglstc, 2020-02-06:
问题原因找到，和ssrplus无关

closed 2020-02-06 by jackwonglstc

repairwa11, 2020-02-28:
一加6T 系统版本号升级到Android10后也出现了这种情况，现在没法解决

3329415183, 2020-03-22:
你好 我现在也是这个问题 这么解决的？我都烦死了

repairwa11, 2020-03-22:
切换一下网络就好了，wifi和流量互切

zatenchan, 2020-03-30:
> 我貌似发现了更深源的原因。如果是通过MIUI自己安装的框架，内置的Google服务配置文件，使用的是services.googleapis.cn。而通过apk mirror之类安装的框架，则是用的services.googleapis.com。这样导致了路由的GFW列表识别不出而不走梯子。我试了一下在SSP P+的强制走代理加入services.googleapis.cn，同时把加州google的216.58.197.195加入强制走代理WAN IP（216.58.197.0/24），手机上的google play、相册、油土鳖什么的app，都能顺利通过登录和连接了。

我草 谢谢大佬，用adguard还以为是规则问题折腾了很久没解决，iphone的手机，用自带小火箭丝般顺滑，用路由翻各种卡顿，但是关了adguard所有规则又会好，刚刚看到你说的查了下adguard，规则里没有提到你说的这两个，抱着试试看加到了ssr里，结果竟然正常了。

wa008, 2020-04-25:
clash for Android 用户，可以在pc端clash生成的yml文件中添加services.googleapis.cn域名（我添加了com 和 cn），拷贝到手机上生成配置文件，问题就解决了
要判断是账号地域问题还是googleapis域名问题，可以用老王的梯子分辨一下

gozap99, 2020-05-03:
> clash for Android 用户，可以在pc端clash生成的yml文件中添加services.googleapis.cn域名（我添加了com 和 cn），拷贝到手机上生成配置文件，问题就解决了
> 要判断是账号地域问题还是googleapis域名问题，可以用老王的梯子分辨一下

我改了配置文件依然不行。。请问下是直接加域名和ip到yml文件里就可以了吗?

wa008, 2020-05-03:
> > clash for Android 用户，可以在pc端clash生成的yml文件中添加services.googleapis.cn域名（我添加了com 和 cn），拷贝到手机上生成配置文件，问题就解决了
> > 要判断是账号地域问题还是googleapis域名问题，可以用老王的梯子分辨一下
> 
> 我改了配置文件依然不行。。请问下是直接加域名和ip到yml文件里就可以了吗?

对的，我当时试的是可以的
但最近又不行了，感觉还有其他原因。。。

gozap99, 2020-05-03:
> > > clash for Android 用户，可以在pc端clash生成的yml文件中添加services.googleapis.cn域名（我添加了com 和 cn），拷贝到手机上生成配置文件，问题就解决了
> > > 要判断是账号地域问题还是googleapis域名问题，可以用老王的梯子分辨一下
> > 
> > 
> > 我改了配置文件依然不行。。请问下是直接加域名和ip到yml文件里就可以了吗?
> 
> 对的，我当时试的是可以的
> 但最近又不行了，感觉还有其他原因。。。

看来只能在手机端用clash才行了。。。
也有用这个办法的[https://www.xiaomi.cn/post/5317273](url)
你可以试试

gozap99, 2020-05-03:
> > > > clash for Android 用户，可以在pc端clash生成的yml文件中添加services.googleapis.cn域名（我添加了com 和 cn），拷贝到手机上生成配置文件，问题就解决了
> > > > 要判断是账号地域问题还是googleapis域名问题，可以用老王的梯子分辨一下
> > > 
> > > 
> > > 我改了配置文件依然不行。。请问下是直接加域名和ip到yml文件里就可以了吗?
> > 
> > 
> > 对的，我当时试的是可以的
> > 但最近又不行了，感觉还有其他原因。。。
> 
> 看来只能在手机端用clash才行了。。。
> 也有用这个办法的https://www.xiaomi.cn/post/5317273
> 你可以试试

forthxu, 2020-05-23:
> 最后的原因可能你想都不想不到。难怪我这几天折腾了几乎所有代理软件，几乎结果都一样！
> 其实代理全部没有问题，因为我用的是小米手机！
> 小米手机的miui系统对google 账号的认证进行了禁止，不允许他后台运行，好像是为了省电！但是google账号认证需要定期后台启动检查。于是悲剧了！
> 解决办法也简单，就是把智能省电模式对googleplay 应用进行关闭就一切正常了！

就是这个原因

zzddsg, 2020-06-10:
> 最后的原因可能你想都不想不到。难怪我这几天折腾了几乎所有代理软件，几乎结果都一样！
> 其实代理全部没有问题，因为我用的是小米手机！
> 小米手机的miui系统对google 账号的认证进行了禁止，不允许他后台运行，好像是为了省电！但是google账号认证需要定期后台启动检查。于是悲剧了！
> 解决办法也简单，就是把智能省电模式对googleplay 应用进行关闭就一切正常了！

miui11，我尝试了将services.googleapis.cn指定为216.58.197.26，结果google play无法运行，然后在miui中修改了智能省电模式：将Google Play改成无限制后，可以正常使用了! 所以miui11中还是有关闭google play后端运行的问题。

WinjayYu, 2020-06-25:
> 最后的原因可能你想都不想不到。难怪我这几天折腾了几乎所有代理软件，几乎结果都一样！
> 其实代理全部没有问题，因为我用的是小米手机！
> 小米手机的miui系统对google 账号的认证进行了禁止，不允许他后台运行，好像是为了省电！但是google账号认证需要定期后台启动检查。于是悲剧了！
> 解决办法也简单，就是把智能省电模式对googleplay 应用进行关闭就一切正常了！

太棒了 这个问题困扰了我好久

allbutone, 2020-07-23:
> 最后的原因可能你想都不想不到。难怪我这几天折腾了几乎所有代理软件，几乎结果都一样！
> 其实代理全部没有问题，因为我用的是小米手机！
> 小米手机的miui系统对google 账号的认证进行了禁止，不允许他后台运行，好像是为了省电！但是google账号认证需要定期后台启动检查。于是悲剧了！
> 解决办法也简单，就是把智能省电模式对googleplay 应用进行关闭就一切正常了！

终于解决了, 谢谢!

geitwhat, 2020-09-23:
> 最后的原因可能你想都不想不到。难怪我这几天折腾了几乎所有代理软件，几乎结果都一样！
> 其实代理全部没有问题，因为我用的是小米手机！
> 小米手机的miui系统对google 账号的认证进行了禁止，不允许他后台运行，好像是为了省电！但是google账号认证需要定期后台启动检查。于是悲剧了！
> 解决办法也简单，就是把智能省电模式对googleplay 应用进行关闭就一切正常了！

> 我用的其他固件的SSR也出现了这样的问题，就一个安卓手机这个问题，应该是DNS的问题，我换节点可以上GOOGLE PLAY，但是YouTube就是视频加载不出来，现在也不知道怎么搞

就是这样的, 只有youtube app这样, 在手机上上开个代理就可以访问了, 关闭代理之后youtube就可以访问, 长时间不用youtube, 就又不行了.

yangeren, 2020-10-30:
> 测试更新，域名强制走代理加入services.googleapis.cn即可，不用强制加IP。各位不妨测试一下。

效果立竿见影，谢啦！

renzhezhu1234, 2020-11-12:
> 切换一下网络就好了，wifi和流量互切

你这个问题才是正解，切了一下秒好

zatenchan, 2020-11-12:
> > 切换一下网络就好了，wifi和流量互切
> 
> 你这个问题才是正解，切了一下秒好

其实手机上访问看不了youtube app，电脑上正常，直接用手机梯子也正常的原因你们都没说到点子上，是因为列表的地址不全导致app上前面的视频广告域名没办法正确解析没加载出来，所以卡在了广告加载页面，解决办法很简单，
1、去广告 
2、把app上的几个广告域名强制走解析
最简单就是全局路由或者国内白名单模式，不信你们试试开全局路由是不是就能看了。这也就是为什么你们说的切换一下网络就会好的原因。

DvW1130, 2020-12-01:
> > 最后的原因可能你想都不想不到。难怪我这几天折腾了几乎所有代理软件，几乎结果都一样！
> > 其实代理全部没有问题，因为我用的是小米手机！
> > 小米手机的miui系统对google 账号的认证进行了禁止，不允许他后台运行，好像是为了省电！但是google账号认证需要定期后台启动检查。于是悲剧了！
> > 解决办法也简单，就是把智能省电模式对googleplay 应用进行关闭就一切正常了！
> 
> 就是这个原因

一加五氢10系统关闭google play store的省电优化真的好了hhh谢谢

xinng7, 2021-01-22:
我的iPhone和iPad也有这个问题,跟小米没有什么关系.

hzren120, 2021-03-03:
软路由，连接WiFi，我的Google手机打不开所有需要翻墙的app，但网页没问题；sony电视也不能用；但iPhone和mbp没问题。我的软路由是之前安装好的，印象中好像之前没有这种情况；最近换了个机场，出现了问题。而且WiFi在pixel手机上频繁掉线，提示“连接的网络有安全问题”。太崩溃了，找不到原因，有好心大神指教一下吗？万谢！

gethappy555, 2021-03-17:
> 问题原因找到，和ssrplus无关

有点关系的，同样的节点，我用passwall就可以连接，ssrplus不行

jouleken, 2021-07-09:
> 软路由，连接WiFi，我的Google手机打不开所有需要翻墙的app，但网页没问题；sony电视也不能用；但iPhone和mbp没问题。我的软路由是之前安装好的，印象中好像之前没有这种情况；最近换了个机场，出现了问题。而且WiFi在pixel手机上频繁掉线，提示“连接的网络有安全问题”。太崩溃了，找不到原因，有好心大神指教一下吗？万谢！

你好，请问sony电视不要用的问题解决了吗？

hzren120, 2021-07-09:
没有解决，问题依旧，大佬有解决方案吗？

jouleken, 2021-07-09:
> 没有解决，问题依旧，大佬有解决方案吗？
> […](#)

我也没找到有效的方法

Migeking, 2021-07-13:
Turbo ACC 网络加速设置 里面关闭 dns 加速  ok！！！！

Migeking, 2021-07-13:
[image]

yan2035, 2021-07-14:
我也是这个情况，用v2ray打不开谷歌商店和YouTube，用老王vpn就能打开。不知道怎么办，

ajunajun133, 2021-09-01:
目前 我的  没有解决  添加了强制代理域名   也没有解决   但是切换手机自带的扶墙  可以连上  再切回路由器 就可以正常联网  路由器 开了UDP转发了     期待大佬给出 有效针对性的解决办法    这个确实有点闹心啊

wanggongwei, 2021-11-06:
各位大佬，我找到了一个原因，！
手机和平板都是三星，然后我把平板刷成港版系统后，谷歌系app全部正常，手机是国行系统，谷歌系app就用不了，
翻各种帖子，区别是三星港行没有屏蔽FCM，但是国行屏蔽了，所以我觉得可能和这个有关系，给大佬提供一个思路，希望有大佬可以解决

hzren120, 2021-11-07:
跟我可能没关系，我是原生Google系统连不上

wanggongwei, 2021-11-07:
[image]
我是在看这个人说的  我以为是这样子的
而且也确实和FCM有关系，我看我能连上Play商店的时候FCM是连接，FCM连不上，Play商店也连不上

[7 more replies]