coolsnowwolf/lede #2692: 经过的好心人能否指导一下如何编译lean的openwrt让他支持用主路由ros网关也可以翻墙的

norikohxb, 2020-01-05:
看了这个栏目的帖子慢慢学会了自己编译lean的openwrt，但是现在编译出来的lean的openwrt启用了v2ray以后，局域网的电脑网关必须是旁路由才可以科学上网，如果局域网的电脑网关是主路由是无法科学上网
虽然自己家里因为还有智能家居（博联设备）并不需要这种旁路由启用了v2ray就可以绑架主路由让他可以科学上网的功能
但是还是耿耿于怀想要知道是怎么编译出这种样子的openwrt，
经过的好心人高手们能否指点一二，哪怕是一点点线索也可以。谢谢了。

ferre-han, 2020-01-05:
换个功能强大的路由器啊

CYaNu, 2020-01-05:
主路由拨号
主路由网关设成旁路由 DNS设成旁路由 开启DHCP 网关为旁路由 DNS为旁路由
旁路路由网关设成主路由 DNS设成自己 关闭DHCP 关闭防火墙
试试吧

norikohxb, 2020-01-05:
感谢两位回帖你们都没有明白我的意思，恩山论坛一些人编译好的lean的openwrt，启动了v2ray以后，你局域网的网关照样是主路由，一样可以翻墙，不需要经过什么dhcp设置。你们并没有真正明白我的意思。

sqliuchang, 2020-01-05:
没听说过不需要设置旁路由作为网关就可以代理的固件。。。。

jonsonjordan, 2020-01-05:
不知道是你问的有问题还是我理解有问题，你这个问题就不是个问题，管你什么主路由旁路由，只要你设备流量经过科学的openwrt路由器，它就可以科学上网

shinyzgj, 2020-01-06:
你说的那些固件只是把配置直接固话进去了而已。这个不是功能单纯是配置问题。

CYaNu, 2020-01-06:
我说的情况基于ros+lede 情况  功能路由只有一个 
也许你双路由都是功能路由吧 帮不了你

legendchi, 2020-01-06:
你的问题很容易理解错误，首先科学上网要想成功就必须让流量经过你openwrt的v2ray代理，然而你的openwrt只是挂在ros下面的旁路由，这个要么你设置旁路由的网关和DNS，如果想用ros的网关和DNS，就必须在ros上做流量转发，把需要代理的浏览转发到openwrt的v2ray上。或者你把openwrt挂到你ros的上一级去，这是最简单的。

fishba, 2020-01-10:
@norikohxb 这样说吧.其实很简单.ros的lan口接入openwrt的lan口,然后这个口配置静态IP,加一条静态路由指向ros网关,ssr p开全局.           然后在ros的firewall里把国内IP反标记路由.这样就是国外的IP了..再在ip-routes加条静态路由指向openwrt的那个IP .routing mark选你标记的那个反路由.ros的dns设置成你的openwrt ip ...ros的dhcp里的dns指定为你的ROS IP  ..这样就完美了.

norikohxb, 2020-01-10:
没有你们说的那么复杂，设置来设置去，如果你是主路由旁路由都在虚拟机的试试看我说的这个openwrt
https://www.right.com.cn/forum/thread-1234564-1-1.html， 我不是推荐他的也不是广告，我现在也自己用lean的教程自己编译。但是这个人的openwrt，启用以后你只要开了ssr，你用主路由作为网关还是旁路由作为网关都可以科学上网，不需要设置什么routing mark，啥都不需要。能理解我的意思吗，你那些，其实很简单的，也还都是要操作。这家伙的openwrt，你启动以后设置好ssr，全网劫持，直接翻墙。也不需要你说的什么全局翻墙设置，我现在是绕过国内的设置，试试看吧，很神奇，我其实也没有用的openwrt，但是我按照lean大神的编译教程我就无法编译出劫持整个网络的openwrt

norikohxb, 2020-01-10:
请不会回复我，怎么在ros上设置或者修改，或者透明网桥之类的东西，我用他的openwrt，是根本就不需要设置，只要旁路由刷入了lusty的openwrt，并且启用了SSR 或者v2ray以后，整个局域网就自动翻墙了。甚至如果你是虚拟机上有好几个openwrt作为好几个各自分开的旁路由，这这个lusty的openwrt直接把你的整个局域网其他旁路由都给搞不正常了，一统天下，劫持你所有以主路由为网关的所有设备的访问。

9l, 2020-01-13:
是DHCP服务器开了强制选项，接管了你网络里所有的DHCP广播

norikohxb, 2020-01-13:
很不幸，不是的，我的ros没有做任何透明网桥和其他特别的设置。
始作俑不是在我的主路由上面，而是在lusty编译（不只是他）还有不少其他人编译的openwrt，启动以后就直接接管你的主路由了。
你不要怀疑什么我吧主路由又做了什么其他特殊的设置，没有，一点都没有。

greennyreborn, 2020-01-13:
> 很不幸，不是的，我的ros没有做任何透明网桥和其他特别的设置。
> 始作俑不是在我的主路由上面，而是在lusty编译（不只是他）还有不少其他人编译的openwrt，启动以后就直接接管你的主路由了。
> 你不要怀疑什么我吧主路由又做了什么其他特殊的设置，没有，一点都没有。

什么很不幸，你看清楚你楼上的人的回复了么，我觉得他说的很对。是你openwrt的dhcp设置覆盖了主路由的

9l, 2020-01-13:
> 很不幸，不是的，我的ros没有做任何透明网桥和其他特别的设置。
> 始作俑不是在我的主路由上面，而是在lusty编译（不只是他）还有不少其他人编译的openwrt，启动以后就直接接管你的主路由了。
> 你不要怀疑什么我吧主路由又做了什么其他特殊的设置，没有，一点都没有。

兄弟，我说的是openwrt开启DHCP服务器强制选项，没说你的ros

[image]

norikohxb, 2020-01-13:
opewrt没有开通dhcp，因为我的dhcp是由ros开的，openwrt里面的dhcp关闭的死死的，所以不是这个问题，你如果是主路由旁路由都是虚拟机的，你可以试试看我给你的编译的那个openwrt，很奇妙，一开启v2ray，绑架局域网的所有不管网关是主路由还是二级路由的，统统绑架。

9l, 2020-01-13:
不试，没有意义

jzhou404, 2020-01-14:
只要通过openwrt部署V2的客户端， 就都可以科学上网的， 和主路由没什么关系的

norikohxb, 2020-01-14:
只通过openwrt的旁路由部署了V2ray服务以后，客户机网关还是主路由依然可以翻墙。不需要把客户机的网关指向旁路由。而且主路由也没有做任何特别的设定，每一台客户机的网关都是主路由。

CYaNu, 2020-01-14:
且不说dhcp劫持 先说从旁路由转一圈很快吗 nat2次很好吗 没有杠的意思 就问题说问题

greennyreborn, 2020-01-14:
> 且不说lean开dhcp劫持就行了 先说从旁路由转一圈很快吗 nat2次很好吗 没有杠的意思 就问题说问题

只有科学上网流量会两次NAT，这个没有办法，正常流量的话openwrt可以相当于一个交换机，只会在主路由有一次nat

CYaNu, 2020-01-14:
> > 且不说lean开dhcp劫持就行了 先说从旁路由转一圈很快吗 nat2次很好吗 没有杠的意思 就问题说问题
> 
> 只有科学上网流量会两次NAT，这个没有办法，正常流量的话openwrt可以相当于一个交换机，只会在主路由有一次nat

根本不必走openwrt  ros做定向分流就行 国内直接出ros 国外走openwrt绕一圈

fishba, 2020-01-14:
@CYaNu 我说的方法.就是这样.只有国外走openwrt这个网关......国内都走ROS

Fddh2012, 2020-01-16:
楼上各位，你们别乱讲了，人家是要一个类似二层防火墙的东西，其实就是在网关和OP之间开了个透明网桥，网关什么都不用动，直接OP接入内网就可以实现内网全局FQ，我来给LZ答案把：因为我看到你在恩山的帖子了：把kmod-br-netfilter编译上就可以了。

CYaNu, 2020-01-16:
没有说乱讲 只是说怎么样效率更好 就算用了netfilter也是把包拉到旁路由去匹配路由表 要知道它是运行在旁路由上的 不藐视 不抬高 只是如是说

Fddh2012, 2020-01-17:
> 没有说乱讲 只是说怎么样效率更好 就算用了netfilter也是把包拉到旁路由去匹配路由表 要知道它是运行在旁路由上的 不藐视 不抬高 只是如是说

别人并没让你比较哪个效率更好，另外，两种方案我都有在使用，各有各的优势，并不需要死抓一条不放。

norikohxb, 2020-01-18:
> > 没有说乱讲 只是说怎么样效率更好 就算用了netfilter也是把包拉到旁路由去匹配路由表 要知道它是运行在旁路由上的 不藐视 不抬高 只是如是说
> 
> 别人并没让你比较哪个效率更好，另外，两种方案我都有在使用，各有各的优势，并不需要死抓一条不放。

果然是这个

norikohxb, 2020-01-18:
> 楼上各位，你们别乱讲了，人家是要一个类似二层防火墙的东西，其实就是在网关和OP之间开了个透明网桥，网关什么都不用动，直接OP接入内网就可以实现内网全局FQ，我来给LZ答案把：因为我看到你在恩山的帖子了：把kmod-br-netfilter编译上就可以了。

果然是这个，太感谢了，问了很久。其实我很奇怪，为什么那么多人不知道？包括用的人？我在恩山问，还有在这里问，回帖的大多数都是那种不知道的，甚至不知道有这么一个东西，都一直说我是ros我装了什么东西，或者是什么ros网关再指向旁路由（这种循环的方式很搞笑）当然我知道大神们笑而不语。其实几个月前Lean大大就说很早就有这个功能了，。今天终于按照你的方式也劫持了ros路由的网关和dns，甚至开了过滤软件，你dns用主路由的一样也可以过滤广告。

但是其实我还是不用这个方式的，因为我家里还有200多个博联智能家具需要不翻墙上网，这个方式开了就全部给我搞成都翻墙了。
我主要是一直不知道怎么实现耿耿于怀，今夜可以做个美梦了。这个就好像很早以前听到的一首歌，其实并不是自己现在想要去听，就是非要知道当时自己喜欢的歌曲叫什么名字。

非常感谢你解答！谢谢了，恩山那个回帖也是你回帖的对吧？

norikohxb, 2020-01-18:
> 楼上各位，你们别乱讲了，人家是要一个类似二层防火墙的东西，其实就是在网关和OP之间开了个透明网桥，网关什么都不用动，直接OP接入内网就可以实现内网全局FQ，我来给LZ答案把：因为我看到你在恩山的帖子了：把kmod-br-netfilter编译上就可以了。

能否再请教你一个问题，我最近在编译R1S的H5的friendlywrt，编译最新版本的之后我的adbyby无法更新 adblock的host lists，不管按手动更新多少次，始终都是940lines，，无法更新到5609lines，请问你知道如何解决这个问题吗？
我猜想是在 make menuconfig里面的配置问题，导致adbyby在friendlywrt里面无法正常更新。

norikohxb, 2020-01-18:
Sun Jan 19 04:25:47 2020 daemon.notice netifd: wan (1550): udhcpc: sending renew to 192.168.8.252
Sun Jan 19 04:25:47 2020 daemon.notice netifd: wan (1550): udhcpc: lease of 192.168.16.244 obtained, lease time 600
Sun Jan 19 04:26:10 2020 cron.info crond[6975]: crond (busybox 1.30.1) started, log level 5
Sun Jan 19 04:26:10 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 19 04:26:10 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 19 04:26:14 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 19 04:26:14 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 19 04:26:17 2020 daemon.crit dnsmasq[7632]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:26:17 2020 daemon.crit dnsmasq[7632]: FAILED to start up
Sun Jan 19 04:26:22 2020 daemon.crit dnsmasq[7700]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:26:22 2020 daemon.crit dnsmasq[7700]: FAILED to start up
Sun Jan 19 04:26:27 2020 daemon.crit dnsmasq[7713]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:26:27 2020 daemon.crit dnsmasq[7713]: FAILED to start up
Sun Jan 19 04:26:32 2020 daemon.crit dnsmasq[7790]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:26:32 2020 daemon.crit dnsmasq[7790]: FAILED to start up
Sun Jan 19 04:26:33 2020 cron.info crond[7817]: crond (busybox 1.30.1) started, log level 5
Sun Jan 19 04:26:33 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 19 04:26:33 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 19 04:26:37 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 19 04:26:37 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 19 04:26:40 2020 daemon.crit dnsmasq[8474]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:26:40 2020 daemon.crit dnsmasq[8474]: FAILED to start up
Sun Jan 19 04:26:42 2020 cron.info crond[8520]: crond (busybox 1.30.1) started, log level 5
Sun Jan 19 04:26:43 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 19 04:26:43 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 19 04:26:47 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 19 04:26:47 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 19 04:26:50 2020 daemon.crit dnsmasq[9170]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:26:50 2020 daemon.crit dnsmasq[9170]: FAILED to start up
Sun Jan 19 04:26:55 2020 daemon.crit dnsmasq[9186]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:26:55 2020 daemon.crit dnsmasq[9186]: FAILED to start up
Sun Jan 19 04:27:00 2020 daemon.crit dnsmasq[9199]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:27:00 2020 daemon.crit dnsmasq[9199]: FAILED to start up
Sun Jan 19 04:27:05 2020 daemon.crit dnsmasq[9208]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:27:05 2020 daemon.crit dnsmasq[9208]: FAILED to start up
Sun Jan 19 04:27:10 2020 daemon.crit dnsmasq[9218]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:27:10 2020 daemon.crit dnsmasq[9218]: FAILED to start up
Sun Jan 19 04:27:15 2020 daemon.crit dnsmasq[9228]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 04:27:15 2020 daemon.crit dnsmasq[9228]: FAILED to start up
Sun Jan 19 04:27:15 2020 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
这个是错误代码，。就是adbyby无法更新 hosts list，有人可以赐教吗？

Fddh2012, 2020-01-18:
> 果然是这个，太感谢了，问了很久。其实我很奇怪，为什么那么多人不知道？包括用的人？我在恩山问，还有在这里问，回帖的大多数都是那种不知道的，甚至不知道有这么一个东西，都一直说我是ros我装了什么东西，或者是什么ros网关再指向旁路由（这种循环的方式很搞笑）当然我知道大神们笑而不语。其实几个月前Lean大大就说很早就有这个功能了，。今天终于按照你的方式也劫持了ros路由的网关和dns，甚至开了过滤软件， [...]

SSR不是有访问控制么？只不过你这200多设置起来是很麻烦。
你上面这问题可以在群里问问，日志看起来是dnsmasq的问题，可能是DNS解析出了问题。

norikohxb, 2020-01-19:
> SSR不是有访问控制么？只不过你这200多设置起来是很麻烦。
> 你上面这问题可以在群里问问，日志看起来是dnsmasq的问题，可能是DNS解析出了问题。

ssr有访问控制，可以加入智能家居的ip段，然后adbbyby也有访问控制也可以加入智能家具的ip段，我都做过，但是感觉这样让200多个设备来openwrt跑一遭，判断一下再回流到ROS的主网关觉得耗费计算量，毕竟智能家居的配件有200多个ip，所以我其实就是圆一下当时自己觉得happyzhang，后来的lusty编译的openwrt那种可以劫持网关和dns的梦。
不过请教你一下你会认为这样劫持效率更高吗？
我主要是担心200个智能家居的设备请求每次请求都给openwrt造成一次返回ros的判断。
加上我ROS是可以根据不同ip段分配网关的。ROS的dhcp里面对于智能家居的网段分配的网关是ROS自身，对于非智能家居的网关分配的是Lean的Openwrt这样起来我觉得效率会高一点。

closed 2020-01-20 by norikohxb

freedomzn, 2020-05-08:
> 看了这个栏目的帖子慢慢学会了自己编译lean的openwrt，但是现在编译出来的lean的openwrt启用了v2ray以后，局域网的电脑网关必须是旁路由才可以科学上网，如果局域网的电脑网关是主路由是无法科学上网
> 虽然自己家里因为还有智能家居（博联设备）并不需要这种旁路由启用了v2ray就可以绑架主路由让他可以科学上网的功能
> 但是还是耿耿于怀想要知道是怎么编译出这种样子的openwrt，
> 经过的好心人高手们能否指点一二，哪怕是一点点线索也可以。谢谢了。

老兄，编译成功了吗？固件是否可以发一下呢？谢谢

Johnny-qian, 2022-11-29:
老兄您好，最近我个人也在研究这一块大概的逻辑知道但是具体的实现办法还在探索中
你说的模式应该是叫做二层透明网桥模式，软路由的2个接口工作在二层，数据包进到软路由中自动进行分流动作，加速走加速内网走内网国内走国内（这块之前有了解OVS可以实现，具体比较复杂）
目前个人正在搞二层透明网桥加DNS IPSET在加策略路由的方式看是否能实现。

nkcfc, 2024-01-24:
ipv6如何支持透明网桥呢？大家有兴趣讨论下吗？