coolsnowwolf/lede #2695: K2P (mt7621) WiFi won't up after startup (w/ temp fix)

oudoubleyang, 2020-01-05:
## K2P (mt7621) WiFi won't up after startup (w/ temp fix)
- Model: `Phicomm K2P (CpuMark : 4529.883072 Scores)`
- Ver.: `OpenWrt R9.12.03 / LuCI Master (git-19.338.43082-9e87e69)`

Issue: WiFi won't automatically turn up after startup. There's also no switch in wireless settings.
Fix: see code block below.

---

## K2P (mt7621) WiFi 启动后不会自启（附临时解决方案）
问题：可能由于K2P使用闭源驱动，启动后WiFi不会自启动。
有一种比较简单粗暴的解决方法：
在启动项中加入：
```bash
# start up script
ifconfig rax0 up
ifconfig ra0 up
ifconfig apclix0 up
ifconfig apcli0 up
```

KevinMX, 2020-01-07:
Same issue.
不过我这里是自己调整过wifi信道之后才出现的无法自启……调整5G信道为2之后自启就好了。

paldier, 2020-01-08:
5g的信道有2？

KevinMX, 2020-01-08:
@paldier 打错，是2.4G，不过感觉问题应该不在信道设置这里。我这里已经正常了。

chobitsos, 2020-01-08:
配置文件发一下呗

| |
Kevin
|
|
seedyoo@126.com
|
On 01/08/2020 12:15, KevinMX wrote:

@paldier 打错，是2.4G，不过感觉问题应该不在信道设置这里。我这里已经正常了。

KevinMX, 2020-01-08:
[wireless.tar.gz](https://github.com/coolsnowwolf/lede/files/4033610/wireless.tar.gz)

felix-fly, 2020-01-09:
同样的问题，不知何故

Javascript-Ninja, 2020-01-13:
Lean's lede already resolve this issue at coolsnowwolf/lede@87ddda1.
If you want use official openwrt, see Javascript-Ninja/openwrt@bad1269.

KevinMX, 2020-01-13:
似乎问题依旧存在。
再现步骤：在LuCI中保存配置刷入sysupgrade，重启后无线未启动。
手动reload之后再次重启，WiFi自启正常。
所以没有自启的话手动Reload一下就行啦。

vwxc2019, 2020-01-16:
刷入sysupgrade后第一次重启后无线未启动，手动reload之后再次重启，WiFi自启正常。
但是有时候重启后还会出现WIFI未启动的情况，重启测试十几次再次遇到WIFI未启动的情况。
有没有脚本可以代替Reload点击操作？

closed 2020-02-19 by coolsnowwolf