coolsnowwolf/lede #2704: 讨论下关于SSR检查服务器无法连接然后切换的问题

luckyyyyy, 2020-01-06:
目前的检测无法连接导致服务器切换的逻辑如下，在我使用这段时间我发现有个问题，有时候只是很小一段时间导致google无法访问，但是实际上浏览器可以正常使用，然后过段时间就恢复，我觉得是不是这段逻辑可以增加以下重试次数？大家觉得呢

```shell
#判断代理是否正常
check_proxy() {
/usr/bin/ssr-check www.google.com  80 $switch_time 1
if [ "$?" == "0" ]; then
 return 0
else
 /usr/bin/ssr-check www.baidu.com  80 $switch_time 1
 if [ "$?" == "0" ]; then
 #goole不通baidu通则不正常
 return 1
 else
 return 2
 fi
fi
return 0
}

```

luckyyyyy, 2020-01-06:
简单修改后 逻辑如下
```shell
check_proxy() {
  local result=0
  for i in {1..3}
  do
    /usr/bin/ssr-check www.google.com 80 $switch_time 1
    if [ "$?" == "0" ]; then
      echo "$(date "+%Y-%m-%d %H:%M:%S") Check Google Proxy Success, count=$i" >> /tmp/ssrplus.log
      result=0
      break
    else
      echo "$(date "+%Y-%m-%d %H:%M:%S") Check Google Proxy Fail, count=$i" >> /tmp/ssrplus.log
      /usr/bin/ssr-check www.baidu.com 80 $switch_time 1
      if [ "$?" == "0" ]; then
        result=1
      else
        result=2
      fi
    fi
    sleep 1;
  done
  return $ret;
}
```

luckyyyyy, 2020-01-06:
附上日志 的确有这个现象

[image]

coolsnowwolf, 2020-01-06:
有道理

luckyyyyy, 2020-01-06:
> 有道理

晚上我给UI加个设置后提交

PR #2706 (2020-01-06): luci-app-ssr-plus: added try count
有时候只不过是短暂性的无法访问, 多尝试一次就可以.
所以增加尝试次数, 默认每秒钟检查一次.

close #2704
closed 2020-01-06 by coolsnowwolf