coolsnowwolf/lede #2708: 重启后外接硬盘无法自动挂在

kyarazhan, 2020-01-06:
设备：竞斗云
固件：OpenWrt R9.12.31 / LuCI Master (git-19.338.43082-9e87e69)
路由器重启后，不能自动挂在外接硬盘和识别外接硬盘，需要将外接硬盘设备断电重启，才能识别。外接硬盘格式为ext4，西数4T、机械。

ferre-han, 2020-01-06:
之前能自动吗,
编译的时候选择自动挂载了吗,
你这问题直接Google就解决了

coolsnowwolf, 2020-01-06:
GPT 分区可能有问题，日志看看

kyarazhan, 2020-01-06:
> GPT 分区可能有问题，日志看看

Microsoft Windows [Version 10.0.18363.535]

shang@YAN-HOME C:\Users\shang
$ ssh root@192.168.2.1
root@192.168.2.1's password:

BusyBox v1.30.0 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r1848-aeceaafa
 -----------------------------------------------------

root@OpenWrt:~# logread
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000] Linux version 4.14.160 (kara@debian) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r1845-db9b77a6)) #0 SMP Mon Jan 6 07:04:00 2020
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] OF: fdt: Machine model: P&W R619AC 128M
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Mon Jan  6 23:53:54 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 130560
Mon Jan  6 23:53:54 2020 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat c0a26900, node_mem_map dfb79000
Mon Jan  6 23:53:54 2020 kern.debug kernel: [    0.000000]   Normal zone: 1152 pages used for memmap
Mon Jan  6 23:53:54 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mon Jan  6 23:53:54 2020 kern.debug kernel: [    0.000000]   Normal zone: 130560 pages, LIFO batch:31
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x3e4 with crng_init=0
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s29452 r8192 d23796 u61440
Mon Jan  6 23:53:54 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s29452 r8192 d23796 u61440 alloc=15*4096
Mon Jan  6 23:53:54 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 rootfstype=squashfs
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] Memory: 509308K/522240K available (4545K kernel code, 156K rwdata, 1244K rodata, 1024K init, 234K bss, 12932K reserved, 0K cma-reserved, 0K highmem)
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000] Virtual kernel memory layout:
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000]       .text : 0xc0208000 - 0xc0770460   (5538 kB)
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000]       .data : 0xc0a00000 - 0xc0a27200   ( 157 kB)
Mon Jan  6 23:53:54 2020 kern.notice kernel: [    0.000000]        .bss : 0xc0a29000 - 0xc0a63b18   ( 235 kB)
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] Hierarchical RCU implementation.
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000009] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000023] Switching to timer-based delay loop, resolution 20ns
Mon Jan  6 23:53:54 2020 kern.info kernel: [    0.000250] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[... cut ...]
Mon Jan  6 23:54:57 2020 daemon.info hostapd: wlan0: STA b4:69:21:ef:6f:7d WPA: pairwise key handshake completed (RSN)
Mon Jan  6 23:55:00 2020 cron.info crond[7294]: crond (busybox 1.30.0) started, log level 8
Mon Jan  6 23:55:00 2020 daemon.info procd: - init complete -
Mon Jan  6 23:55:00 2020 daemon.err urandom_seed[7307]: /sbin/urandom_seed: line 10: getrandom: not found
Mon Jan  6 23:55:00 2020 daemon.err urandom_seed[7307]: An error occured
Mon Jan  6 23:55:05 2020 authpriv.info dropbear[7396]: Child connection from 192.168.2.197:63172

kyarazhan, 2020-01-06:
> 之前能自动吗,
> 编译的时候选择自动挂载了吗,
> 你这问题直接Google就解决了

编译的时候勾选了

CYaNu, 2020-01-06:
Mon Jan 6 23:53:54 2020 kern.info kernel: [ 6.558566] usbcore: registered new interface driver usb-storage
Mon Jan 6 23:53:54 2020 kern.warn kernel: [ 11.913119] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
Mon Jan 6 23:53:54 2020 kern.warn kernel: [ 12.133370] usb 2-1: Device not responding to setup address.
Mon Jan 6 23:53:54 2020 kern.err kernel: [ 12.353073] usb 2-1: device not accepting address 2, error -71

改下BIOS中的USB设置 hand-off 都打开试试

kyarazhan, 2020-01-07:
> Mon Jan 6 23:53:54 2020 kern.info kernel: [ 6.558566] usbcore: registered new interface driver usb-storage
> Mon Jan 6 23:53:54 2020 kern.warn kernel: [ 11.913119] xhci-hcd xhci-hcd.0.auto: Timeout [...]

硬路由中如何设置呢？

CYaNu, 2020-01-07:
> 硬路由中如何设置呢？

重新menuconfig 配置下kernel的usb设置试试吧

kyarazhan, 2020-01-08:
> 重新menuconfig 配置下kernel的usb设置试试吧

多谢，原因已找到，竞斗云原装电源只能支持usb2.0，虽说接口是3.0的。在kernel的添加usb2.0的驱动就可以正常识别了

closed 2020-01-08 by kyarazhan