coolsnowwolf/lede #2711: 请老板: 官方版本加入雕神源码 后台运行插件失效问题

ferre-han, 2020-01-07:
### **切换线路或者修改后ssr+**
Tue Jan  7 13:45:17 2020 daemon.warn pdnsd[16048]: Caught signal 15. Exiting.
Tue Jan  7 13:45:17 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jan  7 13:45:17 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jan  7 13:45:21 2020 cron.err crond[18224]: crond (busybox 1.31.1) started, log level 5
Tue Jan  7 13:45:22 2020 daemon.info pdnsd[18422]: pdnsd-1.2.9b-par starting.
Tue Jan  7 13:45:22 2020 daemon.err pdnsd[18422]: Could not listen on tcp socket: Not a socket
Tue Jan  7 13:45:22 2020 daemon.err pdnsd[18422]: error in UDP recv: Not a socket
Tue Jan  7 13:45:22 2020 daemon.err pdnsd[18422]: error in UDP recv: Not a socket
.......
### **手动/etc/init.d/shadowsocksr restart** 没问题
Tue Jan  7 13:47:15 2020 daemon.warn pdnsd[18422]: Caught signal 15. Exiting.
Tue Jan  7 13:47:15 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jan  7 13:47:15 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jan  7 13:47:18 2020 cron.err crond[18991]: crond (busybox 1.31.1) started, log level 5
Tue Jan  7 13:47:19 2020 daemon.info pdnsd[19181]: pdnsd-1.2.9b-par starting.
Tue Jan  7 13:47:19 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jan  7 13:47:19 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

雕神之前说:[权限问题。trunk 版本的luci默认不再具有读写关键目录权限，需要显式声明，所以无法完成提交](https://github.com/coolsnowwolf/lede/issues/2676#issuecomment-570565201)

有没有哪位吊大老板给解决一下 @luckyyyyy

slitilanber, 2020-01-07:
我用起来完全正常。
环境: alpinelinux+lxc中运行18.06.4+自编译SSR PLUS+adbyby;  lxc中运行自编译大雕的版本也正常。

ferre-han, 2020-01-07:
> 我用起来完全正常。
> 环境: alpinelinux+lxc中运行18.06.4+自编译SSR PLUS+adbyby; lxc中运行自编译大雕的版本也正常。

哎呦 我用的19.07,一会尔换18.06试一试

ferre-han, 2020-01-07:
> 我用起来完全正常。
> 环境: alpinelinux+lxc中运行18.06.4+自编译SSR PLUS+adbyby; lxc中运行自编译大雕的版本也正常。

19.07 有问题

Her0R0cket, 2020-01-08:
大雕不喜欢新版luci,所以在他改变之前只能自己动手了

EnnawYang, 2020-01-08:
> 大雕不喜欢新版luci,所以在他改变之前只能自己动手了

不是说不喜欢 而是在等稳定

ferre-han, 2020-01-08:
> 大雕不喜欢新版luci,所以在他改变之前只能自己动手了

不折腾了dnsforwarder凑活配合使用吧

closed 2020-01-08 by ferre-han