coolsnowwolf/lede #2718: 最新版的gl.net b1300无法编译

jyk1987, 2020-01-08:
make[6]: Leaving directory `/home/jyk/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
make[6]: Entering directory `/home/jyk/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
[  1%] Building C object libbb/CMakeFiles/bb.dir/all_read.c.o
[  3%] Building C object libbb/CMakeFiles/bb.dir/concat_path_file.c.o
[  5%] Building C object libbb/CMakeFiles/bb.dir/copy_file.c.o
[  7%] Building C object libbb/CMakeFiles/bb.dir/copy_file_chunk.c.o
[  8%] Building C object libbb/CMakeFiles/bb.dir/gzip.c.o
[ 10%] Building C object libbb/CMakeFiles/bb.dir/gz_open.c.o
[ 12%] Building C object libbb/CMakeFiles/bb.dir/last_char_is.c.o
[ 14%] Building C object libbb/CMakeFiles/bb.dir/make_directory.c.o
[ 16%] Building C object libbb/CMakeFiles/bb.dir/mode_string.c.o
[ 17%] Building C object libbb/CMakeFiles/bb.dir/parse_mode.c.o
[ 19%] Building C object libbb/CMakeFiles/bb.dir/safe_strncpy.c.o
[ 21%] Building C object libbb/CMakeFiles/bb.dir/time_string.c.o
[ 23%] Building C object libbb/CMakeFiles/bb.dir/unarchive.c.o
/home/jyk/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f/libbb/unarchive.c: In function 'extract_archive':
/home/jyk/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f/libbb/unarchive.c:239:9: warning: ignoring return value of 'chown', declared with attribute warn_unused_result [-Wunused-result]
    chown(full_name, file_entry->uid, file_entry->gid);
         ^
[ 25%] Building C object libbb/CMakeFiles/bb.dir/unzip.c.o
/home/jyk/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f/libbb/unzip.c: In function 'unzip':
/home/jyk/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f/libbb/unzip.c:1051:7: warning: ignoring return value of 'fread', declared with attribute warn_unused_result [-Wunused-result]
  fread(buf, 1, 8, in_file);
       ^
[ 26%] Building C object libbb/CMakeFiles/bb.dir/wfopen.c.o
[ 28%] Building C object libbb/CMakeFiles/bb.dir/xfuncs.c.o
[ 30%] Building C object libbb/CMakeFiles/bb.dir/xreadlink.c.o
[ 32%] Linking C static library libbb.a
/usr/bin/ranlib: libbb.a: No space left on device
make[6]: *** [libbb/libbb.a] Error 1
make[6]: *** Deleting file `libbb/libbb.a'
make[6]: Leaving directory `/home/jyk/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
make[5]: *** [libbb/CMakeFiles/bb.dir/all] Error 2
make[5]: Leaving directory `/home/jyk/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/jyk/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
make[3]: *** [/home/jyk/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f/.built] Error 2
make[3]: Leaving directory `/home/jyk/lede/package/system/opkg'
time: package/system/opkg/host-compile#1.00#0.20#4.81
make[2]: *** [package/system/opkg/host/compile] Error 2
make[2]: Leaving directory `/home/jyk/lede'
make[1]: *** [/home/jyk/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/jyk/lede'
make: *** [world] 错误 2

Her0R0cket, 2020-01-08:
/usr/bin/ranlib: libbb.a: No space left on device     设备空间不足,编译的话至少留10g可用空间吧

jyk1987, 2020-01-08:
> /usr/bin/ranlib: libbb.a: No space left on device 设备空间不足,编译的话至少留10g可用空间吧

谢谢

closed 2020-01-08 by jyk1987