coolsnowwolf/lede #2721: luci-app-ssr-plus 彻底解决更新订阅慢的问题，顺便兼容SSD节点

luckyyyyy, 2020-01-08:
## 重要
由于各家机场不一样，如果需要更多的支持，`给我发邮件告诉我订阅地址我帮你适配`，当然非常小众或者魔改的不适配，只适配大众的，或者在下面自己更新这个代码，在下面给出更新后的版本，这个脚本过几天就提交 pull request，

## 使用方式
直接scp到你的路由器中 lua xxx.lua 即可

## 改善的问题
对比 shell 脚本加了很多智能判断，并且针对 shell 脚本有 io 上更彻底的优化，shell的 io 太重了，处理这种字符串的东西会很累。

## 说明
有BUG下面跟帖直接反馈即可

```lua
require 'nixio'
require 'luci.util'
require 'luci.jsonc'
require 'luci.sys'

-- these global functions are accessed all the time by the event handler
-- so caching them is worth the effort
local tinsert = table.insert
local ssub, slen, schar, srep, sbyte, sformat, sgsub =
      string.sub, string.len, string.char, string.rep, string.byte, string.format, string.gsub
local nodeResult = {} -- update result
local cache = {}

local function split(full, sep)
    full = full:gsub("%z", "")  -- 这里不是很清楚 有时候结尾带个\0
    local off, result = 1, {}
    while true do
        local nEnd = full:find(sep, off)
        if not nEnd then
            local res = ssub(full, off, slen(full))
            if #res > 0 then -- 过滤掉 \0
                tinsert(result, res)
            end
            break
        else
            tinsert(result, ssub(full, off, nEnd - 1))
            off = nEnd + slen(sep)
        end
    end
    return result
end

local function get_urlencode(c)
	return sformat("%%%02X", sbyte(c))
end

function urlEncode(szText)
    local str = szText:gsub("([^0-9a-zA-Z ])", get_urlencode)
    str = str:gsub(" ", "+")
    return str
end

local function trim(text)
    if not text or text == "" then
        return ""
    end
    return (sgsub(text, "^%s*(.-)%s*$", "%1"))
end

local function md5(content)
    -- lua md5sum 不太好调 有特殊字符 要转义太麻烦了
    return trim(luci.sys.exec('echo \"' .. urlEncode(content) .. '\" | md5sum | cut -d \" \"  -f1'))
end

local function base64Decode(text, safe)
    if not text then return '' end
    text = text:gsub("%z", "")
    if safe then
        text = text:gsub("_", "/")
        text = text:gsub("-", "+")
        local mod4 = #text % 4
        text = text .. string.sub('====', mod4 + 1)
    end
    return nixio.bin.b64decode(text):gsub("%z", "")
end

local function processData(szType, content)
    local result = {
        auth_enable = '0',
        switch_enable = '1',
        type = szType,
        local_port = 1234,
        timeout = 60, -- 不太确定 好像是死的
        fast_open = 0,
        kcp_enable = 0,
        kcp_port = 0,
        kcp_param = '--nocomp'
    }
    local hash
    if type(content) == 'string' then
        hash = md5(content)
    else
        hash = md5(luci.jsonc.stringify(content))
    end
    result.hashkey = hash
    if szType == 'ssr' then
        local dat = split(content, "/\\?")
        local hostInfo = split(dat[1], ':')
        result.server = hostInfo[1]
        result.server_port = hostInfo[2]
        result.protocol = hostInfo[3]
        result.encrypt_method = hostInfo[4]
        result.obfs = hostInfo[5]
        result.password = base64Decode(hostInfo[6], true)
        local params = {}
        for k, v in pairs(split(dat[2], '&')) do
            local t = split(v, '=')
            params[t[1]] = t[2]
        end
        result.obfs_param = base64Decode(params.bfsparam, true);
        result.protocol_param = base64Decode(params.protoparam, true);
        result.alias = "【" .. base64Decode(params.group, true) .. "】"
        result.alias = result.alias .. base64Decode(params.remarks, true)

    elseif szType == 'vmess' then
        local info = luci.jsonc.parse(content)
        result.type = 'v2ray'
        result.server = info.add
        result.server_port = info.port
        result.tcp_guise = "none"
        result.transport = info.net
        result.alter_id = info.aid
        result.vmess_id = info.id
        result.alias = info.ps
        result.ws_host = info.host
        result.ws_path = info.path
        result.h2_host = info.host
        result.h2_path = info.path
        if not info.security then
            result.security = 'auto'
        end
        if info.tls == 'tls' or info.tls == '1' then
            result.tls = '1'
        else
            result.tls = '0'
        end
    elseif szType == 'ssd' then
        result.type = 'ss'
        result.server = content.server
        result.server_port = content.port
        result.password = content.password
        result.encrypt_method_ss = content.encryption
        result.alias = '[' .. content.airport .. '] ' .. content.remarks
    end
    return result, hash
end

local function wget(url, count)
    count = count and count + 1 or 1
    if count > 4 then return end
    local raw = luci.sys.exec('wget-ssl --user-agent="Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.90 Safari/537.36" --no-check-certificate -t 3 -T 30 -O- ' .. url)
    if not raw then
        return wgetUrl(url, count)
    end
    return trim(raw)
end

local name = 'shadowsocksr'
local ucic = luci.model.uci.cursor()
local subscribe_url = ucic:get_first(name, 'server_subscribe', 'subscribe_url', {})
local proxy = ucic:get_first(name, 'server_subscribe', 'proxy', '0')
if proxy == '0' then -- 不使用代理更新的话先暂停
    print('服务正在暂停')
    luci.sys.init.stop(name)
end
for _, url in ipairs(subscribe_url) do
    local groupHash = md5(url)
    cache[groupHash] = {}
    nodeResult[groupHash] = setmetatable({}, { __index = cache[groupHash] })
    local raw = wget(url)
    local node, szType
    -- if raw:sub(-1) == string.char(0) then -- 这里不是很清楚 有时候结尾带个\0
    --     raw = raw:sub(1, #raw - 1)
    -- end
    -- SSD 似乎是这种格式 ssd:// 开头的
    -- 不知道 SS 什么格式 没支持
    -- 我不清楚其他机场是否这样
    if raw:find('ssd://') then
        szType = 'ssd'
        local nEnd = select(2, raw:find('ssd://'))
        node = base64Decode(raw:sub(nEnd + 1, #raw), true)
        node = luci.jsonc.parse(node)
        local extra = {
            airport = node.airport,
            port = node.port,
            encryption = node.encryption,
            password = node.password
        }
        local servers = {}
        -- SS里面包着 干脆直接这样
        for _, server in ipairs(node.servers) do
            tinsert(servers, setmetatable(server, { __index = extra }))
        end
        node = servers
    else
        -- ssd 外的格式
        node = split(base64Decode(raw, true), "\n")
    end
    for _, v in ipairs(node) do
        if v then
            local result, hash
            if szType == 'ssd' then
                result, hash = processData(szType, v)
                print('成功解析: ' .. szType ..' 节点')
            elseif not szType then
                local dat = split(v, "://")
                if dat and dat[1] and dat[2] then
                    result, hash = processData(dat[1], base64Decode( dat[2], true))
                    print('成功解析: ' .. dat[1] ..' 节点')
                end
            else
                print('跳过未知类型: ' .. szType)
            end
            -- print(hash, result)
            if hash and result then
                result.groupHash = groupHash
                tinsert(nodeResult[groupHash], result)
                cache[groupHash][hash] = nodeResult[groupHash][#nodeResult[groupHash]]
            end
        end
    end
    print('成功解析节点数量: ' ..#node)
end

-- diff
do
    local uciType = 'servers'
    local ucic = luci.model.uci.cursor()
    local add, del = 0, 0
    ucic:foreach(name, uciType, function(old)
        if
            (not nodeResult[old.groupHash] or not nodeResult[old.groupHash][old.hashkey]) or
            (nodeResult[old.groupHash][old.hashkey] and not nodeResult[old.groupHash][old.hashkey])
        then
            ucic:delete(name, old['.name'])
            del = del + 1
        else
            local dat = nodeResult[old.groupHash][old.hashkey]
            ucic:tset(name, old['.name'], dat)
            -- 标记一下
            setmetatable(nodeResult[old.groupHash][old.hashkey], { __index =  { _ignore = true } })
        end
    end)
    for k, v in pairs(nodeResult) do
        for kk, vv in ipairs(v) do
            if not vv._ignore then
                local section = ucic:add(name, uciType)
                ucic:tset(name, section, vv)
                add = add + 1
            end

        end
    end
    ucic:commit(name)
    -- 如果服务器已经不见了把帮换一个
    local globalServer = ucic:get_first(name, 'global', 'global_server', '')
    local firstServer = ucic:get_first(name, 'servers')
    if not ucic:get(name, globalServer) then
        if firstServer then
            ucic:set(name, ucic:get_first(name, 'global'), 'global_server', firstServer)
            ucic:commit(name)
            print('当前主服务器已更新，正在自动更换。')
        end
    end

    if proxy == '0' and firstServer then -- 不使用代理更新的话先暂停
        print('更新成功服务正在启动')
        luci.sys.init.start(name)
    end
    print('新增节点数量: ' ..add, '删除节点数量: ' .. del)
end

```

luckyyyyy, 2020-01-08:
@coolsnowwolf

coolsnowwolf, 2020-01-08:
是个很好的建议，目前的 shell 虽然已经很成熟，但是订阅超过 100 个节点在低性能设备上很容易超时（7620 单核之类），但重新造轮子也可能有新的bug，大家可以帮助提交下意见

denver9527, 2020-01-09:
半路出家，路过学习来的

denver9527, 2020-01-09:
小白路过，进来学习一番

greennyreborn, 2020-01-09:
不如还是整合到UI里面？

caicaicai21, 2020-01-11:
今天测试了一下订阅，好像出现解析错误

> 2020-01-11 21:24:55 Shadowsocks/ShadowsocksR 1 Threads Started!
> 	[C]: ?
> 	/usr/share/shadowsocksr/subscribe.lua:284: in main chunk
> 	[C]: in function 'xpcall'
> 	/usr/share/shadowsocksr/subscribe.lua:206: in function </usr/share/shadowsocksr/subscribe.lua:160>
> 	/usr/share/shadowsocksr/subscribe.lua:79: in function 'base64Decode'
> 	/usr/share/shadowsocksr/subscribe.lua:293: in function </usr/share/shadowsocksr/subscribe.lua:284>
> 2020-01-11 21:24:53 stack traceback:
> 2020-01-11 21:24:53 更新失败服务正在恢复
> 2020-01-11 21:24:53 发生错误, 正在恢复服务

好像是解析节点有问题，测试节点：
**ss://YWVzLTEyOC1nY206dGVzdDEyMzQ1Njc4@test.server:1111#%E6%B5%8B%E8%AF%95%E6%9C%8D%E5%8A%A1**

luckyyyyy, 2020-01-11:
> 今天测试了一下订阅，好像出现解析错误
>
>
> [image]
>
> 好像是解析节点有问题，测试节点：
>
> ss://YWVzLTEyOC1nY206dGVzdDEyMzQ1Njc4@test.server:1111#%E6%B5%8B%E8%AF%95%E6%9C%8D%E5%8A%A1

方便把订阅链接发给我吗？
邮件 我帮你改一下 
这个不是解析错误 是没支持 ss的我不知道什么格式 这看起来 有urlencode 和base64 不知是否所有ss都这样

caicaicai21, 2020-01-11:
> 方便把订阅链接发给我吗？
> 邮件 我帮你改一下
> 这个不是解析错误 是没支持 ss的我不知道什么格式 这看起来 有urlencode 和base64 不知是否所有ss都这样

订阅连接内容有涉及到某些信息，所以不是很方便分享
循环是正常的，只是按你说的，不支持这个格式的
单条内容大致是ss://base64编码的加密方式和密码(例如：aes-128-gcm:12345678)@服务器地址:端口号#节点备注

luckyyyyy, 2020-01-11:
你发几个例子给我即可 我明天处理&nbsp;

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: caicaicai21 <notifications@github.com&gt;
发送时间: 2020年1月12日 00:44
收件人: coolsnowwolf/lede <lede@noreply.github.com&gt;
抄送: William Chan <root@williamchan.me&gt;, Author <author@noreply.github.com&gt;
主题: 回复：[coolsnowwolf/lede] luci-app-ssr-plus 彻底解决更新订阅慢的问题，顺便兼容SSD节点 (#2721)

adamw92, 2020-01-11:
> 方便把订阅链接发给我吗？
> 邮件 我帮你改一下
> 这个不是解析错误 是没支持 ss的我不知道什么格式 这看起来 有urlencode 和base64 不知是否所有ss都这样

这是ss协议sip002格式的链接，关于sip002格式的相关信息可以查看这里：https://shadowsocks.org/en/spec/SIP002-URI-Scheme.html

hzjnet, 2020-01-11:
# 太好了，谢谢大家

luckyyyyy, 2020-01-12:
> 这是ss协议sip002格式的链接，关于sip002格式的相关信息可以查看这里：https://shadowsocks.org/en/spec/SIP002-URI-Scheme.html

谢谢 这样就知道了 不是很麻烦 我下午回来改

denver9527, 2020-01-12:
我编译的这个订阅功能没有问题。就是隔一段时间 出现err /usr/bin/ssr-redir[22369]: server recv:
Connection reset peer，从定向链接被节点重置。那位大侠知道怎么回事吗？

EnnawYang, 2020-01-12:
昨天晚上用的好好的 今天早上起来就发现一大堆错误日志，前面的已经被覆盖了 看不到了，
Sun Jan 12 08:31:28 2020 user.notice shadowsocksr: ssr local error.restart!
Sun Jan 12 08:31:40 2020 user.notice shadowsocksr: ssr local error.restart!
Sun Jan 12 08:31:58 2020 user.notice shadowsocksr: ssr local error.restart!
Sun Jan 12 08:32:10 2020 user.notice shadowsocksr: ssr local error.restart!
Sun Jan 12 08:32:28 2020 user.notice shadowsocksr: ssr local error.restart!
Sun Jan 12 08:32:40 2020 user.notice shadowsocksr: ssr local error.restart!
......

luckyyyyy, 2020-01-12:
> 我编译的这个订阅功能没有问题。就是隔一段时间 出现err /usr/bin/ssr-redir[22369]: server recv: Connection reset peer，从定向链接被节点重置。那位大侠知道怎么回事吗？ 在 2020年1月12日星期日，hzjnet <notifications@github.com> 写道：
> […](#)

> 昨天晚上用的好好的 今天早上起来就发现一大堆错误日志，前面的已经被覆盖了 看不到了，
> Sun Jan 12 08:31:28 2020 user.notice shadowsocksr: ssr local error.restart!
> Sun Jan 12 08:31:40 2020 user.notice shadowsocksr: ssr local error.restart!
> Sun [...]

手动点击下更新订阅  如果依然这个错误 可以把订阅链接发给我看一下

luckyyyyy, 2020-01-12:
> 我编译的这个订阅功能没有问题。就是隔一段时间 出现err /usr/bin/ssr-redir[22369]: server recv: Connection reset peer，从定向链接被节点重置。那位大侠知道怎么回事吗？ 在 2020年1月12日星期日，hzjnet <notifications@github.com> 写道：
> […](#)

你这个 。。嗯 你懂的

caicaicai21, 2020-01-12:
> […](#)

被强制断开了，它会自动重新连接，可能是机场的问题，也有可能是GFW的问题，反正是正常的，不需要理会。

EnnawYang, 2020-01-12:
> > 我编译的这个订阅功能没有问题。就是隔一段时间 出现err /usr/bin/ssr-redir[22369]: server recv: Connection reset peer，从定向链接被节点重置。那位大侠知道怎么回事吗？ 在 2020年1月12日星期日，hzjnet [notifications@github.com](mailto:notifications@github.com) 写道：
> > […](#)
> 
> > 昨天晚上用的好好的 今天早上起来就发现一大堆错误日志，前面的已经被覆盖了 看不到了，
> > Sun Jan 12 08:31:28 2020 user.notice shadowsocksr: ssr local error.restart!
> > Sun Jan 12 08:31:40 2020 user.notice shadowsocksr: ssr local error.restart!
> > Sun Jan 12 08:31:58 2020 user.notice shadowsocksr: ssr local error.restart!
> > Sun Jan 12 08:32:10 2020 user.notice shadowsocksr: ssr local error.restart!
> > Sun Jan 12 08:32:28 2020 user.notice shadowsocksr: ssr local error.restart!
> > Sun Jan 12 08:32:40 2020 user.notice shadowsocksr: ssr local error.restart!
> > ......
> 
> 手动点击下更新订阅 如果依然这个错误 可以把订阅链接发给我看一下

问题找到了，不是订阅的锅，是我socks5代理设置的问题

luckyyyyy, 2020-01-12:
好的

发自我的iPhone

luckyyyyy, 2020-01-12:
> 今天测试了一下订阅，好像出现解析错误
>
>
> [image]
>
> 好像是解析节点有问题，测试节点：
>
> ss://YWVzLTEyOC1nY206dGVzdDEyMzQ1Njc4@test.server:1111#%E6%B5%8B%E8%AF%95%E6%9C%8D%E5%8A%A1

已经更新 看是否可以

luckyyyyy, 2020-01-12:
#2755

caicaicai21, 2020-01-12:
> 已经更新 看是否可以

[image]
不行，算了，我把订阅连接发给你吧，邮箱是root@williamchan.me？

luckyyyyy, 2020-01-12:
> > > 今天测试了一下订阅，好像出现解析错误
> > > > 2020-01-11 21:24:55 Shadowsocks/ShadowsocksR 1 Threads Started!
> > > > [C]: ?
> > > > /usr/share/shadowsocksr/subscribe.lua:284: in main chunk
> > > > [C]: in function 'xpcall'
> > > > /usr/share/shadowsocksr/subscribe.lua:206: in function </usr/share/shadowsocksr/subscribe.lua:160>
> > > > /usr/share/shadowsocksr/subscribe.lua:79: in function 'base64Decode'
> > > > /usr/share/shadowsocksr/subscribe.lua:293: in function </usr/share/shadowsocksr/subscribe.lua:284>
> > > > 2020-01-11 21:24:53 stack traceback:
> > > > 2020-01-11 21:24:53 更新失败服务正在恢复
> > > > 2020-01-11 21:24:53 发生错误, 正在恢复服务
> > > 
> > > 
> > > [image]
> > > 好像是解析节点有问题，测试节点：
> > > ss://YWVzLTEyOC1nY206dGVzdDEyMzQ1Njc4@test.server:1111#%E6%B5%8B%E8%AF%95%E6%9C%8D%E5%8A%A1
> > 
> > 
> > 已经更新 看是否可以
> 
> [image]
> 不行，算了，我把订阅连接发给你吧，邮箱是[root@williamchan.me](mailto:root@williamchan.me)？

对的 因为我测试你上面发的那两条是通了

caicaicai21, 2020-01-12:
> 对的 因为我测试你上面发的那两条是通了

我怀疑里面夹在了一些奇怪的东西，base64解码是用网页上的，可能少了一些。
邮件已经发送，注意查收。

luckyyyyy, 2020-01-12:
@caicaicai21 改是改好了 但是你提供给我的 格式很奇怪我不清楚哪一种才是正确的格式

因为的连接中 包含两种 我先提交上来 你试一下

caicaicai21, 2020-01-12:
> @caicaicai21 改是改好了 但是你提供给我的 格式很奇怪我不清楚哪一种才是正确的格式
> 
> 因为的连接中 包含两种 我先提交上来 你试一下

> @caicaicai21 改是改好了 但是你提供给我的 格式很奇怪我不清楚哪一种才是正确的格式
> 
> 因为的连接中 包含两种 我先提交上来 你试一下

[image]
测试通过。有两种格式的话，有一种应该就是其他的信息，例如：剩余流量xxx。

luckyyyyy, 2020-01-12:
我帮你测过时没问题的

ss的连接太奇怪了 password部分 有些base64有些没有
有些整个连接base64 有些部分字段base64 搞不明白

caicaicai21, 2020-01-12:
> 我帮你测过时没问题的
> 
> ss的连接太奇怪了 password部分 有些base64有些没有
> 有些整个连接base64 有些部分字段base64 搞不明白

每个机场都瞎搞呗，有些又不遵循规则。

adamw92, 2020-01-12:
> 我帮你测过时没问题的
> 
> ss的连接太奇怪了 password部分 有些base64有些没有
> 有些整个连接base64 有些部分字段base64 搞不明白

是这样的，ss的链接有两种格式。
一种是最开始就支持的，格式如下：

BASE64 encoded URI format configs:

ss://BASE64-ENCODED-STRING-WITHOUT-PADDING#TAG

也就是对method:password@ip:port部分整体进行一次base64，你可以看这个页面 https://shadowsocks.org/en/config/quick-guide.html

另一种就是新增的sip002格式，只对userinfo部分进行base64。

目前国内机场提供的订阅链接通常都是sip002格式的。
然而因为ss官方客户端都不支持sip002格式，于是机场们单个节点提供链接的时候就又是第一种格式了（目前sspanel的似乎都是这样）…

luckyyyyy, 2020-01-12:
我发现我进了个坑啊，之前shell处理的应该也没有那么全。

发自我的iPhone

luckyyyyy, 2020-01-12:
@adamw92  看样子SS是最麻烦的 其他的都还好 怪不得我以前用SS客户端的时候 有些很乱

adamw92, 2020-01-12:
> 我发现我进了个坑啊，之前shell处理的应该也没有那么全。 发自我的iPhone
> […](#)

之前的shell处理的是sip002格式的，机场们提供的订阅也是sip002。
ssr+插件单独添加节点信息的那里的导入节点链接处理的就是这个格式。
订阅支持sip002格式就好了，验证的时候也用sip002去验证。
需要sip002格式的订阅用于验证我可以发给你

luckyyyyy, 2020-01-12:
这个我已经支持了，两种都做了容错处理。

发自我的iPhone

9l, 2020-01-12:
用v2rayN客户端导出的服务器订阅内容不能被解析，报错:

 	2020-01-13 00:33:57 发生错误, 正在恢复服务

 	 	[C]: ?
 	 	/usr/share/shadowsocksr/subscribe.lua:284: in main chunk
 	 	[C]: in function 'xpcall'
 		/usr/share/shadowsocksr/subscribe.lua:206: in function 
 	 </usr/share/shadowsocksr/subscribe.lua:160>
 		/usr/share/shadowsocksr/subscribe.lua:79: in function 'base64Decode'
 		/usr/share/shadowsocksr/subscribe.lua:286: in function
 	</usr/share/shadowsocksr/subscribe.lua:284>
 	 2020-01-13 00:33:57 stack traceback:
 	 2020-01-13 00:33:57 /usr/share/shadowsocksr/subscribe.lua:79: attempt to index a nil value

 	 2020-01-13 00:33:57 (2.99 MB/s) - written to stdout [441/441]

 	      0K                                                       100% 2.99M=0s

 	 Saving to: 'STDOUT'
 	 Length: 441 [application/octet-stream]
 	 HTTP request sent, awaiting response... 200 OK

之前的shell版本可以识别的虽然也报有一个错误提示 `base64: invalid input`，但是至少能用

v2rayN客户端导出的服务器订阅内容是这样的:

`dm1lc3M6Ly9ldzBLSUNBaWRpSTZJQ0l5SWl3TkNpQWdJbkJ6SWpvZ0luUmxjM1FpTEEwS0lDQWlZV1JrSWpvZ0luZDNkeTVtY21WbE5qWTJOaTV1WlhRaUxBMEtJQ0FpY0c5eWRDSTZJQ0kwTkRNaUxBMEtJQ0FpYVdRaU9pQWlNVE00TW1WbFpHSXRaR0UzWlMwMFpqUm1MVGczWkRRdE1qbGhaREF3WlRneE9EZGlJaXdOQ2lBZ0ltRnBaQ0k2SUNJMk5DSXNEUW9nSUNKdVpYUWlPaUFpZDNNaUxBMEtJQ0FpZEhsd1pTSTZJQ0p1YjI1bElpd05DaUFnSW1odmMzUWlPaUFpSWl3TkNpQWdJbkJoZEdnaU9pQWlMekU0TkRJMU56TTVMeUlzRFFvZ0lDSjBiSE1pT2lBaWRHeHpJZzBLZlE9PQ0K`

luckyyyyy, 2020-01-12:
根据最新的pr修改看是否可以

发自我的iPhone

 2020-01-13 00:33:57 发生错误, 正在恢复服务  [C]: ?  /usr/share/shadowsocksr/subscribe.lua:284: in main chunk  [C]: in function 'xpcall' /usr/share/shadowsocksr/subscribe.lua:206: in function   </usr/share/shadowsocksr/subscribe.lua:160&gt; /usr/share/shadowsocksr/subscribe.lua:79: in function 'base64Decode' /usr/share/shadowsocksr/subscribe.lua:286: in function </usr/share/shadowsocksr/subscribe.lua:284&gt;  2020-01-13 00:33:57 stack traceback:  2020-01-13 00:33:57 /usr/share/shadowsocksr/subscribe.lua:79: attempt to index a nil value  2020-01-13 00:33:57 (2.99 MB/s) - written to stdout [441/441]       0K                                                       100% 2.99M=0s  Saving to: 'STDOUT'  Length: 441 [application/octet-stream]  HTTP request sent, awaiting response... 200 OK

9l, 2020-01-12:
> 根据最新的pr修改看是否可以 发自我的iPhone
> […](#)

试了一下你新pr的，可以解析正常，谢谢

hzjnet, 2020-01-13:
好东东 👍

greennyreborn, 2020-01-13:
楼主厉害，订阅更新非常快

SharerMax, 2020-01-13:
1. 有一种可能需要处理一下，将非标准的Base64中```_-```替换为```+/```后再进行encode
```因为编码后的+/=字符，标准的Base64并不适合直接放在URL里传输，有一些Base64的变种，它们将+/等符号转换为其他符号（如_-），这样就能安全的在URL中传输（Url Safe）了```
[摘自 https://1024tools.com/base64](https://1024tools.com/base64)
2. 节点的分割还有一种情况是按空格分隔，可能也需要处理一下，代码中好像只处理了按```\n```分割的节点信息

luckyyyyy, 2020-01-13:
第一个处理了 第二个不是很清楚现在

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: Max Zhao <notifications@github.com&gt;
发送时间: 2020年1月13日 17:01
收件人: coolsnowwolf/lede <lede@noreply.github.com&gt;
抄送: William Chan <root@williamchan.me&gt;, Author <author@noreply.github.com&gt;
主题: 回复：[coolsnowwolf/lede] luci-app-ssr-plus 彻底解决更新订阅慢的问题，顺便兼容SSD节点 (#2721)

SharerMax, 2020-01-13:
> 第一个处理了 第二个不是很清楚现在 发自我的iPhone
> […](#)
> ------------------ 原始邮件 ------------------ 发件人: Max Zhao <notifications@github.com&gt; 发送时间: 2020年1月13日 17:01 收件人: coolsnowwolf/lede <lede@noreply.github.com&gt; [...]

我的错，看到代码了，第一个处理了，
第二种应该是没处理的，我这触发了个错误
```
2020-01-13 16:20:29 发生错误, 正在恢复服务
	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:328: in main chunk
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:249: in function </usr/share/shadowsocksr/subscribe.lua:199>
	/usr/share/shadowsocksr/subscribe.lua:125: in function 'processData'
	/usr/share/shadowsocksr/subscribe.lua:29: in function 'split'
	/usr/share/shadowsocksr/subscribe.lua:330: in function </usr/share/shadowsocksr/subscribe.lua:328>
2020-01-13 16:20:29 stack traceback:
2020-01-13 16:20:29 /usr/share/shadowsocksr/subscribe.lua:29: attempt to index local 'full' (a nil value)
```

我手动修改了一下```subscribe.lua```中[这行代码](https://github.com/coolsnowwolf/lede/blob/d78dd0b948c54e70e21675a50bc481bb7619c0b6/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua#L235)为```node = split(base64Decode(raw, true):gsub(" ", "\n"), "\n")```后可以正确处理了

ps: 不会lua 现查的lua的文档，可能处理的不好

luckyyyyy, 2020-01-13:
没什么大问题 就这样改吧 你提pr

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: Max Zhao <notifications@github.com&gt;
发送时间: 2020年1月13日 17:45
收件人: coolsnowwolf/lede <lede@noreply.github.com&gt;
抄送: William Chan <root@williamchan.me&gt;, Author <author@noreply.github.com&gt;
主题: 回复：[coolsnowwolf/lede] luci-app-ssr-plus 彻底解决更新订阅慢的问题，顺便兼容SSD节点 (#2721)

RealKiro, 2020-01-14:
> 我手动修改了一下`subscribe.lua`中[这行代码](https://github.com/coolsnowwolf/lede/blob/d78dd0b948c54e70e21675a50bc481bb7619c0b6/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua#L235)为`node = split(base64Decode(raw, true):gsub(" ", "\n"), "\n")`后可以正确处理了

[image]
俺也一样，但是略有不同

SharerMax, 2020-01-14:
> > 我手动修改了一下`subscribe.lua`中[这行代码](https://github.com/coolsnowwolf/lede/blob/d78dd0b948c54e70e21675a50bc481bb7619c0b6/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua#L235)为`node = split(base64Decode(raw, true):gsub(" ", "\n"), "\n")`后可以正确处理了
> 
> [image]
> 俺也一样，但是略有不同，这样改还是不行

你的情况可能和我的不一样，看错误，你应该使用的是v2ray的订阅链接，其返回内容的json格式可能不对，你可以检查一下

luckyyyyy, 2020-01-14:
> 你的情况可能和我的不一样，看错误，你应该使用的是v2ray的订阅链接，其返回内容的json格式可能不对，你可以检查一下

v2ray 订阅 

> > 我手动修改了一下`subscribe.lua`中[这行代码](https://github.com/coolsnowwolf/lede/blob/d78dd0b948c54e70e21675a50bc481bb7619c0b6/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua#L235)为`node = split(base64Decode(raw, true):gsub(" ", "\n"), "\n")`后可以正确处理了
> 
> [image]
> 俺也一样，但是略有不同，这样改还是不行

v2ray 订阅 json解析失败了 又是什么独特的格式吗？

SimonsQiu, 2020-01-14:
可以加入对obfs插件的支援么？

RealKiro, 2020-01-14:
> v2ray 订阅 json解析失败了 又是什么独特的格式吗？

不知道，[#2748 ](https://github.com/coolsnowwolf/lede/pull/2748)

SharerMax, 2020-01-14:
> > v2ray 订阅 json解析失败了 又是什么独特的格式吗？
> 
> 不知道,报错虽稍微不同但仍然是这个Bug[#2748 ](https://github.com/coolsnowwolf/lede/pull/2748)

擦，两个换行，我好像发现问题了

luckyyyyy, 2020-01-14:
各位 剩下的靠你们解决一下 我写好了 trim 函数的 我最近会有些忙

SharerMax, 2020-01-14:
> > v2ray 订阅 json解析失败了 又是什么独特的格式吗？
> 
> 不知道,报错虽稍微不同但仍然是这个Bug[#2748 ](https://github.com/coolsnowwolf/lede/pull/2748)
已处理
PR #2770

luckyyyyy, 2020-01-14:
节点为空 应该和trim还有\0有关 lua的行为和c是一致的 可以从这个方面排查下

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: Max Zhao <notifications@github.com&gt;
发送时间: 2020年1月15日 00:16
收件人: coolsnowwolf/lede <lede@noreply.github.com&gt;
抄送: William Chan <root@williamchan.me&gt;, Author <author@noreply.github.com&gt;
主题: 回复：[coolsnowwolf/lede] luci-app-ssr-plus 彻底解决更新订阅慢的问题，顺便兼容SSD节点 (#2721)

SharerMax, 2020-01-15:
v2ray的订阅格式有两个版本了，之前的version 1已经找不到说明了，只有version 2版本的（[分享链接格式说明(ver-2)](https://github.com/2dust/v2rayN/wiki/%E5%88%86%E4%BA%AB%E9%93%BE%E6%8E%A5%E6%A0%BC%E5%BC%8F%E8%AF%B4%E6%98%8E(ver-2))）
是否还要继续支持version1的订阅格式

luckyyyyy, 2020-01-15:
> v2ray的订阅格式有两个版本了，之前的version 1已经找不到说明了，只有version 2版本的（[分享链接格式说明(ver-2)](https://github.com/2dust/v2rayN/wiki/%E5%88%86%E4%BA%AB%E9%93%BE%E6%8E%A5%E6%A0%BC%E5%BC%8F%E8%AF%B4%E6%98%8E(ver-2))）
> 是否还要继续支持version1的订阅格式

哈哈 太复杂了 v2ray 我看了之前的 shell 脚本支持的也不全 我看PR有人改了一点

coolsnowwolf, 2020-01-15:
所以为啥机场用 SSR 这么多明白了吧？ 因为 SSR 的标准很明确

RealKiro, 2020-01-16:
> v2ray的订阅格式有两个版本了，之前的version 1已经找不到说明了，只有version 2版本的（[分享链接格式说明(ver-2)]

我找到了这个，不知有没有帮助 [添加v2ray订阅ver1格式的支持](https://gitlab.com/blademainer/fancyss/commit/dfd5fa40929020e36a24a222f729d3ad451d344f)

adamw92, 2020-01-16:
@SimonsQiu ssr+本身不支持obfs插件，订阅支持了有啥用？

MxCen, 2020-01-17:
> @SimonsQiu ssr+本身不支持obfs插件，订阅支持了有啥用？

是的，SSR不支持呀。订阅支持没有任何的意义，SS才支持。

closed 2020-01-22 by coolsnowwolf