coolsnowwolf/lede #2728: 请问可以添加NVME的驱动吗？

michael6555, 2020-01-09:


sqliuchang, 2020-01-09:
有人刷op还用nvme硬盘的吗…是真土豪啊

michael6555, 2020-01-10:
用16g的傲腾，三十块算土豪吗？

发自我的iPhone

sqliuchang, 2020-01-10:
> 用16g的傲腾，三十块算土豪吗？ 发自我的iPhone

可以，很机智，原来傲腾还可以这么用！
但是看起来默认是选中了nvme驱动的啊
https://github.com/coolsnowwolf/lede/blob/4730b1260e3f3f72ca7771e0744c1ef50e4d1bb7/target/linux/x86/config-4.19#L84

michael6555, 2020-01-10:
> > 用16g的傲腾，三十块算土豪吗？ 发自我的iPhone
> 
> 可以，很机智，原来傲腾还可以这么用！
> 但是看起来默认是选中了nvme驱动的啊
> https://github.com/coolsnowwolf/lede/blob/4730b1260e3f3f72ca7771e0744c1ef50e4d1bb7/target/linux/x86/config-4.19#L84

默认有开的话，开机找不到系统

jackiezhang29, 2020-01-10:
> 有人刷op还用nvme硬盘的吗…是真土豪啊

我貌似就是950 pro

michael6555, 2020-01-10:
> > 有人刷op还用nvme硬盘的吗…是真土豪啊
> 
> 我貌似就是950 pro

你的用950pro能开机吗？

jackiezhang29, 2020-01-10:
> > > 有人刷op还用nvme硬盘的吗…是真土豪啊
> > 
> > 
> > 我貌似就是950 pro
> 
> 你的用950pro能开机吗？

我是win2019+hyperv

coolsnowwolf, 2020-01-11:
EFI 的 IMG 才能支持 NVME

michael6555, 2020-01-12:
> EFI 的 IMG 才能支持 NVME

怎样才能编译成EFI的？

MxCen, 2020-01-12:
Target Images  --->Build EFI GRUB images (Linux x86 or x86_64 host only)

michael6555, 2020-01-14:
> Target Images --->Build EFI GRUB images (Linux x86 or x86_64 host only)
编译x86_64版本CONFIG_EFI_IMAGES=y是默认的，能说简单明了一点吗?

MxCen, 2020-01-14:
make menuconfig 里面选起Target Images --->Build EFI GRUB images (Linux x86 or x86_64 host only)，编译出来的固件包efi的就是。

closed 2020-02-19 by coolsnowwolf