coolsnowwolf/lede #2731: 偶尔上网巨慢无响应

zhangdp, 2020-01-09:
排查了一圈似乎跟udp连接数有关系，pt下载一多时打开网页啥的都巨慢或者连不上，udp连接数有4k+，导致dns查询不了了？
Thu Jan  9 23:47:51 2020 daemon.warn pdnsd[6348]: Caught signal 15. Exiting.
Thu Jan  9 23:47:51 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jan  9 23:47:51 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jan  9 23:47:54 2020 cron.info crond[30208]: crond (busybox 1.30.0) started, log level 8

dolphinpaopao, 2020-01-09:
这个不是固件问题，首先检查是不是运营商限制了连接数和端口数，测试2W连接是没有问题的，另外这种问题首先去搜索、论坛或者QQ群等渠道获取答案，不要随意开issue，作者已经强调过多次，请理解issue的含义。谢谢，感谢你的谨慎对此做出的贡献。

zhangdp, 2020-01-09:
感谢你的回复。我已经找这个问题好久了，已经测试了无数遍了，100%确定是固件的问题或者说哪里没设置好可能。我是当2级路由使用的，当这个固件的路由下的设备无法上网的时候，1级路由下的设备是正常的。而且我换了其他固件，下载同样的东西确实不会断网。真心是很喜欢这个固件，但我自己水平有限，也搜索过好多，实在不知道啥原因才开issue的。很早以前有开过一次issue，说要把FullCone-NAT关闭，有改善但问题依旧。

Her0R0cket, 2020-01-10:
你这个是系统日志,还有一个是内核日志,你需要再看一下内核日志

我之前有段时间网络有问题,也是会断网,内核日志老是报  "ip_conntrack: table full, dropping packet",是nf_conntrack连接数到达了限制.搜索下得到了解决方案
降低 ip_conntrack_tcp_timeout_established 时间, 增加nf_conntrack_max最大连接数.
最后在启动项里添加如下两行解决.

echo 1200 > /proc/sys/net/ipv4/netfilter/ip_conntrack_tcp_timeout_established
echo 100000 > /proc/sys/net/netfilter/nf_conntrack_max

在启动项里需要重启生效,不过可以手动在命令行里运行这两条命令试下.

CYaNu, 2020-01-10:
> 排查了一圈似乎跟udp连接数有关系，pt下载一多时打开网页啥的都巨慢或者连不上，udp连接数有4k+，导致dns查询不了了？
> Thu Jan 9 23:47:51 2020 daemon.warn pdnsd[6348]: Caught signal 15. Exiting.
> Thu Jan 9 23:47:51 2020 user.notice dnsmasq: DNS rebinding [...]

可以说你的问题和这段log没有关系，如果想解决请放上完整点的log。

caicaicai21, 2020-01-10:
固件里面有个Turbo ACC 网络加速，有可能是这个驱动的原因，可以尝试 开/关 测试一下。

dolphinpaopao, 2020-01-10:
固件默认编译默认设置，再次测试2W连接没有问题，虽然此时网页打开速度受到影响变慢。所以这个问题得自己排查，因为每个人得使用环境差异太大。首先恢复默认设置测试问题是否重现再逐一排查。怀疑是连接数和端口方面的原因。

coolsnowwolf, 2020-01-10:
最好有 netstats 的统计，话说 Fulcone 的 UDP 连接表性能问题确实存在，但家用不太会遇到

CYaNu, 2020-01-11:
> 感谢你的回复。我已经找这个问题好久了，已经测试了无数遍了，100%确定是固件的问题或者说哪里没设置好可能。我是当2级路由使用的，当这个固件的路由下的设备无法上网的时候，1级路由下的设备是正常的。而且我换了其他固件，下载同样的东西确实不会断网。真心是很喜欢这个固件，但我自己水平有限，也搜索过好多，实在不知道啥原因才开issue的。很早以前有开过一次issue，说要把FullCone-NAT关闭，有改善但问题依旧。

我的网络环境和你的类似ROS + LEDE，为了避免问题 我会关掉 lede的FULLCONE NAT加速 DNS加速，并且关掉防火墙，做成透明网关，所以NAT操作都交由ROS处理。LEDE只做DNS服务器和科学工具。

zhangdp, 2020-01-12:
瞎几把折腾了下，现在似乎不会断网了，就是下载多的时候dns解析可能很慢或者解析不了。回忆了下，这两天做的有：全新编译最新代码全新安装，关闭FULLCONE，关闭Turbo ACC加速；Esxi里网络->默认TCP/IP堆栈->连接数改为最大值20000，虽然不知道这个连接数指的是啥。
[image]

closed 2020-01-18 by coolsnowwolf