coolsnowwolf/lede #2734: 选择了VPN-Softether编译后 然而功能模块却不出现在Web界面里

TerryWang159, 2020-01-10:
1. make menuconfig  and then  selected  Network-->VPN->Softethervpn base | softethervpn-bridge |softethervpn-client|softethervpn-server
2. Save config file and run command  " make -j1 V=s"
3. make success ,But find nothing about softethervpn in the main menu.
Please try above, Mybe my local-mistake. thanks the project .

xinxijishuwyq, 2020-01-10:
这个本来就没有web界面吧。。。有界面的是luci-app-softethervpn

TerryWang159, 2020-01-10:
我看了一下在NETWork-->VPN-->选择的PPTPVPN 会出现在界面主菜单VPN里,所以按照同样的逻辑, 选择的softether 也应该出现在主菜单吧, 如果界面配置入口都没有,那怎么使用呢?

TerryWang159, 2020-01-10:
> 
> 
> 这个本来就没有web界面吧。。。有界面的是luci-app-softethervpn

不一定在luci选的东西才会出现在界面吧 ,在其他地方选择的也会出现的.  
我看了一下在NETWork-->VPN-->选择的PPTPVPN 会出现在界面主菜单VPN里,所以按照同样的逻辑, 选择的softether 也应该出现在主菜单吧, 如果界面配置入口都没有,那怎么使用呢?

Aes64X, 2020-01-10:
关了吧，issues不是讨论这种问题的，测试过功能正常，自己找原因。

TerryWang159, 2020-01-10:
> 
> 
> 关了吧，issues不是讨论这种问题的，测试过功能正常，自己找原因。

我是昨晚git最新版本的代码编译的 ,你测试的是最新的源么 ? 现在我正在重新编译,待完成后再看看

caicaicai21, 2020-01-10:
pptp有web界面，因为固件默认勾选了了luci-app-pptp-server，如果你要softether有web界面，那么请你把luci-app-softethervpn这个勾上，它自动会加载相关依赖。

TerryWang159, 2020-01-10:
> 
> 
> pptp有web界面，因为固件默认勾选了luci-app-pptp-server，如果你要softether有web界面，那么请你把luci-app-softethervpn这个勾上，它自动会加载相关依赖。另外单独勾选softether并没有web界面，因为它并不需要luci-app-softethervpn依赖。

Got it!  谢谢大神

closed 2020-01-10 by TerryWang159