coolsnowwolf/lede #2735: R7800云编译固件5G偶尔掉线

pk49800, 2020-01-10:
问题描述：R7800云编译固件5G 偶尔掉线，掉线后需要启停一次无线的5G开关才能搜到5G信号，2.4G频段正常

主机型号 | Netgear Nighthawk X4S R7800

固件版本 | OpenWrt R9.12.31 / LuCI Master (git-19.338.43082-9e87e69)

内核版本 | 4.14.160

系统日志如下：
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27827]: listening at 0.0.0.0:1234
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27827]: running from root user
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27829]: initializing ciphers... aes-256-ctr
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27829]: tcp port reuse enabled
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27829]: listening at 0.0.0.0:1234
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27829]: running from root user
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27869]: initializing ciphers... aes-256-ctr
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27869]: UDP relay enabled
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27869]: udp port reuse enabled
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27869]: TCP relay disabled
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27869]: listening at 0.0.0.0:1234
Thu Jan  9 20:33:12 2020 kern.info /usr/bin/ssr-redir[27869]: running from root user
Thu Jan  9 20:33:12 2020 daemon.info pdnsd[27885]: pdnsd-1.2.9b-par starting.
Thu Jan  9 20:33:13 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jan  9 20:33:13 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jan  9 20:33:26 2020 kern.err /usr/bin/ssr-redir[27829]: server recv: Connection reset by peer
Thu Jan  9 20:33:27 2020 kern.err /usr/bin/ssr-redir[27829]: server recv: Connection reset by peer
Thu Jan  9 20:34:20 2020 daemon.info hostapd: wlan1: STA a0:57:e3:4d:2b:8d IEEE 802.11: authenticated
Thu Jan  9 20:34:20 2020 daemon.info hostapd: wlan1: STA a0:57:e3:4d:2b:8d IEEE 802.11: associated (aid 5)
Thu Jan  9 20:34:20 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED a0:57:e3:4d:2b:8d
Thu Jan  9 20:34:20 2020 daemon.info hostapd: wlan1: STA a0:57:e3:4d:2b:8d RADIUS: starting accounting session 8D46D7B577BF93FD
Thu Jan  9 20:34:20 2020 daemon.info hostapd: wlan1: STA a0:57:e3:4d:2b:8d WPA: pairwise key handshake completed (RSN)
Thu Jan  9 20:34:21 2020 kern.err /usr/bin/ssr-redir[27829]: server recv: Connection reset by peer
Thu Jan  9 20:35:01 2020 daemon.info hostapd: wlan1: STA a0:57:e3:4d:2b:8d IEEE 802.11: disconnected due to excessive missing ACKs
Thu Jan  9 20:35:01 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED a0:57:e3:4d:2b:8d
Thu Jan  9 20:35:31 2020 daemon.info hostapd: wlan1: STA a0:57:e3:4d:2b:8d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Jan  9 20:36:13 2020 daemon.info hostapd: wlan0: STA a0:57:e3:4d:2b:8d IEEE 802.11: authenticated
Thu Jan  9 20:36:13 2020 daemon.info hostapd: wlan0: STA a0:57:e3:4d:2b:8d IEEE 802.11: associated (aid 3)
Thu Jan  9 20:36:13 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED a0:57:e3:4d:2b:8d
Thu Jan  9 20:36:13 2020 daemon.info hostapd: wlan0: STA a0:57:e3:4d:2b:8d RADIUS: starting accounting session 90CB277BB3B2EE2D
Thu Jan  9 20:36:13 2020 daemon.info hostapd: wlan0: STA a0:57:e3:4d:2b:8d WPA: pairwise key handshake completed (RSN)
Thu Jan  9 20:36:15 2020 kern.err /usr/bin/ssr-redir[27829]: server recv: Connection reset by peer
Thu Jan  9 20:44:20 2020 kern.err /usr/bin/ssr-redir[27827]: server recv: Connection reset by peer
Thu Jan  9 20:48:43 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED c4:86:e9:41:11:74
Thu Jan  9 20:49:25 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 IEEE 802.11: authenticated
Thu Jan  9 20:49:25 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 IEEE 802.11: associated (aid 1)
Thu Jan  9 20:49:25 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED c4:86:e9:41:11:74
Thu Jan  9 20:49:25 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 RADIUS: starting accounting session D8ACDF0982CCDE1C
Thu Jan  9 20:49:25 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 WPA: pairwise key handshake completed (RSN)
Thu Jan  9 20:55:27 2020 kern.err /usr/bin/ssr-redir[27829]: server recv: Connection reset by peer
Thu Jan  9 21:03:34 2020 kern.err /usr/bin/ssr-redir[27829]: server recv: Connection reset by peer
Thu Jan  9 21:04:06 2020 kern.err /usr/bin/ssr-redir[27829]: server recv: Connection reset by peer
Thu Jan  9 21:04:32 2020 kern.err /usr/bin/ssr-redir[27829]: server recv: Connection reset by peer
Thu Jan  9 21:05:07 2020 kern.err /usr/bin/ssr-redir[27827]: server recv: Connection reset by peer
Thu Jan  9 21:11:42 2020 kern.err /usr/bin/ssr-redir[27827]: server recv: Connection reset by peer
Thu Jan  9 21:11:51 2020 kern.err /usr/bin/ssr-redir[27827]: server recv: Connection reset by peer
Thu Jan  9 21:11:51 2020 kern.err /usr/bin/ssr-redir[27827]: server recv: Connection reset by peer
Thu Jan  9 21:11:51 2020 kern.err /usr/bin/ssr-redir[27829]: server recv: Connection reset by peer
Thu Jan  9 21:17:42 2020 kern.err /usr/bin/ssr-redir[27827]: server recv: Connection reset by peer
Thu Jan  9 21:22:25 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED c4:86:e9:41:11:74
Thu Jan  9 21:23:02 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 IEEE 802.11: authenticated
Thu Jan  9 21:23:02 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 IEEE 802.11: associated (aid 1)
Thu Jan  9 21:23:02 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED c4:86:e9:41:11:74
Thu Jan  9 21:23:02 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 RADIUS: starting accounting session 2AA7F411EF7E362E
Thu Jan  9 21:23:02 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 WPA: pairwise key handshake completed (RSN)
Thu Jan  9 21:23:04 2020 kern.err /usr/bin/ssr-redir[27829]: server recv: Connection reset by peer
Thu Jan  9 21:25:39 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED c4:86:e9:41:11:74
Thu Jan  9 21:25:39 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 IEEE 802.11: authenticated
Thu Jan  9 21:25:39 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 IEEE 802.11: associated (aid 1)
Thu Jan  9 21:25:39 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED c4:86:e9:41:11:74
Thu Jan  9 21:25:39 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 RADIUS: starting accounting session 2AA7F411EF7E362E
[... 641 lines cut ...]
Fri Jan 10 10:14:08 2020 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 74:c1:4f:a7:2a:ef
Fri Jan 10 10:15:28 2020 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 94:65:2d:8f:f3:57
Fri Jan 10 10:16:14 2020 daemon.info hostapd: wlan1: STA 38:00:25:aa:91:a5 IEEE 802.11: authenticated
Fri Jan 10 10:16:14 2020 daemon.info hostapd: wlan1: STA 38:00:25:aa:91:a5 IEEE 802.11: associated (aid 2)
Fri Jan 10 10:16:14 2020 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED 38:00:25:aa:91:a5 off
Fri Jan 10 10:16:14 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 38:00:25:aa:91:a5
Fri Jan 10 10:16:14 2020 daemon.info hostapd: wlan1: STA 38:00:25:aa:91:a5 RADIUS: starting accounting session 6D02DF1A32883F45
Fri Jan 10 10:16:14 2020 daemon.info hostapd: wlan1: STA 38:00:25:aa:91:a5 WPA: pairwise key handshake completed (RSN)
Fri Jan 10 10:16:16 2020 kern.err /usr/bin/ssr-redir[27247]: server recv: Connection reset by peer
Fri Jan 10 10:16:16 2020 kern.err /usr/bin/ssr-redir[27249]: server recv: Connection reset by peer
Fri Jan 10 10:16:16 2020 kern.err /usr/bin/ssr-redir[27247]: server recv: Connection reset by peer
Fri Jan 10 10:16:17 2020 kern.err /usr/bin/ssr-redir[27249]: server recv: Connection reset by peer
Fri Jan 10 10:16:17 2020 kern.err /usr/bin/ssr-redir[27247]: server recv: Connection reset by peer
Fri Jan 10 10:16:17 2020 kern.err /usr/bin/ssr-redir[27247]: server recv: Connection reset by peer
Fri Jan 10 10:16:17 2020 kern.err /usr/bin/ssr-redir[27247]: server recv: Connection reset by peer
Fri Jan 10 10:16:17 2020 kern.err /usr/bin/ssr-redir[27247]: server recv: Connection reset by peer
Fri Jan 10 10:16:19 2020 kern.err /usr/bin/ssr-redir[27249]: server recv: Connection reset by peer
Fri Jan 10 10:16:19 2020 kern.err /usr/bin/ssr-redir[27247]: server recv: Connection reset by peer
Fri Jan 10 10:16:19 2020 kern.err /usr/bin/ssr-redir[27249]: server recv: Connection reset by peer
Fri Jan 10 10:16:19 2020 kern.err /usr/bin/ssr-redir[27247]: server recv: Connection reset by peer
Fri Jan 10 10:16:19 2020 kern.err /usr/bin/ssr-redir[27247]: server recv: Connection reset by peer
Fri Jan 10 10:16:44 2020 kern.err /usr/bin/ssr-redir[27247]: server recv: Connection reset by peer
Fri Jan 10 10:17:03 2020 kern.err /usr/bin/ssr-redir[27249]: server recv: Connection reset by peer
Fri Jan 10 10:17:19 2020 kern.err /usr/bin/ssr-redir[27249]: server recv: Connection reset by peer
Fri Jan 10 10:19:15 2020 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 74:c1:4f:a7:2a:ef

mygit5678, 2020-01-10:
信道改为44试试

coolsnowwolf, 2020-01-10:
信道固定 36， AES 加密方式

pk49800, 2020-01-11:
> 信道固定 36， AES 加密方式

改了也还是这样，不知道有没有设置正确，下面是设置截图
[image]

xieshiyudada, 2020-01-11:
我之前也遇到过，这样的问题，我的解决方案是，重置，wifi频段什么的不要动，设置个wifi密码。

pk49800, 2020-01-12:
> 信道固定 36， AES 加密方式

昨天信道改成固定36，频宽改成80MHz（以前是160MHz）就没有出现掉线的问题了。
[image]

下面是昨晚到今天的系统日志。
Sat Jan 11 22:28:07 2020 kern.err /usr/bin/ssr-redir[17824]: send: Connection reset by peer
Sat Jan 11 22:31:06 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:31:09 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:31:09 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:31:09 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 22:31:09 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 22:34:46 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 22:35:55 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 IEEE 802.11: authenticated
Sat Jan 11 22:35:55 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 IEEE 802.11: associated (aid 5)
Sat Jan 11 22:35:55 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED c4:86:e9:41:11:74
Sat Jan 11 22:35:55 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 RADIUS: starting accounting session 6019D8F126684A4E
Sat Jan 11 22:35:55 2020 daemon.info hostapd: wlan1: STA c4:86:e9:41:11:74 WPA: pairwise key handshake completed (RSN)
Sat Jan 11 22:35:56 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:36:42 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:36:45 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 22:36:50 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 22:37:37 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 22:37:39 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:38:59 2020 kern.err /usr/bin/ssr-redir[17824]: send: Connection reset by peer
Sat Jan 11 22:39:12 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:39:14 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:39:24 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 22:39:28 2020 kern.err /usr/bin/ssr-redir[17824]: send: Connection reset by peer
Sat Jan 11 22:39:35 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:39:39 2020 kern.err /usr/bin/ssr-redir[17826]: send: Connection reset by peer
Sat Jan 11 22:42:30 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:42:38 2020 kern.err /usr/bin/ssr-redir[17826]: send: Connection reset by peer
Sat Jan 11 22:42:42 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:42:45 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:42:59 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:43:04 2020 kern.err /usr/bin/ssr-redir[17826]: send: Connection reset by peer
Sat Jan 11 22:43:05 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:43:08 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 22:43:49 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:45:00 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:50:21 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:57:24 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 22:58:12 2020 daemon.info hostapd: wlan1: STA 20:0d:b0:14:e7:a5 IEEE 802.11: disconnected due to excessive missing ACKs
Sat Jan 11 22:58:12 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 20:0d:b0:14:e7:a5
Sat Jan 11 22:58:42 2020 daemon.info hostapd: wlan1: STA 20:0d:b0:14:e7:a5 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Jan 11 23:01:29 2020 kern.err /usr/bin/ssr-redir[17824]: send: Connection reset by peer
Sat Jan 11 23:07:30 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 23:07:30 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 23:07:30 2020 kern.err /usr/bin/ssr-redir[17824]: server recv: Connection reset by peer
Sat Jan 11 23:12:36 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 23:13:12 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 23:15:00 2020 kern.err /usr/bin/ssr-redir[17826]: server recv: Connection reset by peer
Sat Jan 11 23:19:48 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 4c:57:ca:d1:6a:8b
Sat Jan 11 23:19:48 2020 daemon.info hostapd: wlan1: STA 4c:57:ca:d1:6a:8b IEEE 802.11: authenticated
Sat Jan 11 23:19:48 2020 daemon.info hostapd: wlan1: STA 4c:57:ca:d1:6a:8b IEEE 802.11: associated (aid 2)
Sat Jan 11 23:19:48 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 4c:57:ca:d1:6a:8b
Sat Jan 11 23:19:48 2020 daemon.info hostapd: wlan1: STA 4c:57:ca:d1:6a:8b RADIUS: starting accounting session 026B62E6A04CF33A
Sat Jan 11 23:19:48 2020 daemon.info hostapd: wlan1: STA 4c:57:ca:d1:6a:8b WPA: pairwise key handshake completed (RSN)
[... cut ...]
Sun Jan 12 10:27:40 2020 kern.err /usr/bin/ssr-redir[12313]: server recv: Connection reset by peer
Sun Jan 12 10:27:40 2020 kern.err /usr/bin/ssr-redir[12313]: server recv: Connection reset by peer
Sun Jan 12 10:27:40 2020 kern.err /usr/bin/ssr-redir[12313]: server recv: Connection reset by peer
Sun Jan 12 10:27:40 2020 kern.err /usr/bin/ssr-redir[12313]: server recv: Connection reset by peer
Sun Jan 12 10:27:40 2020 kern.err /usr/bin/ssr-redir[12313]: server recv: Connection reset by peer
Sun Jan 12 10:27:40 2020 kern.err /usr/bin/ssr-redir[12313]: server recv: Connection reset by peer

closed 2020-01-18 by coolsnowwolf