coolsnowwolf/lede #2739: kmod-fs-cifs 与openvpn ssl安装不上去，有没有openvpn服务器的配置教程

lone1985, 2020-01-10:
如题，网络硬盘mount不了，安装kmod-fs-cifs 又安装不上去

root@OpenWrt:~# mount -t cifs -o username="",password="" //192.168.4.250/d969 /mnt/sdb/nas
mount: /mnt/sdb/nas: unknown filesystem type 'cifs'.

mount -t nfs
mount -t smbfs

提示都是一样的错误

然后网站找了很久了没解决
https://www.right.com.cn/forum/thread-315798-1-1.html
这个帖子也一样'
CIFS在OpenWrt安装起来很简单：

opkg update
opkg install kmod-fs-cifs
opkg install kmod-nls-utf8 kmod-nls-base kmod-crypto-hmac kmod-crypto-md5 kmod-crypto-misc cifsmount
简单使用的示例：

mount -t cifs //cifs-server/share /localfolder -o user=username,password=password
匿名访问：

mount -t cifs //cifs-server/share /localfolder -o guest

我安装的时候提示这个
root@OpenWrt:~# opkg install kmod-fs-cifs
Installing kmod-fs-cifs (4.19.93-1) to root...
Downloading http://downloads.openwrt.org/sna ... .19.93-1_x86_64.ipk
Collected errors:
* satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-cifs:
*      kernel (= 4.19.93-1-0b3dff97a9c6f0730b67aeb33483aec1) *         kernel (= 4.19.93-1-0b3dff97a9c6f0730b67aeb33483aec1) *
* opkg_install_cmd: Cannot install package kmod-fs-cifs.
————————————————————————————————————————

openVPN 的服务器 配置到生成证书 秘钥就错误了

1、安装搭建OpenVPN服务器所需要的软件
登录服务器，更新软件源并安装必须的软件包：

ssh root@192.168.1.1
opkg update
opkg install openvpn-openssl openvpn-easy-rsa luci-i18n-openvpn-zh-cn
————————————————————————
**### 到这里提示 openvpn-openssl 错误。**

2、使用easy-rsa生成所需的证书文件

cd /etc/easy-rsa
先备份一下参数文件：

cp vars vars-bak
按需修改vars参数：

vi vars
修改第45、65、76、84-89、97、105、113、117、134、139、171、180、192行，大部分参数无需修改，只是去除#号使其生效：

set_var EASYRSA                 "$PWD"
set_var EASYRSA_PKI             "$EASYRSA/pki"
set_var EASYRSA_DN              "cn_only"
set_var EASYRSA_REQ_COUNTRY     "CN"
set_var EASYRSA_REQ_PROVINCE    "Guangdong"
set_var EASYRSA_REQ_CITY        "Guangzhou"
set_var EASYRSA_REQ_ORG         "OpenVPN Certificate Authority"
set_var EASYRSA_REQ_EMAIL       "test@test.com"
set_var EASYRSA_REQ_OU          "OpenVPN EASY CA"
set_var EASYRSA_KEY_SIZE        4096
set_var EASYRSA_ALGO            rsa
set_var EASYRSA_CA_EXPIRE       3650
set_var EASYRSA_CERT_EXPIRE     3650
set_var EASYRSA_NS_SUPPORT      "no"
set_var EASYRSA_NS_COMMENT      "OpenVPN Certificate Authority"
set_var EASYRSA_EXT_DIR         "$EASYRSA/x509-types"
set_var EASYRSA_SSL_CONF        "$EASYRSA/openssl-1.0.cnf"
set_var EASYRSA_DIGEST          "sha256"
现在可以开始生成证书了，注意：
1）、在build-ca指令之后，加了nopass参数，以避免每次使用证书都需要输入口令。如需更强的安全性，可不加该参数，后面生成服务端证书的时候也使用了nopass参数。
2）、在脚本运行过程中可能会暂停，等待确认，通常方括号[]内的预设内容如不需要改可直接回车，有些提示输入’yes’的则需照输，否则脚本会退出。

cd /etc/easy-rsa/
easyrsa init-pki
easyrsa build-ca nopass

这里就生成不了了

Her0R0cket, 2020-01-10:
如果不是官方正式版的系统的话基本只能靠自己编译集成,openvpn也需要自己集成,luci基础环境已经和官方不兼容了

AmadeusGhost, 2020-01-10:
kmod这些需要自己编译的时候选进去，官方源的会冲突。

lone1985, 2020-01-11:
可是我不会编译啊，有没有简易的编译教程？，我现在刷的是这个内核【2020-01-08更新】4.19.91内核 魔改版+原版不可描述双版齐发，更有丰富插件闪耀加持https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=1234564&page=1&authorid=318678。linux基础也会一点 求指导

Her0R0cket, 2020-01-11:
http://repo.elecity.top:8880/openwrt/
自己找找吧,里面应该都有,内核安装的话要加--force-depends,内核版本差不多理论上是可以用的

AmadeusGhost, 2020-01-11:
> 可是我不会编译啊，有没有简易的编译教程？，我现在刷的是这个内核【2020-01-08更新】4.19.91内核 [...]

编译的话网上教程一堆啊，百度/谷歌找下就有了。

lone1985, 2020-01-11:
> 编译的话网上教程一堆啊，百度/谷歌找下就有了。

-----------------------------------------
按照这个操作对吗？手把手教你用OpenWRT单独编译ipk插件
https://www.right.com.cn/forum/thread-598816-1-1.html
先把原来的IPK下载下来

lone1985, 2020-01-11:
> http://repo.elecity.top:8880/openwrt/
> 自己找找吧,里面应该都有,内核安装的话要加--force-depends,内核版本差不多理论上是可以用的

谢谢我安装成功了，不过现在提示
mount: /mnt/sdb/nas: special device //192.168.4.250/d969 does not exist.
我明明两个分区是在的，怎么提示找不到？

AmadeusGhost, 2020-01-11:
> 按照这个操作对吗？手把手教你用OpenWRT单独编译ipk插件
> https://www.right.com.cn/forum/thread-598816-1-1.html
> 先把原来的IPK下载下来

我指的是编译整个固件而不是ipk，自己想要什么就选上什么。

lone1985, 2020-01-11:
> http://repo.elecity.top:8880/openwrt/
> 自己找找吧,里面应该都有,内核安装的话要加--force-depends,内核版本差不多理论上是可以用的

已经搞定了samba4对旧版共享协议smb1已经不支持，需要加上vers=1.0，
mount -t cifs -o username="",password="",file_mode=0777,dir_mode=0777,vers=1.0 //192.168.4.250/d969 /mnt/sdb/nas
谢谢了

closed 2020-02-19 by coolsnowwolf