coolsnowwolf/lede #2742: K2T编译出来的固件有问题。。

NickLiu635, 2020-01-10:
K2T编译有问题
在ar7xxx/9xxx和ath79里面都能找到K2T这个设备，
当选择ar7xxx/9xxx编译时，刷到K2T后，无线信号正常，但系统不稳定（我这儿平均一天崩两次，每次崩的现象就是无法通过IP访问设备，SSID正常发射但其它设备无法获取IP，看起来就像是外设正常工作但CPU的网络停止工作，且未触发看门狗）
当选择ath79编译刷到K2T后，K2T的无线中不显示5G设备

AmadeusGhost, 2020-01-10:
ath79有问题是因为package/base-files少了caldata.sh等文件，可以试试我的pr #2690

coolsnowwolf, 2020-01-10:
ATH10K 占用内存过大，偏偏 K2T 才 64MB

Aes64X, 2020-01-11:
你可以尝试 ath10k-ct-smallbuffers ，说明是给小内存设备使用，我还在测试。。。。

NickLiu635, 2020-01-13:
目前用ar7xxx的，去掉一些不必要的组件（如流控，adbyby）后编译出来的K2t，实测12h没重启了（昨晚更新编译的）

NickLiu635, 2020-01-13:
@coolsnowwolf 使用ath79添加ath10k驱动后仍然报错
ath10k相关日志如下

> [   12.717321] NET: Registered protocol family 24
[   12.760640] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   12.767479] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   12.982393] firmware ath10k!pre-cal-pci-0000:00:00.0.bin: firmware_loading_store: map pages failed
[   13.040967] firmware ath10k!cal-pci-0000:00:00.0.bin: firmware_loading_store: map pages failed
[   13.099359] firmware ath10k!QCA9888!hw2.0!firmware-6.bin: firmware_loading_store: map pages failed
[   13.880093] ath10k_pci 0000:00:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   13.889661] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   13.910305] ath10k_pci 0000:00:00.0: firmware ver 10.4-3.9.0.2-00040 api 5 features no-p2p,mfp,peer-flow-ctrl,allows-mesh-bcast,no-ps crc32 0a26c866
[   14.198691] ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,vendor=168c,device=0056,subsystem-vendor=0000,subsystem-device=0000 from ath10k/QCA9888/hw2.0/board-2.bin
[   14.240600] firmware ath10k!QCA9888!hw2.0!board.bin: firmware_loading_store: map pages failed
[   14.249614] ath10k_pci 0000:00:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9888/hw2.0
[   14.259341] ath10k_pci 0000:00:00.0: failed to fetch board file: -11
[   14.266093] ath10k_pci 0000:00:00.0: could not probe fw (-11)
[   14.358851] ath: EEPROM regdomain: 0x0
[   14.358858] ath: EEPROM indicates default country code should be used
[   14.358861] ath: doing EEPROM country->regdmn map search
[   14.358874] ath: country maps to regdmn code: 0x3a
[   14.358878] ath: Country alpha2 being used: US
[   14.358881] ath: Regpair used: 0x3a

AmadeusGhost, 2020-01-13:
说过了ath79有问题是因为没有caldata.sh等底层文件，不信你去看hotplug.d的实现
而且这些文件也不是说加上就加上了，兼容 报错一堆问题要修
我也不会天天花费时间在这上面

PR #2919 (2020-02-03): ath79: sync and fix wireless
fix #2742 
fix #2878 
fix #2891
closed 2020-02-03 by coolsnowwolf