coolsnowwolf/lede #2749: SSR+ 界面添加服务器节点原地跳转

kfitfk, 2020-01-11:
路由器是 lenovo newifi y1s，固件得小于 15MB 的才能刷。我在 `make menuconfig` 这一步就选了这样：

```
Target System (MediaTek Ralink MIPS)
Subtarget (MT7620 based boards)
Target Profile (Lenovo Y1S)
```

按上面这样配了会清空大多数默认配置。我搜索了一下默认的 `.config` 文件里和 shadowsocks 有关的内容，只勾选了与之相关的那一些。下面附上我的 `.config` 文件。

[y1s_config_final.txt](https://github.com/coolsnowwolf/lede/files/4049102/y1s_config_final.txt)

刷机完成后发现在 ShadowsocksR Plus+ 的“服务器节点”配置中，点击“添加”永远在原地跳转。

[image]

搜了一下午没找到有用的信息。有人遇到过类似问题吗？想知道这是因为编译配置的问题，还是有其他途径可修复？谢谢。

luckyyyyy, 2020-01-11:
我实测没有这个问题

kfitfk, 2020-01-11:
> 我实测没有这个问题

能麻烦你建一个节点类型为 trojan 的 server，看一下生成的 shadowsocksr config 文件吗？我试试能不能直接 ssh 上去手动改 config 文件来把节点配置给加上。

luckyyyyy, 2020-01-11:
```
config servers
	option type 'trojan'
	option alias '1'
	option server 'www.baidu.com'
	option server_port '3'
	option password '4'
	option insecure '1'
	option tls '1'
	option tls_host '5'
	option mux '1'
	option concurrency '8'
	option switch_enable '1'
	option local_port '1234'
```

kfitfk, 2020-01-11:
非常感谢，搞定了！

另外，经你这么一说，我觉得更多是跟我的机器环境有关系。结果发现界面上一直 302 是浏览器的问题。在 macOS Google Chrome 79.0.3945.117 版本下我一直进不了服务器节点添加界面，换 Firefox Nightly 就好了。

closed 2020-01-11 by kfitfk

luckyyyyy, 2020-01-11:
哈哈 我的系统和浏览器版本应该和你一样的 解决了就好

发自我的iPhone