coolsnowwolf/lede #2751: 在/utils/选中mdadm进行编译后，一直出现这个，之后就没有之后了

tcsr200722, 2020-01-12:
make[4]: Entering directory '/home/lenovo/lede/build_dir/target-x86_64_musl/mdadm-4.0'
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/lenovo/lede/build_dir/target-x86_64_musl/mdadm-4.0:mdadm-4.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -DHAVE_STDINT_H -DNO_COROSYNC -DNO_DLM -DUSE_PTHREADS -DCONFFILE='"/var/etc/mdadm.conf"' -DMAP_DIR='"/var/run/mdadm"' -DMDMON_DIR='"/var/run/mdadm"' -DFAILED_SLOTS_DIR='"/var/run/mdadm/failed-slots"' -I/home/lenovo/lede/staging_dir/target-x86_64_musl/usr/include -I/home/lenovo/lede/staging_dir/target-x86_64_musl/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -DUSE_PTHREADS -DBINDIR="/sbin" -c -o mdadm.o mdadm.c
mdadm.c: In function 'main':
mdadm.c:1306:24: warning: implicit declaration of function 'minor'; did you mean 'mknod'? [-Wimplicit-function-declaration]
ident.super_minor = minor(stb.st_rdev);
^~~~~
mknod
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/lenovo/lede/build_dir/target-x86_64_musl/mdadm-4.0:mdadm-4.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -DHAVE_STDINT_H -DNO_COROSYNC -DNO_DLM -DUSE_PTHREADS -DCONFFILE='"/var/etc/mdadm.conf"' -DMAP_DIR='"/var/run/mdadm"' -DMDMON_DIR='"/var/run/mdadm"' -DFAILED_SLOTS_DIR='"/var/run/mdadm/failed-slots"' -I/home/lenovo/lede/staging_dir/target-x86_64_musl/usr/include -I/home/lenovo/lede/staging_dir/target-x86_64_musl/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -DUSE_PTHREADS -DBINDIR="/sbin" -c -o config.o config.c
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/lenovo/lede/build_dir/target-x86_64_musl/mdadm-4.0:mdadm-4.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -DHAVE_STDINT_H -DNO_COROSYNC -DNO_DLM -DUSE_PTHREADS -DCONFFILE='"/var/etc/mdadm.conf"' -DMAP_DIR='"/var/run/mdadm"' -DMDMON_DIR='"/var/run/mdadm"' -DFAILED_SLOTS_DIR='"/var/run/mdadm/failed-slots"' -I/home/lenovo/lede/staging_dir/target-x86_64_musl/usr/include -I/home/lenovo/lede/staging_dir/target-x86_64_musl/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -DUSE_PTHREADS -DBINDIR="/sbin" -c -o policy.o policy.c
policy.c: In function 'disk_path':
policy.c:215:23: warning: implicit declaration of function 'makedev'; did you mean 'map_dev'? [-Wimplicit-function-declaration]
if (stb.st_rdev != makedev(disk->disk.major, disk->disk.minor))
^~~~~~~
map_dev
policy.c: In function 'devid_policy':
policy.c:425:20: warning: implicit declaration of function 'major' [-Wimplicit-function-declaration]
disk.disk.major = major(dev);
^~~~~
policy.c:426:20: warning: implicit declaration of function 'minor'; did you mean 'mknod'? [-Wimplicit-function-declaration]
disk.disk.minor = minor(dev);
^~~~~
mknod
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/lenovo/lede/build_dir/target-x86_64_musl/mdadm-4.0:mdadm-4.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -DHAVE_STDINT_H -DNO_COROSYNC -DNO_DLM -DUSE_PTHREADS -DCONFFILE='"/var/etc/mdadm.conf"' -DMAP_DIR='"/var/run/mdadm"' -DMDMON_DIR='"/var/run/mdadm"' -DFAILED_SLOTS_DIR='"/var/run/mdadm/failed-slots"' -I/home/lenovo/lede/staging_dir/target-x86_64_musl/usr/include -I/home/lenovo/lede/staging_dir/target-x86_64_musl/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -DUSE_PTHREADS -DBINDIR="/sbin" -c -o mdstat.o mdstat.c
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/lenovo/lede/build_dir/target-x86_64_musl/mdadm-4.0:mdadm-4.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -DHAVE_STDINT_H -DNO_COROSYNC -DNO_DLM -DUSE_PTHREADS -DCONFFILE='"/var/etc/mdadm.conf"' -DMAP_DIR='"/var/run/mdadm"' -DMDMON_DIR='"/var/run/mdadm"' -DFAILED_SLOTS_DIR='"/var/run/mdadm/failed-slots"' -I/home/lenovo/lede/staging_dir/target-x86_64_musl/usr/include -I/home/lenovo/lede/staging_dir/target-x86_64_musl/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -DUSE_PTHREADS -DBINDIR="/sbin" -c -o ReadMe.o ReadMe.c
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/lenovo/lede/build_dir/target-x86_64_musl/mdadm-4.0:mdadm-4.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -DHAVE_STDINT_H -DNO_COROSYNC -DNO_DLM -DUSE_PTHREADS -DCONFFILE='"/var/etc/mdadm.conf"' -DMAP_DIR='"/var/run/mdadm"' -DMDMON_DIR='"/var/run/mdadm"' -DFAILED_SLOTS_DIR='"/var/run/mdadm/failed-slots"' -I/home/lenovo/lede/staging_dir/target-x86_64_musl/usr/include -I/home/lenovo/lede/staging_dir/target-x86_64_musl/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -DUSE_PTHREADS -DBINDIR="/sbin" -c -o util.o util.c
util.c: In function 'md_get_version':
util.c:275:6: warning: implicit declaration of function 'major' [-Wimplicit-function-declaration]
if (major(stb.st_rdev) == MD_MAJOR)
^~~~~
util.c: In function 'test_partition_from_id':
util.c:464:35: warning: implicit declaration of function 'minor'; did you mean 'mknod'? [-Wimplicit-function-declaration]
sprintf(buf, "%d:%d", major(id), minor(id));
^~~~~
mknod
util.c: In function 'devnm2devid':
util.c:955:11: warning: implicit declaration of function 'makedev'; did you mean 'map_dev'? [-Wimplicit-function-declaration]
return makedev(mjr, mnr);
^~~~~~~
map_dev
util.c: In function 'open_container':
util.c:1535:8: error: called object 'major' is not a function or function pointer
(int)major(st.st_rdev), (int)minor(st.st_rdev));
^~~~~
util.c:1529:6: note: declared here
int major, minor;
^~~~~
util.c:1535:32: error: called object 'minor' is not a function or function pointer
(int)major(st.st_rdev), (int)minor(st.st_rdev));
^~~~~
util.c:1529:13: note: declared here
int major, minor;
^~~~~
: recipe for target 'util.o' failed
make[4]: *** [util.o] Error 1
make[4]: Leaving directory '/home/lenovo/lede/build_dir/target-x86_64_musl/mdadm-4.0'
Makefile:70: recipe for target '/home/lenovo/lede/build_dir/target-x86_64_musl/mdadm-4.0/.built' failed
make[3]: *** [/home/lenovo/lede/build_dir/target-x86_64_musl/mdadm-4.0/.built] Error 2
make[3]: Leaving directory '/home/lenovo/lede/package/utils/mdadm'
time: package/utils/mdadm/compile#1.87#0.25#4.20
package/Makefile:107: recipe for target 'package/utils/mdadm/compile' failed
make[2]: *** [package/utils/mdadm/compile] Error 2
make[2]: Leaving directory '/home/lenovo/lede'
package/Makefile:103: recipe for target '/home/lenovo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/lenovo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lenovo/lede'
/home/lenovo/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

closed 2020-01-13 by coolsnowwolf

AmadeusGhost, 2020-01-13:
已修复

tcsr200722, 2020-01-13:
谢谢!