coolsnowwolf/lede #2754: 树莓派4B最新代码编译失败

EnnawYang, 2020-01-12:
看到树莓派系列有代码更新，顺便编译了一下
```
CONFIG_TARGET_brcm2708=y
CONFIG_TARGET_brcm2708_bcm2711=y
CONFIG_TARGET_brcm2708_bcm2711_DEVICE_rpi-4=y
```
其他设置保持默认，编译快结束时出现如下错误
```
  CC      drivers/of/device.o
  CC      drivers/of/platform.o
  CC      drivers/of/configfs.o
  CC      drivers/of/dynamic.o
  CC      drivers/of/fdt.o
  CC      drivers/of/address.o
  CC      drivers/of/irq.o
  CC      drivers/of/of_net.o
drivers/of/of_net.c: In function 'read_mtd_mac_address':
drivers/of/of_net.c:58:32: error: passing argument 4 of 'mtd_read' from incompatible pointer type [-Werror=incompatible-pointer-types]
  return mtd_read(mtd, from, 6, &retlen, mac);
                                ^
In file included from drivers/of/of_net.c:14:0:
./include/linux/mtd/mtd.h:418:5: note: expected 'size_t * {aka long unsigned int *}' but argument is of type 'int *'
 int mtd_read(struct mtd_info *mtd, loff_t from, size_t len, size_t *retlen,
     ^~~~~~~~
drivers/of/of_net.c: In function 'read_mtd_mac_address_ascii':
drivers/of/of_net.c:66:30: error: passing argument 4 of 'mtd_read' from incompatible pointer type [-Werror=incompatible-pointer-types]
  if (mtd_read(mtd, from, 12, &retlen, buf)) {
                              ^
In file included from drivers/of/of_net.c:14:0:
./include/linux/mtd/mtd.h:418:5: note: expected 'size_t * {aka long unsigned int *}' but argument is of type 'int *'
 int mtd_read(struct mtd_info *mtd, loff_t from, size_t len, size_t *retlen,
     ^~~~~~~~
drivers/of/of_net.c:74:32: error: passing argument 4 of 'mtd_read' from incompatible pointer type [-Werror=incompatible-pointer-types]
  if (mtd_read(mtd, from+12, 5, &retlen, buf+12)) {
                                ^
In file included from drivers/of/of_net.c:14:0:
./include/linux/mtd/mtd.h:418:5: note: expected 'size_t * {aka long unsigned int *}' but argument is of type 'int *'
 int mtd_read(struct mtd_info *mtd, loff_t from, size_t len, size_t *retlen,
     ^~~~~~~~
At top level:
drivers/of/of_net.c:88:3: warning: 'mtd_mac_address_properties' defined but not used [-Wunused-variable]
 } mtd_mac_address_properties[] = {
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:303: recipe for target 'drivers/of/of_net.o' failed
make[7]: *** [drivers/of/of_net.o] Error 1
scripts/Makefile.build:544: recipe for target 'drivers/of' failed
make[6]: *** [drivers/of] Error 2
Makefile:1052: recipe for target 'drivers' failed
make[5]: *** [drivers] Error 2
make[5]: Leaving directory '/home/xw/openwrt/lede-pi4b/build_dir/target-aarch64_cortex-a72_musl/linux-brcm2708_bcm2711/linux-4.19.93'
Makefile:34: recipe for target '/home/xw/openwrt/lede-pi4b/build_dir/target-aarch64_cortex-a72_musl/linux-brcm2708_bcm2711/linux-4.19.93/.image' failed
make[4]: *** [/home/xw/openwrt/lede-pi4b/build_dir/target-aarch64_cortex-a72_musl/linux-brcm2708_bcm2711/linux-4.19.93/.image] Error 2
make[4]: Leaving directory '/home/xw/openwrt/lede-pi4b/target/linux/brcm2708'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/xw/openwrt/lede-pi4b/target/linux'
time: target/linux/install#224.30#19.45#243.74
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/xw/openwrt/lede-pi4b'
target/Makefile:19: recipe for target '/home/xw/openwrt/lede-pi4b/staging_dir/target-aarch64_cortex-a72_musl/stamp/.target_install' failed
make[1]: *** [/home/xw/openwrt/lede-pi4b/staging_dir/target-aarch64_cortex-a72_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/xw/openwrt/lede-pi4b'
/home/xw/openwrt/lede-pi4b/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

AmadeusGhost, 2020-01-12:
我的锅Orz
树莓派我已经加上了对应的firmware，应该没问题了。
这个```target/linux/generic/pending-4.19/681-NET-add-of_get_mac_address_mtd.patch```是为了在```device-tree```支持```mtd-mac-address-ascii```，初衷是为了解决一些把mac以ascii方式写入设备的sb产品的随机mac问题，不过看起来bcm2708不兼容Orz，可以先把这个patch改回去看看树莓派还有没有问题。

sdf8057, 2020-01-12:
编译qemu armv8也有同样的问题

Guwalgiyakuan, 2020-01-12:
树莓派3B 三小时前刚同步的代码出现同样的错误。
```
CONFIG_TARGET_brcm2708=y
CONFIG_TARGET_brcm2708_bcm2710=y
CONFIG_TARGET_brcm2708_bcm2710_DEVICE_rpi-3=y
```
```
  CC      drivers/of/base.o
  CC      drivers/of/device.o
  CC      drivers/of/platform.o
  CC      drivers/of/property.o
  CC      drivers/of/kobj.o
  CC      drivers/of/configfs.o
  CC      drivers/of/dynamic.o
  CC      drivers/of/fdt.o
  CC      drivers/of/fdt_address.o
  CC      drivers/of/address.o
  CC      drivers/of/irq.o
  CC      drivers/of/of_net.o
drivers/of/of_net.c: In function 'read_mtd_mac_address':
drivers/of/of_net.c:58:32: error: passing argument 4 of 'mtd_read' from incompatible pointer type [-Werror=incompatible-pointer-types]
  return mtd_read(mtd, from, 6, &retlen, mac);
                                ^
In file included from drivers/of/of_net.c:14:0:
./include/linux/mtd/mtd.h:418:5: note: expected 'size_t * {aka long unsigned int *}' but argument is of type 'int *'
 int mtd_read(struct mtd_info *mtd, loff_t from, size_t len, size_t *retlen,
     ^~~~~~~~
drivers/of/of_net.c: In function 'read_mtd_mac_address_ascii':
drivers/of/of_net.c:66:30: error: passing argument 4 of 'mtd_read' from incompatible pointer type [-Werror=incompatible-pointer-types]
  if (mtd_read(mtd, from, 12, &retlen, buf)) {
                              ^
In file included from drivers/of/of_net.c:14:0:
./include/linux/mtd/mtd.h:418:5: note: expected 'size_t * {aka long unsigned int *}' but argument is of type 'int *'
 int mtd_read(struct mtd_info *mtd, loff_t from, size_t len, size_t *retlen,
     ^~~~~~~~
drivers/of/of_net.c:74:32: error: passing argument 4 of 'mtd_read' from incompatible pointer type [-Werror=incompatible-pointer-types]
  if (mtd_read(mtd, from+12, 5, &retlen, buf+12)) {
                                ^
In file included from drivers/of/of_net.c:14:0:
./include/linux/mtd/mtd.h:418:5: note: expected 'size_t * {aka long unsigned int *}' but argument is of type 'int *'
 int mtd_read(struct mtd_info *mtd, loff_t from, size_t len, size_t *retlen,
     ^~~~~~~~
At top level:
drivers/of/of_net.c:88:3: warning: 'mtd_mac_address_properties' defined but not used [-Wunused-variable]
 } mtd_mac_address_properties[] = {
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:303: recipe for target 'drivers/of/of_net.o' failed
make[7]: *** [drivers/of/of_net.o] Error 1
scripts/Makefile.build:544: recipe for target 'drivers/of' failed
make[6]: *** [drivers/of] Error 2
Makefile:1052: recipe for target 'drivers' failed
make[5]: *** [drivers] Error 2
make[5]: Leaving directory '/home/git/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/linux-4.19.93'
Makefile:34: recipe for target '/home/git/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/linux-4.19.93/.image' failed
make[4]: *** [/home/git/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/linux-4.19.93/.image] Error 2
make[4]: Leaving directory '/home/git/lede/target/linux/brcm2708'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/git/lede/target/linux'
time: target/linux/install#424.43#49.34#477.33
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/git/lede'
target/Makefile:19: recipe for target '/home/git/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install' failed
make[1]: *** [/home/git/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/git/lede'
/home/git/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

xiexianghua, 2020-01-13:
我用香橙派zero plus也出错了
 if (mtd_read(mtd, from, 12, &retlen, buf)) {
                              ^
In file included from drivers/of/of_net.c:13:0:
./include/linux/mtd/mtd.h:437:5: note: expected 'size_t * {aka long unsigned int *}' but argument is of type 'int *'
 int mtd_read(struct mtd_info *mtd, loff_t from, size_t len, size_t *retlen,
     ^~~~~~~~
drivers/of/of_net.c:73:32: error: passing argument 4 of 'mtd_read' from incompatible pointer type [-Werror=incompatible-pointer-types]
  if (mtd_read(mtd, from+12, 5, &retlen, buf+12)) {
                                ^
In file included from drivers/of/of_net.c:13:0:
./include/linux/mtd/mtd.h:437:5: note: expected 'size_t * {aka long unsigned int *}' but argument is of type 'int *'
 int mtd_read(struct mtd_info *mtd, loff_t from, size_t len, size_t *retlen,
     ^~~~~~~~
cc1: some warnings being treated as errors
make[7]: *** [drivers/of/of_net.o] Error 1
make[6]: *** [drivers/of] Error 2
make[5]: *** [drivers] Error 2
make[5]: Leaving directory `/home/xiexianghua95/lede/build_dir/target-aarch64_cortex-a53_musl/linux-sunxi_cortexa53/linux-4.14.162'
make[4]: *** [/home/xiexianghua95/lede/build_dir/target-aarch64_cortex-a53_musl/linux-sunxi_cortexa53/linux-4.14.162/.image] Error 2
make[4]: Leaving directory `/home/xiexianghua95/lede/target/linux/sunxi'
make[3]: *** [install] Error 2
make[3]: Leaving directory `/home/xiexianghua95/lede/target/linux'
time: target/linux/install#16.08#1.96#37.04
make[2]: *** [target/linux/install] Error 2

AmadeusGhost, 2020-01-13:
git pull后再试试，如果没问题了可以关闭。

xiexianghua, 2020-01-13:
git pull后没问题了

EnnawYang, 2020-01-13:
> git pull后再试试，如果没问题了可以关闭。

好的 我等会试下，issues先关闭了

closed 2020-01-13 by EnnawYang

AmadeusGhost, 2020-03-27:
> 顺利编译成功

请问树莓派4B刷入openwrt后支持5G开启80MHz吗？有人跟我说不能诶

EnnawYang, 2020-03-27:
> > 顺利编译成功
> 
> 请问树莓派4B刷入openwrt后支持5G开启80MHz吗？有人跟我说不能诶

不支持

AmadeusGhost, 2020-03-27:
> 不支持

所以是默认没无线，需要改成40MHz才能用吗？

EnnawYang, 2020-03-27:
> > 不支持
> 
> 所以是默认没无线，需要改成40MHz才能用吗？

默认开启，但是好像要固定信道才能有信号，用到现在只能到20MHz，不能上40MHz和80MHz（切换到40MHz但连接后显示为20MHz，切换到80MHz就没无线信号了）

AmadeusGhost, 2020-03-27:
> 默认开启，但是好像要固定信道才能有信号，用到现在只能到20MHz，不能上40MHz和80MHz（切换到40MHz但连接后显示为20MHz，切换到80MHz就没无线信号了）

那我把默认80改成40吧🤔

EnnawYang, 2020-03-27:
> > 默认开启，但是好像要固定信道才能有信号，用到现在只能到20MHz，不能上40MHz和80MHz（切换到40MHz但连接后显示为20MHz，切换到80MHz就没无线信号了）
> 
> 那我把默认80改成40吧🤔

```
Fri Mar 27 22:12:21 2020 daemon.notice netifd: Interface 'lan' is enabled
Fri Mar 27 22:12:21 2020 daemon.notice netifd: Interface 'lan' is setting up now
Fri Mar 27 22:12:21 2020 daemon.notice netifd: Interface 'lan' is now up
Fri Mar 27 22:12:21 2020 daemon.notice netifd: Network device 'lo' link is up
Fri Mar 27 22:12:21 2020 daemon.notice netifd: Interface 'loopback' has link connectivity
Fri Mar 27 22:12:21 2020 daemon.notice netifd: radio0 (1301): command failed: No error information (-524)
Fri Mar 27 22:12:21 2020 user.notice mac80211: Failed command: iw dev wlan0 del
Fri Mar 27 22:12:21 2020 user.notice mac80211: Failed command: iw phy phy0 set antenna 0xffffffff 0xffffffff
Fri Mar 27 22:12:21 2020 daemon.notice netifd: radio0 (1301): command failed: I/O error (-5)
Fri Mar 27 22:12:21 2020 user.notice mac80211: Failed command: iw phy phy0 set distance 0
Fri Mar 27 22:12:21 2020 daemon.notice netifd: radio0 (1301): command failed: Too many open files in system (-23)
Fri Mar 27 22:12:21 2020 user.notice mac80211: Failed command: iw phy phy0 interface add wlan0 type __ap
Fri Mar 27 22:12:21 2020 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Fri Mar 27 22:12:21 2020 kern.info kernel: [    9.816909] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Fri Mar 27 22:12:21 2020 kern.info kernel: [    9.825045] br-lan: port 1(wlan0) entered blocking state
Fri Mar 27 22:12:21 2020 kern.info kernel: [    9.830367] br-lan: port 1(wlan0) entered disabled state
Fri Mar 27 22:12:21 2020 kern.info kernel: [    9.835864] device wlan0 entered promiscuous mode
Fri Mar 27 22:12:21 2020 kern.info kernel: [    9.840649] br-lan: port 1(wlan0) entered blocking state
Fri Mar 27 22:12:21 2020 kern.info kernel: [    9.845972] br-lan: port 1(wlan0) entered forwarding state
Fri Mar 27 22:12:21 2020 daemon.notice netifd: bridge 'br-lan' link is up
Fri Mar 27 22:12:21 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Fri Mar 27 22:12:21 2020 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Fri Mar 27 22:12:21 2020 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Fri Mar 27 22:12:21 2020 kern.info kernel: [    9.851660] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Fri Mar 27 22:12:21 2020 daemon.warn hostapd: wlan0: Could not connect to kernel driver
Fri Mar 27 22:12:21 2020 daemon.err hostapd: Using interface wlan0 with hwaddr dc:a6:32:24:f9:af and ssid "Pi4B"
```
这信息有用么？ 一星期之前编译的版本，最近树莓派一直没拿出来用