coolsnowwolf/lede #2757: ShadowsocksR Plus+在K2P上运行不正常

KevinMX, 2020-01-12:
设备：Phicomm K2P，编译配置[在此](https://github.com/KevinMX/openwrt-phicomm-k2p-build)
v2ray：h2+tls，绕过大陆ip模式，仅代理常用端口，使用PDNSD TCP查询并缓存
症状：v2ray进程不断重启/中断，且此时整个设备都无法正常联网。中途有出现过OOM的情况，将UDP中继关闭，改为单线程后不再OOM，但v2ray依旧会中断/重启。
log已贴上。
[dmesg.log](https://github.com/coolsnowwolf/lede/files/4050305/dmesg.log)
[k2pdmesg.log](https://github.com/coolsnowwolf/lede/files/4050306/k2pdmesg.log)
[system.log](https://github.com/coolsnowwolf/lede/files/4050307/system.log)

hxlls, 2020-01-12:
K2P我记得好像是128M内存的把，可能V2ray运行内存太小，把zram 编译上去看看

coolsnowwolf, 2020-01-13:
大概率是内存不足，考虑增大内存或者更换其他路由

KevinMX, 2020-01-13:
> K2P我记得好像是128M内存的把，可能V2ray运行内存太小，把zram 编译上去看看

```root@Phicomm-K2P-OpenWrt:~# free -m
              total        used        free      shared  buff/cache   available
Mem:         123156       86500       27224        1052        9432        8512
Swap:         61436       27136       34300
```
多线程并发转发（自动），UDP中继（开启），添加了zram-swap后的效果，看起来大概是内存不足的问题了，有时候占用已经超过K2P的128M了。现在勉强能稳定运行了。
不过以前并没有出现过类似情况，怀疑可能新版本有内存溢出？
我开单线程再观察一下。

KevinMX, 2020-01-13:
目前开单线程基本正常，内存/zram占用基本稳定下来了。
```
              total        used        free      shared  buff/cache   available
Mem:         123156       84768       28196         640       10192       18596
Swap:         61436       16384       45052
```

closed 2020-01-13 by KevinMX