coolsnowwolf/lede #2758: 新原来编译的比原来大了很多，是怎么回事呢？

xinpenghan, 2020-01-12:
之前12.03版本的时候编译x86平台的固件是50几兆，现在编译出来是100多兆，大了很多

MxCen, 2020-01-12:
屁事水，直接开启GZIP压缩就是了。

ssitssit, 2020-01-12:
问的问题太水了！

dolphinpaopao, 2020-01-12:
作者真的宽容。请关闭并理解issue含义，开启前请仔细搜索通过论坛QQ群等其他渠道获取信息。这里不是垃圾场。

AmadeusGhost, 2020-01-12:
不选上docker就小很多了。

xinpenghan, 2020-01-12:
> 屁事水，直接开启GZIP压缩就是了。

GZIP早开了，一样

xinpenghan, 2020-01-12:
> 作者真的宽容。请关闭并理解问题含义，开启前请仔细搜索通过论坛QQ群等其他渠道获取信息。这里不是垃圾场。

你不正面回答问题，你不觉得反而是在倒垃圾吗？

lifetyper, 2020-01-13:
自己选了哪些组件也不说，config文件怎么样也不说，就在这里说固件太大，能回答你这个问题的恐怕是神仙。

9l, 2020-01-13:
你预装了太多软件进去吧，我就留了两个luci软件，出来的才20MB多一些

xinpenghan, 2020-01-14:
配置文件一直没改，12.03版本编译出来，combined-squashfs.img有53mb，新版本编译出来有185mb
         # 编译x64固件:
          cat >> .config <<EOF
          CONFIG_TARGET_x86=y
          CONFIG_TARGET_x86_64=y
          CONFIG_TARGET_x86_64_Generic=y
          EOF

          # 固件压缩:
          cat >> .config <<EOF
          CONFIG_TARGET_IMAGES_GZIP=y
          EOF
          

          # 编译UEFI固件:
          # cat >> .config <<EOF
          # CONFIG_EFI_IMAGES=y
          # EOF
          

          # IPv6支持:
          cat >> .config <<EOF
          CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
          CONFIG_PACKAGE_ipv6helper=y
          EOF

          
          # 多文件系统支持:
          cat >> .config <<EOF
          CONFIG_PACKAGE_kmod-fs-nfs=y
          CONFIG_PACKAGE_kmod-fs-nfs-common=y
          CONFIG_PACKAGE_kmod-fs-nfs-v3=y
          CONFIG_PACKAGE_kmod-fs-nfs-v4=y
          CONFIG_PACKAGE_kmod-fs-ntfs=y
          CONFIG_PACKAGE_kmod-fs-squashfs=y
          CONFIG_PACKAGE_kmod-fs-cifs=y
          CONFIG_PACKAGE_kmod-nls-utf8=y
          CONFIG_PACKAGE_kmod-nls-base=y
          CONFIG_PACKAGE_kmod-crypto-hmac=y
          CONFIG_PACKAGE_kmod-crypto-md5=y
          CONFIG_PACKAGE_kmod-crypto-misc=y
          CONFIG_PACKAGE_kmod-fs-ext4=y
          CONFIG_PACKAGE_kmod-fs-vfat=y
          EOF

          
          # USB3.0支持:
          cat >> .config <<EOF
          CONFIG_PACKAGE_kmod-usb-ohci=y
          CONFIG_PACKAGE_kmod-usb-ohci-pci=y
          CONFIG_PACKAGE_kmod-usb2=y
          CONFIG_PACKAGE_kmod-usb2-pci=y
          CONFIG_PACKAGE_kmod-usb3=y
          CONFIG_PACKAGE_kmod-usb-storage=y
          CONFIG_PACKAGE_kmod-usb-storage-extras=y
          CONFIG_PACKAGE_kmod-usb-storage-uas=y
          CONFIG_PACKAGE_kmod-usb-uhci=y
          CONFIG_PACKAGE_kmod-scsi-generic=y
          CONFIG_PACKAGE_kmod-scsi-core=y
          EOF
          

          # 常用LuCI插件选择:
          cat >> .config <<EOF
          CONFIG_PACKAGE_luci-app-adbyby-plus=y
          CONFIG_PACKAGE_luci-app-frpc=y
          CONFIG_PACKAGE_luci-app-hd-idle=y
          CONFIG_PACKAGE_luci-app-sqm=y
          CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Kcptun=y
          CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks=y
          CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=y
          CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Socks=y
          CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=y
          CONFIG_PACKAGE_luci-app-filetransfer=y
          CONFIG_PACKAGE_luci-app-v2ray-server=y
          CONFIG_PACKAGE_luci-app-vsftpd=y
          CONFIG_PACKAGE_luci-app-usb-printer=y
          CONFIG_PACKAGE_luci-app-wol=y
          CONFIG_PACKAGE_luci-app-vlmcsd=y
          CONFIG_PACKAGE_luci-app-syncdial=y
          CONFIG_PACKAGE_luci-app-mwan3=y
          CONFIG_PACKAGE_luci-app-n2n_v2=y
          CONFIG_PACKAGE_luci-app-qos=y
          CONFIG_PACKAGE_luci-app-ramfree=y
          CONFIG_PACKAGE_luci-app-zerotier=y
          CONFIG_PACKAGE_luci-app-nlbwmon=y
          CONFIG_PACKAGE_luci-app-autoreboot=y
          CONFIG_PACKAGE_luci-app-accesscontrol=y
          CONFIG_PACKAGE_luci-app-netdata=y
          CONFIG_PACKAGE_luci-app-samba4=y
          CONFIG_SAMBA4_SERVER_NETBIOS=y
          # CONFIG_PACKAGE_autosamba is not set
          # CONFIG_PACKAGE_luci-app-samba is not set
          # CONFIG_PACKAGE_luci-app-ipsec-vpnd is not set
          # CONFIG_PACKAGE_luci-app-pptp-server is not set
          # CONFIG_PACKAGE_luci-app-qbittorrent is not set
          # CONFIG_PACKAGE_luci-app-xlnetacc is not set
          # CONFIG_PACKAGE_luci-app-wrtbwmon is not set
          # CONFIG_PACKAGE_luci-app-unblockmusic is not set
          EOF
          
          
          # LuCI主题:
          cat >> .config <<EOF
          CONFIG_PACKAGE_luci-theme-argon=y
          CONFIG_PACKAGE_luci-theme-netgear=y
          EOF

          # 常用软件包:
          cat >> .config <<EOF
          CONFIG_PACKAGE_curl=y
          CONFIG_PACKAGE_htop=y
          CONFIG_PACKAGE_nano=y
          CONFIG_PACKAGE_screen=y
          CONFIG_PACKAGE_tree=y
          CONFIG_PACKAGE_vim-fuller=y
          CONFIG_PACKAGE_wget=y
          CONFIG_PACKAGE_nginx=y
          CONFIG_PACKAGE_php7=y
          CONFIG_PHP7_LIBFREETYPE=y
          CONFIG_PACKAGE_php7-fpm=y
          CONFIG_PACKAGE_php7-mod-curl=y
          CONFIG_PACKAGE_php7-mod-gd=y
          CONFIG_PACKAGE_php7-mod-iconv=y
          CONFIG_PACKAGE_php7-mod-json=y
          CONFIG_PACKAGE_php7-mod-opcache=y
          CONFIG_PACKAGE_php7-mod-session=y
          CONFIG_PACKAGE_php7-mod-zip=y
          CONFIG_PACKAGE_php7-mod-mbstring=y
          CONFIG_PACKAGE_zoneinfo-asia=y
          EOF

9l, 2020-01-14:
linux内核升级了

closed 2020-01-21 by coolsnowwolf