coolsnowwolf/lede #2765: 大佬，能不能来个k3的openwrt-ci.yml

jsyzjhj, 2020-01-13:
谢谢！

EnnawYang, 2020-01-13:
自己改咯

jsyzjhj, 2020-01-13:
@EnnawYang 没教程唉。。不知道改哪

yonggedebaqi, 2020-01-13:
我也是改了好久编译成功了
[image]

yonggedebaqi, 2020-01-13:
就是不知道能不能刷机启动机器

9l, 2020-01-13:
CONFIG_TARGET_bcm53xx=y
	CONFIG_TARGET_bcm53xx_DEVICE_phicomm-k3=y

只能帮你到这里了

jsyzjhj, 2020-01-14:
> 就是不知道能不能刷机启动机器

这玩意太难了

CYaNu, 2020-01-14:
> > 就是不知道能不能刷机启动机器
> 
> 这玩意太难了，这是昨了？
> 
> 2020-01-13T12:46:28.7080355Z touch /home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/libiconv-1.11.1/.pkgdir/libiconv-full.installed
> 2020-01-13T12:46:28.7102326Z mkdir -p /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp
> 2020-01-13T12:46:28.7123732Z SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/libiconv-1.11.1/.pkgdir/libiconv-full/. /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
> 2020-01-13T12:46:28.7161574Z touch /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.libiconv-full_installed
> 2020-01-13T12:46:28.7237526Z echo "libiconv-full" >> /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/libiconv-full.default.install
> 2020-01-13T12:46:28.7255162Z make[3]: Leaving directory '/home/runner/work/lede/lede/package/libs/libiconv-full'
> 2020-01-13T12:46:28.7260613Z time: package/libs/libiconv-full/compile#12.87#3.46#18.16
> 2020-01-13T12:46:28.7288234Z make[2]: Leaving directory '/home/runner/work/lede/lede'
> 2020-01-13T12:46:28.7311001Z make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
> 2020-01-13T12:46:28.7311442Z package/Makefile:103: recipe for target '/home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile' failed
> 2020-01-13T12:46:28.7312713Z Build failed - please re-run with -j1 to see the real error message
> 2020-01-13T12:46:28.7313020Z make[1]: Leaving directory '/home/runner/work/lede/lede'
> 2020-01-13T12:46:28.7313904Z make: *** [world] Error 1
> 2020-01-13T12:46:28.7314296Z /home/runner/work/lede/lede/include/toplevel.mk:216: recipe for target 'world' failed
> 2020-01-13T12:46:28.7320408Z ##[error]Process completed with exit code 2.
> 2020-01-13T12:46:28.7340166Z Cleaning up orphan processes

用-j1试试

jsyzjhj, 2020-01-14:
> 用-j1试试

用了的。不知道是不是libiconv这玩意儿有问题

icksky, 2020-01-16:
参考我 FORK 的版本 就是K3的 [lede](https://github.com/icksky/lede)

closed 2020-01-17 by coolsnowwolf