coolsnowwolf/lede #2768: Flow加速问题

git-maven, 2020-01-13:
开了FLOW 加速多拨网速能叠加到1000M,但是外网访问黑裙辉网速就只有几十K/s
如果关闭了FLOW 加速多拨网速只能叠加到500M,但是外网访问黑裙辉网速正常
不知大伙有没有遇到过这个问题
ps:开了端口映射

Albert-Zhan, 2020-01-13:
你应该打开bbr了吧，关闭就好了

git-maven, 2020-01-13:
> 你应该打开bbr了吧，关闭就好了

没开bbr，就是默认刷好的设置，然后关了flow加速，外网访问群晖的速度就正常了- -，好奇怪

caicaicai21, 2020-01-14:
FLOW驱动好像是对外网访问会有影响，至少我的虚拟机里面的如果开启了是有问题的，硬路由没测试过，不清楚。

git-maven, 2020-01-14:
> FLOW驱动好像是对外网访问会有影响，至少我的虚拟机里面的如果开启了是有问题的，硬路由没测试过，不清楚。

我的是华硕ARCH17，还想换个软路由看看能不能改善，看来是没必要换了

Wilson-coder, 2020-01-14:
多谢这个问题的提醒，我把Flow加速关掉了，然后异地远程桌面基本可以直接看视频了。关掉之前点击一下都要等半天反应，远程桌面客户端提示网速质量极差，现在已经变成极好了。

git-maven, 2020-01-14:
> 多谢这个问题的提醒，我把Flow加速关掉了，然后异地远程桌面基本可以直接看视频了。关掉之前点击一下都要等半天反应，远程桌面客户端提示网速质量极差，现在已经变成极好了。

你关掉以后下行速度是不是变慢了？我开启之前可以跑满千兆的，关了就只能500M左右了。

keivenliao, 2020-01-14:
这个应该对X86的没有影响吧，我的没有这个问题。

Wilson-coder, 2020-01-14:
> > 多谢这个问题的提醒，我把Flow加速关掉了，然后异地远程桌面基本可以直接看视频了。关掉之前点击一下都要等半天反应，远程桌面客户端提示网速质量极差，现在已经变成极好了。
> 
> 你关掉以后下行速度是不是变慢了？我开启之前可以跑满千兆的，关了就只能500M左右了。

我的下行速度本来就只有200M，没有发觉到变化。

caicaicai21, 2020-01-14:
> 这个应该对X86的没有影响吧，我的没有这个问题。

我虚拟机下的会有影响，那时候刚开始用L大的固件，我向外网映射内网的web服务器，如果开启flow，访问会变得极慢。

keivenliao, 2020-01-14:
> > 这个应该对X86的没有影响吧，我的没有这个问题。
> 
> 我虚拟机下的会有影响，那时候刚开始用L大的固件，我向外网映射内网的web服务器，如果开启flow，访问会变得极慢。

[image]

只开这两项也会？

caicaicai21, 2020-01-14:
> > > 这个应该对X86的没有影响吧，我的没有这个问题。
> > 
> > 
> > 我虚拟机下的会有影响，那时候刚开始用L大的固件，我向外网映射内网的web服务器，如果开启flow，访问会变得极慢。
> 
> [image]
> 
> 只开这两项也会？

BBR不影响，只有FLOW有影响。

shshunlan, 2020-01-16:
flow和多wan或qos有冲突

hyyz17200, 2020-02-18:
> 这个应该对X86的没有影响吧，我的没有这个问题。

x86也会有，我是在远程桌面的时候发现的，开了外网远程极卡

closed 2020-02-19 by coolsnowwolf