coolsnowwolf/lede #2771: pdnsd 获得的是台湾ip问题

szoyj, 2020-01-13:
按照正常逻辑，pdnsd通过代理（美国的vps）去取ip的时，在cdn的作用下，获得的是ip是美国境内的ip，从代理机ping此ip时延迟是很低的。但是目前发现的情况是pdnsd里缓存的ip是台湾ip。

环境：
```
代理机 位于美国加利福尼亚州洛杉矶，装了v2
ssrp+   
    运行模式：gfw列表模式
    需要代理的端口：所有
    DNS解析方式：使用PDNSD TCP 查询并缓存
    访问国外域名DNS服务器：8.8.4.4

主路由ip 192.168.30.1
旁路由安装openwrt ip 192.168.30.250
内网pc，网关ip 192.168.30.1，并未指向旁路由

```

/var/etc/pdnsd.conf（原生，无额外改动）
```
global {
        perm_cache=1024;
        cache_dir="/var/pdnsd";
        pid_file = /var/run/pdnsd.pid;
        run_as="nobody";
        server_ip = 127.0.0.1;
        server_port = 5335;
        status_ctl = on;
        query_method = tcp_only;
        min_ttl=1h;
        max_ttl=1w;
        timeout=10;
        neg_domain_pol=on;
        proc_limit=2;
        procq_limit=8;
}
server {
        label= "ssr-usrdns";
        ip = 8.8.4.4;
        port = 53;
        timeout=6;
        uptest=none;
        interval=10m;
        purge_cache=off;
}
server {
        label= "ssr-pdnsd";
        ip = 208.67.222.222, 208.67.220.220;
        port = 5353;
        timeout=6;
        uptest=none;
        interval=10m;
        purge_cache=off;
}
```
复现步骤
pc上执行下面命令
```
root@ubuntu:~# dig @192.168.30.250 youtube.com
; <<>> DiG 9.10.6 <<>> @192.168.30.250 youtube.com
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 59216
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1024
;; QUESTION SECTION:
;youtube.com.			IN	A

;; ANSWER SECTION:
youtube.com.		3600	IN	A	172.217.160.78（中国台湾台北市）

;; Query time: 486 msec
;; SERVER: 192.168.30.250#53(192.168.30.250)
;; WHEN: Mon Jan 13 19:34:40 CST 2020
;; MSG SIZE  rcvd: 56
```
openwrt上执行下面命令
```
root@OpenWrt:~# pdnsd-ctl dump
Opening socket /var/pdnsd/pdnsd.status
youtube.com.
    01/13 19:34:39    A       172.217.160.78 （中国台湾台北市）

Succeeded
```
**但是当我删除/var/etc/pdnsd.conf里第二个server后**

pc上执行下面命令
```
root@ubuntu:~# dig @192.168.30.250 youtube.com
; <<>> DiG 9.10.6 <<>> @192.168.30.250 youtube.com
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 42171
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1024
;; QUESTION SECTION:
;youtube.com.			IN	A

;; ANSWER SECTION:
youtube.com.		3600	IN	A	172.217.11.174（美国加利福尼亚州洛杉矶）

;; Query time: 937 msec
;; SERVER: 192.168.30.250#53(192.168.30.250)
;; WHEN: Mon Jan 13 18:07:10 CST 2020
;; MSG SIZE  rcvd: 56

```
openwrt上执行下面命令
```
root@OpenWrt:~# pdnsd-ctl dump
Opening socket /var/pdnsd/pdnsd.status
youtube.com.
    01/13 18:07:10    A       172.217.11.174（美国加利福尼亚州洛杉矶）

Succeeded

```

结论就是
**pdnsd是并发查询，查询server2（未走代理，从国内访问）的速度快于server1（在iptables的作用下走了代理，从国外访问），导致获得的ip在cdn的作用是地理位置最近的ip，台湾。**

szoyj, 2020-01-13:
另一个问题 pdnsd debug模式（命令后加参数-g或conf里加入debug=on;）无法使用，会segment fault

rampageX, 2020-01-13:
这个是 pdnsd 的坑，两个办法解决：

1. 在 global 添加一句：

`par_queries=1;`

pdnsd 默认是 2，然后你的第一组服务器只有一个服务器 8.8.4.4，它会用第二组去查。

2. 在第一组再添加一个服务器比如 `8.8.8.8, 8.8.4.4`

9l, 2020-01-13:
哦，我说呢，怪不得在访问控制的强制域名列表里加了域名，有时生效有时不生效，刚开始还以为是域名格式写的不对，后来又以为是dns服务器设置的不对，搞了好久都不能正常工作。
今天看到这个，改了pdnsd的设置后，就好了，真是个坑

luckyyyyy, 2020-01-20:
这个很好 我说怎么回事 有时候是非预期

Bnosora, 2020-01-23:
我今天测试的结果是不管删除第二个服务器、增加`par_queries=1;`还是把第二组的服务器改成8.8.8.8都会回复台湾的谷歌ip……测试b站ip直接返回我本地的服务器ip，应该是dns直接从本地出去了，没走代理。（服务器用的是新加坡的v2）

closed 2020-02-19 by coolsnowwolf