coolsnowwolf/lede #2773: 小马V1直装op-X64出现大量错误日志

nicepass, 2020-01-14:
Tue Jan 14 12:30:27 2020 daemon.err uhttpd[4353]: Cannot get device settings: No such device
Tue Jan 14 12:30:27 2020 daemon.err uhttpd[4353]: Cannot get device settings: No such device
Tue Jan 14 12:30:28 2020 daemon.err uhttpd[4353]: Cannot get device settings: No such device
Tue Jan 14 12:30:28 2020 daemon.err uhttpd[4353]: Cannot get device settings: No such device
Tue Jan 14 12:30:29 2020 daemon.err uhttpd[4353]: Cannot get device settings: No such device
Tue Jan 14 12:30:29 2020 daemon.err uhttpd[4353]: Cannot get device settings: No such device
编译的插件如下图：

[image]

EnnawYang, 2020-01-14:
同机器 没问题

nicepass, 2020-01-14:
格掉硬盘重新安装还是会报这个错误

0x5826, 2020-01-15:
能正常用就没问题

closed 2020-01-15 by coolsnowwolf