coolsnowwolf/lede #2777: 重新安装的系统，新拉的代码，编译出了错误，那位大神帮忙看下，帮解决

dzhuqi, 2020-01-14:
`OpenWrt-libtool: link: x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/z/lede/build_dir/target-x86_64_musl/openldap-2.4.45:openldap-2.4.45 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -fpic -znow -zrelro -o .libs/apitest apitest.o  -L/home/z/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/z/lede/staging_dir/target-x86_64_musl/lib -L/home/z/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/lib -L/home/z/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib ./.libs/libldap.so /home/z/lede/build_dir/target-x86_64_musl/openldap-2.4.45/libraries/liblber/.libs/liblber.so ../../libraries/liblber/.libs/liblber.so -lpthread ../../libraries/liblutil/liblutil.a -lsasl2 -lssl -lcrypto
/home/z/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.5.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ./.libs/libldap.so: undefined reference to `RAND_egd'
collect2: error: ld returned 1 exit status
Makefile:312: recipe for target 'apitest' failed
make[6]: *** [apitest] Error 1
make[6]: Leaving directory '/home/z/lede/build_dir/target-x86_64_musl/openldap-2.4.45/libraries/libldap'
Makefile:298: recipe for target 'all-common' failed
make[5]: *** [all-common] Error 1
make[5]: Leaving directory '/home/z/lede/build_dir/target-x86_64_musl/openldap-2.4.45/libraries'
Makefile:1348: recipe for target 'all-common' failed
make[4]: *** [all-common] Error 1
make[4]: Leaving directory '/home/z/lede/build_dir/target-x86_64_musl/openldap-2.4.45'
Makefile:151: recipe for target '/home/z/lede/build_dir/target-x86_64_musl/openldap-2.4.45/.built' failed
make[3]: *** [/home/z/lede/build_dir/target-x86_64_musl/openldap-2.4.45/.built] Error 2
make[3]: Leaving directory '/home/z/lede/feeds/packages/libs/openldap'
time: package/feeds/packages/openldap/compile#55.24#12.65#67.06
package/Makefile:107: recipe for target 'package/feeds/packages/openldap/compile' failed
make[2]: *** [package/feeds/packages/openldap/compile] Error 2
make[2]: Leaving directory '/home/z/lede'
package/Makefile:103: recipe for target '/home/z/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/z/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/z/lede'
/home/z/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
z@z-GA-MA770T-UD3P:~/lede$ `

dzhuqi, 2020-01-14:
习惯了l大雕的代码，无奈自己只有初中水平，百度了几天也没有解决，只好到这麻烦大家了

whlsxl, 2020-01-14:
有可能是网络问题，不过可以先关闭`openldap`这个模块试试。
在`make menuconfig` 后按`/`查询

dzhuqi, 2020-01-15:
还是一样

fs20sc, 2020-01-15:
rm -rf tmp
rm -rf bin
然后
make -j1 V=s
保持科学上网
用单线程编译看看
你这起码6线程在搞，错误信息貌似没完整

Aes64X, 2020-01-15:
你需要 make -j1 V=s 来找出具体错误

wang452, 2020-01-15:
/home/jay/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/ioctl.c: In function 'cryptodev_ioctl':
/home/jay/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/ioctl.c:831:4: error: implicit declaration of function 'sys_close'; did you mean 'ksys_close'? [-Werror=implicit-function-declaration]
    sys_close(fd);
    ^~~~~~~~~
    ksys_close
cc1: some warnings being treated as errors
scripts/Makefile.build:303: recipe for target '/home/jay/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/ioctl.o' failed
make[6]: *** [/home/jay/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/ioctl.o] Error 1
Makefile:1520: recipe for target '_module_/home/jay/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04' failed
make[5]: *** [_module_/home/jay/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04] Error 2
make[5]: Leaving directory '/home/jay/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.93'
Makefile:27: recipe for target 'build' failed
make[4]: *** [build] Error 2
make[4]: Leaving directory '/home/jay/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04'Makefile:74: recipe for target '/home/jay/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/.built' failed
make[3]: *** [/home/jay/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/.built] Error 2
make[3]: Leaving directory '/home/jay/lede/feeds/packages/utils/cryptodev-linux'
time: package/feeds/packages/cryptodev-linux/compile#1.60#4.53#10.91
package/Makefile:107: recipe for target 'package/feeds/packages/cryptodev-linux/compile' failed
make[2]: *** [package/feeds/packages/cryptodev-linux/compile] Error 2

ch1993128, 2020-01-16:
需要扶墙。。。。以前大雕的git action编译挺好用的，可以看下，make menuconfig在本地做，看定制出的文件改到线上编译。。

whlsxl, 2020-01-16:
我是编译`kmod-sched-cake`的时候Error了，找到原因了。
之前一直用`docker container` 编译，一是不用配置环境，二是可以直接扔到服务器上。但是最近这个项目更新了依赖，没注意，所以导致有些库编译缺少工具。[最新依赖](https://github.com/coolsnowwolf/lede/blob/49e1a7c9b2a086ec4b6426e1d218364502f3d7ff/README#L15)
也跟大家分享下我的`Dockerfile`
```
FROM ubuntu:18.04
MAINTAINER Hailong

ENV DEBIAN_FRONTEND=noninteractive
RUN \
  touch /etc/apt/sources.list && \
  sed -i 's/archive.ubuntu.com/mirrors.aliyun.com/g' /etc/apt/sources.list && \
  apt-get update && \
  apt-get -y upgrade && \
  apt-get install -y build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev && \
  apt-get install -y lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp && \
  apt-get install -y libssl-dev texinfo libglib2.0-dev && \
  apt-get install -y xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler && \
  apt-get install -y wget curl time  nano && \
  apt-get install -y tzdata && \
  touch /root/.bashrc && \
  ln -fs /usr/share/zoneinfo/Asia/Shanghai /etc/localtime && \
  dpkg-reconfigure --frontend noninteractive tzdata && \
  echo "alias time=/usr/bin/time" > /root/.bashrc && \
  rm -rf /var/lib/apt/lists/* && \
  git clone https://github.com/coolsnowwolf/lede /tmp/lede/ && \
  ./scripts/feeds update -a && \
  ./scripts/feeds install -a
# ENV LANG en_US.utf8
WORKDIR /tmp/lede/
```
用的时候先build image `docker build -t lede:1804 -f [Dockerfile Path] .`
再`docker run -it  --name lede1804 -v [lede存储位置]:/tmp/lede lede:1804 /bin/bash`
[clone 之后](https://github.com/coolsnowwolf/lede/blob/49e1a7c9b2a086ec4b6426e1d218364502f3d7ff/README#L17)，就可以愉快的编译啦。

whlsxl, 2020-01-16:
> 需要扶墙。。。。以前大雕的git action编译挺好用的，可以看下，make menuconfig在本地做，看定制出的文件改到线上编译。。

是完全在没有上墙的情况下编译的，不过各地的情况可能不一样。

ch1993128, 2020-01-16:
.........那百分百失败，不要想了，要么你以前成功过是路由扶过的

closed 2020-02-19 by coolsnowwolf