coolsnowwolf/lede #2788: 关于Softether VPN DDNS 默认配置的问题

Tiexcn, 2020-01-16:
这个软件默认自带的的DDNS，稳定性极差，还会导致莫名其妙的掉HUB，也不知是给人黑掉了还是什么问题，一般1，2天HUB就消失了，VPN连接不上，在禁用了DDNS服务后，一切正常。。。

如果说无法修改默认配置，也建议在插件开启页面上做一个提示，以提醒使用的朋友。

keivenliao, 2020-01-16:
还有这个问题？不过我从来没有用过这个ddns

Tiexcn, 2020-01-16:
网桥是开启了的，  SecureNAT也是关闭着的， 因为OpenVPN默认配置文件使用的是这个软件自带的DDNS，我也是没有修改，先是连接不上，后来发现解析的地址，不是我的外网地址，在丢了2-3次HUB后，我关掉了自带的DDNS，故障便解除了，丢完HUB后，我的管理员密码无法正常连接到Softeher VPN，所以我才怀疑是不是被黑了。

Pantheon-Paradoxes, 2020-01-17:
个人用了两年从4到5，没有发生过掉hub的问题，你这个自查openwrt内是不是与接口相关的应用冲突，mwan,firewall等

coolsnowwolf, 2020-01-21:
已测试，不能重现

bzzjh, 2020-01-22:
没有遇到这个情况，挺好用的。

closed 2020-02-19 by coolsnowwolf

Jefffish09, 2020-05-04:
> 还有这个问题？不过我从来没有用过这个ddns
> 刚才看了一下，默认是开启的， 也就是说我一直开着这个DDNS，也没有发现有不稳定的情况
> 我个人觉得关键有以下两点
> 1、配置本地网桥
> [image]
> 
> 2、关闭该软件的dncp
> [image]

@keivenliao 

我看一些网上教程在网桥设置那里
[image]
选择的是第二项（tap那个选项），不知道这两个选项有啥不同？