coolsnowwolf/lede #2792: SSR PLUS 是否可以添加用v2ray来进行Socks5代理

awter, 2020-01-16:
是否可以添加用v2ray来进行Socks5代理，或者单独弄个全局Socks5的代理功能？
类似问题：
#2476

coolsnowwolf, 2020-01-18:
有独立的 v2ray server 的 ipk

bdhjiajia, 2020-01-21:
独立那个可以加密码。

awter, 2020-01-24:
那个 我试了一下，不可以为空用户、空密码，对我来说意义不大，比如谷歌浏览器就不支持socks5 用户名密码。只需要简单的socks5代理就行了。

hyird, 2020-01-26:
> 那个 我试了一下，不可以为空用户、空密码，对我来说意义不大，比如谷歌浏览器就不支持socks5 用户名密码。只需要简单的socks5代理就行了。

你的需求 passwall 应该可以

trackycn, 2020-02-05:
> 那个 我试了一下，不可以为空用户、空密码，对我来说意义不大，比如谷歌浏览器就不支持socks5 用户名密码。只需要简单的socks5代理就行了。

这个可以自己修改一下的, 两种修改方法:
1. 直接修改源码, 重新编译这个插件, 然后路由器里安装
2. 在路由里修改, 修改完成后, 在luci界面去除用户名和密码保存下就可以生效

修改两个文件
1. config.lua
源码路径: /package/lean/luci-app-v2ray-server/luasrc/model/cbi/v2ray_server/
路由路径: /usr/lib/lua/luci/model/cbi/v2ray_server

2. genv2rayconfig.lua
源码路径: /package/lean/luci-app-v2ray-server/luasrc/model/cbi/v2ray_server/api
路由路径: /usr/lib/lua/luci/model/cbi/v2ray_server/api

修改后的文件放百度云了
`链接: https://pan.baidu.com/s/1DXFlZaBUQ6MNRacagQAK9w 提取码: brv2`

awter, 2020-02-06:
> 这个可以自己修改一下的, 两种修改方法:
>
> 1. 直接修改源码, 重新编译这个插件, 然后路由器里安装
> 2. 在路由里修改, 修改完成后, 在luci界面去除用户名和密码保存下就可以生效
>
> 修改两个文件
>
> 1. config.lua
>    源码路径: /package/lean/luci-app-v2ray-server/luasrc/model/cbi/v2ray_server/ [...]

牜牜牜 试了一下 可以了 谢谢大佬

closed 2020-02-06 by awter