coolsnowwolf/lede #2793: 最新的SSR-plus不兼容19.07正式版的luci

xun188, 2020-01-16:
最新的SSR-plus不兼容19.07正式版，麻烦更新兼容性

hahaher123, 2020-01-16:
把luci-lib-ipkg装上就行

tjuyy, 2020-01-16:
19.07 选择采用了在客户端的浏览器用Javascript渲染界面，而非在服务端用Lua，所有需要选上**luci-compat**

参考 [官方release说明: OpenWrt 19.07.0 first stable release](https://forum.openwrt.org/t/openwrt-19-07-0-first-stable-release/52186)

> 
> **Client-side rendering of the LuCI web interface**
> 
> The new version of LuCI, the integrated web interface for OpenWrt,
> implements client-side rendering of views. This improves performance by
> offloading some work that was done on the device (Lua code) to the
> client browser (Javascript code)
> 
> The LuCI ecosystem is large, and not all LuCI apps have been adapted to
> this change, which may result in crashes involving cbi.lua. In that
> case, install the _luci-compat_ package.
> 
> With this step, Lua usage in LuCI is reduced and LuCI effectively comes
> closer to the goals of the experimental LuCI2 without having to rewrite
> everything from scratch.

sqliuchang, 2020-01-18:
> 19.07 选择采用了在客户端的浏览器用Javascript渲染界面，而非在服务端用Lua，所有需要选上**luci-compat**

原来luci是服务端渲染的啊，现在改用客户端js渲染是好事

closed 2020-01-19 by coolsnowwolf