coolsnowwolf/lede #2801: 大量的Cron.info,不知是什么原因。下载大大最新的译固件，仅安装smartDNS.

ly19811105, 2020-01-17:
ri Jan 17 14:09:27 2020 cron.info crond[4200]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:09:37 2020 cron.info crond[4965]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:09:48 2020 cron.info crond[5779]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:09:59 2020 cron.info crond[6533]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:10:09 2020 cron.info crond[7242]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:10:20 2020 cron.info crond[7916]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:10:30 2020 cron.info crond[9358]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:10:41 2020 cron.info crond[10005]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:14:03 2020 cron.info crond[10968]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:14:14 2020 cron.info crond[11609]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:14:24 2020 cron.info crond[12464]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:14:35 2020 cron.info crond[13755]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:14:45 2020 cron.info crond[14467]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:14:56 2020 cron.info crond[15186]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:15:06 2020 cron.info crond[15905]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:15:17 2020 cron.info crond[16620]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:15:27 2020 cron.info crond[17339]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:15:38 2020 cron.info crond[18129]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:15:48 2020 cron.info crond[18816]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:15:59 2020 cron.info crond[19509]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:16:09 2020 cron.info crond[20202]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:16:20 2020 cron.info crond[20895]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:16:30 2020 cron.info crond[22330]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:16:41 2020 cron.info crond[23023]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:16:52 2020 cron.info crond[23716]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:17:02 2020 cron.info crond[24403]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:17:13 2020 cron.info crond[25096]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:17:23 2020 cron.info crond[25789]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:17:34 2020 cron.info crond[26476]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:17:44 2020 cron.info crond[27169]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:17:55 2020 cron.info crond[27862]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:18:05 2020 cron.info crond[28578]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:18:16 2020 cron.info crond[29266]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:18:27 2020 cron.info crond[29959]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:18:37 2020 cron.info crond[30652]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:18:48 2020 cron.info crond[31339]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:18:58 2020 cron.info crond[32032]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:19:09 2020 cron.info crond[32725]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:19:19 2020 cron.info crond[965]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:19:30 2020 cron.info crond[1693]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:19:40 2020 cron.info crond[2393]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:19:51 2020 cron.info crond[3111]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:20:01 2020 cron.info crond[3801]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:20:12 2020 cron.info crond[4448]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:20:23 2020 cron.info crond[5101]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:20:33 2020 cron.info crond[5742]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:20:44 2020 cron.info crond[6389]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:20:54 2020 cron.info crond[7036]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:21:05 2020 cron.info crond[7677]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:21:15 2020 cron.info crond[8324]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:21:26 2020 cron.info crond[8971]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:21:36 2020 cron.info crond[9612]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:21:47 2020 cron.info crond[10259]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:21:57 2020 cron.info crond[11186]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:22:08 2020 cron.info crond[11873]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:22:18 2020 cron.info crond[12566]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:22:29 2020 cron.info crond[13259]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:22:39 2020 cron.info crond[13946]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:22:50 2020 cron.info crond[14639]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:23:01 2020 cron.info crond[15332]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:23:11 2020 cron.info crond[16019]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:23:22 2020 cron.info crond[16713]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:23:32 2020 cron.info crond[17406]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:23:43 2020 cron.info crond[18093]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:23:53 2020 cron.info crond[18809]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:24:04 2020 cron.info crond[19502]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:24:14 2020 cron.info crond[20189]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:24:25 2020 cron.info crond[20886]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:24:35 2020 cron.info crond[21578]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:24:46 2020 cron.info crond[22266]: crond (busybox 1.30.0) started, log level 5
[... 5 lines cut ...]
Fri Jan 17 14:25:49 2020 cron.info crond[26435]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:26:00 2020 cron.info crond[27128]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:26:11 2020 cron.info crond[27821]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:26:21 2020 cron.info crond[28509]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:26:32 2020 cron.info crond[29202]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:26:42 2020 cron.info crond[29895]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:26:53 2020 cron.info crond[30588]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:27:03 2020 cron.info crond[31298]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:27:14 2020 cron.info crond[31991]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:27:24 2020 cron.info crond[32684]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:27:35 2020 cron.info crond[916]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:27:46 2020 cron.info crond[1628]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:27:56 2020 cron.info crond[2329]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:28:07 2020 cron.info crond[3020]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:28:17 2020 cron.info crond[3717]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:28:28 2020 cron.info crond[4412]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:28:38 2020 cron.info crond[5122]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:28:49 2020 cron.info crond[5815]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:28:59 2020 cron.info crond[6508]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:29:10 2020 cron.info crond[7943]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:29:20 2020 cron.info crond[8636]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:29:31 2020 cron.info crond[9329]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:29:42 2020 cron.info crond[10016]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:29:52 2020 cron.info crond[10709]: crond (busybox 1.30.0) started, log level 5
Fri Jan 17 14:30:03 2020 cron.info crond[11356]: crond (busybox 1.30.0) started, log level 5

CYaNu, 2020-01-17:
计划脚本里面有日志输出 不想看去掉就行了

MxCen, 2020-01-17:
smartDNS默认安装后对SSR有反作用。建议DNS不是太垃圾了，不要安装。

Her0R0cket, 2020-01-17:
你这是计划任务的守护进程不停重启导致的,你去系统的计划任务里面看看有什么计划任务,全部清掉看看正常不

sqliuchang, 2020-01-18:
你的计划任务有问题，停止工作后又进程守护重启了，你检查检查自己的计划任务格式对不对。或者是被什么程序杀死了？

outemancut, 2020-01-19:
你把网易云音乐关了应该就没有了

coolsnowwolf, 2020-01-19:
需要卸载 smartdns，否则 smartdns 重定向了客户端的 dns 但没有包括路由器“自身”的 DNS而导致路由器上面的 ssr plus 不能正确连接到 Google 的正确 IP 而导致进程一再重启

ly19811105, 2020-01-20:
谢谢各位大神的帮助，弱弱的问一句，这个除了卸载Smartdns，有其它方法解决吗？或都有其它dns本地服务器软件推荐吗？多谢！

coolsnowwolf, 2020-01-21:
已内置，不需要再做什么了

closed 2020-01-24 by coolsnowwolf