coolsnowwolf/lede #2803: minidlna请求增加DSD文件支持，谢谢。

coolsuse, 2020-01-17:
因想利用minidlna功能提供dsd音源给播放器，方便更换歌曲，发觉现有的不支持dsd格式。搜索到以下GitHub已增加dsd的支持，直接替换您的package文件后编译出错，因本人能力不足以修正，故请求minidlna增加DSD文件支持，辛苦了，谢谢！
附上支持DSD格式GitHub源： https://github.com/harmy/openwrt-minidlna

coolsnowwolf, 2020-01-18:
已支持，测试汇报

coolsuse, 2020-01-18:
编译出错了，全新编译后错误如下：

make[4]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12'
make  all-recursive
make[5]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12'
Making all in m4
make[6]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/m4'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/m4'
Making all in src
make[6]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/src'
make  all-am
make[7]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/src'
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I..  -DLOCALEDIR=\"/usr/share/locale\"  -I/home/freewind/lede/staging_dir/target-x86_64_musl/usr/include -I/home/freewind/lede/staging_dir/target-x86_64_musl/include -I/home/freewind/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/freewind/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/freewind/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12:libgpg-error-1.12 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic  -MT gpg_error-strerror-sym.o -MD -MP -MF .deps/gpg_error-strerror-sym.Tpo -c -o gpg_error-strerror-sym.o `test -f 'strerror-sym.c' || echo './'`strerror-sym.c
strerror-sym.c: In function 'gpg_strerror_sym':
strerror-sym.c:45:13: warning: implicit declaration of function 'errnos_msgidxof'; did you mean 'msgidxof'? [-Wimplicit-function-declaration]
       idx = errnos_msgidxof (code);
             ^~~~~~~~~~~~~~~
             msgidxof
strerror-sym.c:47:9: error: 'errnos_msgstr' undeclared (first use in this function)
  return errnos_msgstr + errnos_msgidx[idx];
         ^~~~~~~~~~~~~
strerror-sym.c:47:9: note: each undeclared identifier is reported only once for each function it appears in
strerror-sym.c:47:25: error: 'errnos_msgidx' undeclared (first use in this function); did you mean 'errnos_msgstr'?
  return errnos_msgstr + errnos_msgidx[idx];
                         ^~~~~~~~~~~~~
                         errnos_msgstr
make[7]: *** [Makefile:645: gpg_error-strerror-sym.o] Error 1
make[7]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/src'
make[6]: *** [Makefile:386: all] Error 2
make[6]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/src'
Making all in tests
make[6]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/tests'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/tests'
Making all in po
make[6]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/po'
make[6]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/po'
Making all in lang
make[6]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/lang'
Making all in cl
make[7]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/lang/cl'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/lang/cl'
make[7]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/lang'
make[7]: Nothing to be done for 'all-am'.
make[7]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/lang'
make[6]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/lang'
make[6]: Entering directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12'
make[6]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12'
make[5]: *** [Makefile:386: all-recursive] Error 1
make[5]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12'
make[4]: *** [Makefile:317: all] Error 2
make[4]: Leaving directory '/home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12'
make[3]: *** [Makefile:84: /home/freewind/lede/build_dir/target-x86_64_musl/libgpg-error-1.12/.built] Error 2
make[3]: Leaving directory '/home/freewind/lede/feeds/packages/libs/libgpg-error'
time: package/feeds/packages/libgpg-error/compile#0.09#0.02#0.13
make[2]: *** [package/Makefile:109: package/feeds/packages/libgpg-error/compile] Error 2
make[2]: Leaving directory '/home/freewind/lede'
make[1]: *** [package/Makefile:103: /home/freewind/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/freewind/lede'
make: *** [/home/freewind/lede/include/toplevel.mk:218: world] Error 2

AmadeusGhost, 2020-01-19:
试下这个 https://github.com/coolsnowwolf/packages/pull/20 ，ipq40xx编译通过

coolsuse, 2020-01-19:
> 已支持，测试汇报

感谢出手，马上就有！
已确认支持，谢谢！

coolsnowwolf, 2020-01-21:
同时也支持了 APE

coolsuse, 2020-01-21:
> 同时也支持了 APE

多谢大佬

closed 2020-01-24 by coolsnowwolf