coolsnowwolf/lede #2805: 执行 opkg upgrade luci 页面出现异常

CuminLo, 2020-01-18:
刚编译的固件配置上网之后，发现有包可以更新。

于是我执行了 `opkg upgrade luci`，更新过程正常，但是当访问  `192.168.1.1` 之后，出现以下错误：

```
/usr/lib/lua/luci/dispatcher.lua:437: Invalid controller file found
The file '/usr/lib/lua/luci/controller/opkg.lua' contains no index() function.
Please make sure that the controller contains a valid index function and verify the spelling!
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:437: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:452: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:188: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
```

只是界面不能访问，上网和命令行都是正常的。

搜索了一圈，貌似出现这个问题是可能 L大对 `luci` 有修改与原来 `openwrt luci`  不兼容导致的么？因为我发现我更新其他的包几乎都不能成功，类似以下这样的错误：

```
opkg upgrade aria2
...
Collected errors:
 * check_data_file_clashes: Package libstdcpp6 wants to install file /usr/lib/libstdc++.so.6
        But that file is already provided by package  * libstdcpp
```

1. 除了恢复出厂，我现在的情况该如何恢复 `luci`?
2. 如何执行  `opkg upgrade ...`  而不会出现兼容问题？

AmadeusGhost, 2020-01-18:
1. 恢复出厂设置，或者升级固件的时候不保留配置
2. 底层包不要动，类似kmod、aria2这些编译的时候选进去，别upgrade了

CuminLo, 2020-01-18:
@AmadeusGhost 谢谢。有 `upgrade` 强迫症...

closed 2020-01-18 by CuminLo

AmadeusGhost, 2020-01-18:
> @AmadeusGhost 谢谢。有 `upgrade` 强迫症...

主要是官方的会冲突