coolsnowwolf/lede #2813: 大神们能否指导一下？

norikohxb, 2020-01-19:
https://totoro.site/index.php/archives/70/
我按照了这篇文章把lean大的程序复制到friendlywrt对应的目录里面。然后进行编译。
  cd friendlywrt-h5
          git clone https://github.com/coolsnowwolf/lede
          cd friendlywrt
          cp -r ../lede/package/lean package/
          sed -i 's/https:\/\/git.openwrt.org\/project\/luci.git^039ef1f4deba725d3591b159bbc9569885d68131/https:\/\/github.com\/coolsnowwolf\/luci/' feeds.conf.default
          sed -i 's/https:\/\/git.openwrt.org\/feed\/packages.git^00803ffc91e80b16e9c1603ff32106d42e255923/https:\/\/github.com\/coolsnowwolf\/packages/' feeds.conf.default       

我用这个方式编译好了friendlywrt+lean的package（勾选了adbyby plus）并且编译成功了。
其他功能都正常，就是adbyby plus无法更新host lists 貌似adbyby也无法正常启动。
有没有大神可以帮我看一下错误日志帮我分析一下是哪里出问题了？
以下是错误日志

Sun Jan 19 22:33:58 2020 daemon.crit dnsmasq[3622]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 22:33:58 2020 daemon.crit dnsmasq[3622]: FAILED to start up
Sun Jan 19 22:33:58 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 19 22:33:58 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 19 22:34:01 2020 daemon.crit dnsmasq[4195]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 22:34:01 2020 daemon.crit dnsmasq[4195]: FAILED to start up
Sun Jan 19 22:34:01 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 19 22:34:01 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 19 22:34:04 2020 daemon.crit dnsmasq[4277]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 22:34:04 2020 daemon.crit dnsmasq[4277]: FAILED to start up
Sun Jan 19 22:34:09 2020 cron.info crond[4324]: crond (busybox 1.30.1) started, log level 5
Sun Jan 19 22:34:09 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 19 22:34:09 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 19 22:34:13 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 19 22:34:13 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 19 22:34:16 2020 daemon.crit dnsmasq[4935]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 22:34:16 2020 daemon.crit dnsmasq[4935]: FAILED to start up
Sun Jan 19 22:34:21 2020 daemon.crit dnsmasq[4942]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 22:34:21 2020 daemon.crit dnsmasq[4942]: FAILED to start up
Sun Jan 19 22:34:26 2020 daemon.crit dnsmasq[4944]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 22:34:26 2020 daemon.crit dnsmasq[4944]: FAILED to start up
Sun Jan 19 22:34:31 2020 daemon.crit dnsmasq[4945]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 22:34:31 2020 daemon.crit dnsmasq[4945]: FAILED to start up
Sun Jan 19 22:34:36 2020 daemon.crit dnsmasq[4947]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 22:34:36 2020 daemon.crit dnsmasq[4947]: FAILED to start up
Sun Jan 19 22:34:41 2020 daemon.crit dnsmasq[4948]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /var/etc/dnsmasq-adbyby.d/06-dnsmasq.esc
Sun Jan 19 22:34:41 2020 daemon.crit dnsmasq[4948]: FAILED to start up

coolsnowwolf, 2020-01-19:
你的 dnsmasq 没有开启 ipset 支持

norikohxb, 2020-01-19:
好的，谢谢lean大 的解答，我马上测试

norikohxb, 2020-01-19:
lean大您好，我加上了所有的dnsmasq-full的子菜单，编译到friendlywrt里面也编译成功了，也有adbyby的菜单，但是adbyby菜单开启和关闭在log里面都无法找到，虽然菜单显示启动了，但是实际上还是没有启动，体现在 adblock host list无法更新。能否在百忙之中抽空指导一下吗？

norikohxb, 2020-01-19:
每天凌晨6点更新规则并重启 Adblock Plus Host List： 940 Lines
手动强制更新Adblock Plus Host List （无论按多少次）
adblock plus host list的数量还是940
adbyby在friendlywrt最新版本里面是完全没有运行

norikohxb, 2020-01-19:
SSR plus也都正常，其他都正常，Nano Pi R1S H5的adbyby就是不能用，不起作用。

hahaher123, 2020-01-20:
安装wget了么，更新规则需要用到

norikohxb, 2020-01-20:
[https://www.right.com.cn/forum/data/attachment/forum/202001/20/150829im395md2hiw369b2.jpg](url)

norikohxb, 2020-01-20:
是在 make menuconfig 里面的 netowrk-》file transfer 里面的 最后2个wget 吗？我补充勾选了并且编译了。然后用最新编译的固件再刷了一次r1s，还是无法更新 adblocks host  list
在纪录里面没有看到有更新纪录
而且还提示adblock没有启动，我编译并没有选择adblock

Mon Jan 20 15:29:16 2020 user.notice root: adbyby mem mode rules updated!
Mon Jan 20 15:29:16 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 20 15:29:16 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 20 15:29:26 2020 cron.info crond[2755]: crond (busybox 1.30.1) started, log level 5
Mon Jan 20 15:29:27 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 20 15:29:27 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 20 15:29:31 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 20 15:29:31 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 20 15:30:34 2020 cron.info crond[3586]: crond (busybox 1.30.1) started, log level 5
Mon Jan 20 15:30:34 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 20 15:30:34 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 20 15:30:39 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 20 15:30:39 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

norikohxb, 2020-01-20:
更新好了，太感谢你们了，非常感谢你，已经解决了，要勾选wget，否则不会更新，非常非常感谢你们。

norikohxb, 2020-01-20:
感谢热心的lean大大的指导，还有感谢热心的hahaher123大大的提醒。圆满完成自己的编译。谢谢你们了

closed 2020-01-20 by norikohxb