coolsnowwolf/lede #2821: 纯粹作为v2ray+adbyby plus去广告的旁路由有没有必要开启luci-app-flowoffload(turbo acc加速)和luci-app-upnp

norikohxb, 2020-01-20:
lean在make menuconfig的标配都是有
luci-app-upnp 和 luci-app-flowoffload 
第一个是upnp作为主路由是可以让迅雷bt下载更快对吧。那么作为和ros主路由同在一个虚拟机共用一个网卡接口的旁路由来说，这个开不开upnp有没有区别？

还有 luci-app-flowoffload 对应的是turbo acc的选项，在旁路由下开启turbo acc是否是多此一举的做法？

请问有没有人可以解答一下？

9l, 2020-01-21:
没有必要吧

pV-nRT, 2020-01-22:
旁路网关下，upnp会失效，需要自己在主路由和旁路网关上面设置两次端口映射。flowoffload在固件的描述是“开启 Flow Offloading 转发加速. 降低CPU占用 / 增强路由转发性能”，根据描述，转发应该是基于二层，所以旁路网关参与转发应该是有加速的。

norikohxb, 2020-01-22:
我是否可以总结一下你的意思， UPNP对于 路由无效，仅对NAT有效，旁路由的网卡网关直接是同一个局域网的另外一个ip，所以其实都是直接连接的，这个时候UPNP对于这种情况的网络来说其实是多此一举的，但是flow offloading  有转发加速的功能，对于旁路由这种拐一次路由是有用的。可以增加转发性能。我这样理解对吗？

pV-nRT, 2020-01-23:
> 我是否可以总结一下你的意思， UPNP对于 路由无效，仅对NAT有效，旁路由的网卡网关直接是同一个局域网的另外一个ip，所以其实都是直接连接的，这个时候UPNP对于这种情况的网络来说其实是多此一举的，但是flow offloading 有转发加速的功能，对于旁路由这种拐一次路由是有用的。可以增加转发性能。我这样理解对吗？

是的，在旁路网关这种模式下，upnp这种自动端口映射假设是映射到旁路网关，则旁路网关到主路网关之间的映射没有自动打开，假设是映射到主路网关，则设备的所有传出流量根本没到达主路网关，而是去往旁路网关，根据思路，一种简单的解决方法是：主路由关闭upnp，设置dmz指向旁路网关，然后旁路网关开启upnp

pV-nRT, 2020-01-23:
> 我是否可以总结一下你的意思， UPNP对于 路由无效，仅对NAT有效，旁路由的网卡网关直接是同一个局域网的另外一个ip，所以其实都是直接连接的，这个时候UPNP对于这种情况的网络来说其实是多此一举的，但是flow offloading 有转发加速的功能，对于旁路由这种拐一次路由是有用的。可以增加转发性能。我这样理解对吗？

是的，我是这样理解的

pV-nRT, 2020-01-23:
> 我是否可以总结一下你的意思， UPNP对于 路由无效，仅对NAT有效，旁路由的网卡网关直接是同一个局域网的另外一个ip，所以其实都是直接连接的，这个时候UPNP对于这种情况的网络来说其实是多此一举的，但是flow offloading 有转发加速的功能，对于旁路由这种拐一次路由是有用的。可以增加转发性能。我这样理解对吗？

对，但是设备没有直接连接外网，upnp不是多此一举，而是upnp这种自动nat穿透失效，需要自己另外配置。这是旁路网关的一个副作用

norikohxb, 2020-01-23:
这个是我旁路由的快照，我这次编译加入了luci里面的upnp
这个是旁路由的openwrt的服务菜单里面的upnp的菜单。还是会有比特彗星的upnp通过旁路由。
旁路由的LAN的网关立马指向ROS（主路由了）
局域网的电脑网关是旁路由。
按照你的说法，这个UPNP里面应该看不到任何连接。
因为上网的电脑其实每一个的ip和 ROS都是在同一个网段。在255.255.255.0的子网掩码下，其实是可以自由的和ROS这个主路由直接通讯的。

协议 | 外部端口 | 客户端地址 | 客户端端口 | 描述 |  
-- | -- | -- | -- | -- | --
TCP | 34001 | 192.168.16.144 | 49681 | DIRECT_REVERSE_SERVER |  
TCP | 34002 | 192.168.16.142 | 34491 | DIRECT_REVERSE_SERVER |  
UDP | 35555 | 192.168.16.142 | 35555 | wechat voip |  
TCP | 39005 | 192.168.16.144 | 39005 | BJSDK at 192.168.16.144:39005 |  
UDP | 39005 | 192.168.16.144 | 39005 | BJSDK at 192.168.16.144:39005 |  
UDP | 6949 | 192.168.16.144 | 6949 | BJSDK at 192.168.16.144:6949 |  
TCP | 6949 | 192.168.16.144 | 6949 | BJSDK at 192.168.16.144:6949 |  
TCP | 8488 | 192.168.16.99 | 8488 | BitComet TCP |  
UDP | 8488 | 192.168.16.99 | 8488 | BitComet UDP |  
UDP | 51893 | 192.168.16.99 | 51893 | Teredo 192.168.16.99:51893->51893 UDP |

norikohxb, 2020-01-23:
协议 | 外部端口 | 客户端地址 | 客户端端口 | 描述 |  
-- | -- | -- | -- | -- | --
TCP | 34001 | 192.168.16.144 | 49681 | DIRECT_REVERSE_SERVER |  
TCP | 34002 | 192.168.16.142 | 34491 | DIRECT_REVERSE_SERVER |  
UDP | 35555 | 192.168.16.142 | 35555 | wechat voip |  
TCP | 39005 | 192.168.16.144 | 39005 | BJSDK at 192.168.16.144:39005 |  
UDP | 39005 | 192.168.16.144 | 39005 | BJSDK at 192.168.16.144:39005 |  
UDP | 6949 | 192.168.16.144 | 6949 | BJSDK at 192.168.16.144:6949 |  
TCP | 6949 | 192.168.16.144 | 6949 | BJSDK at 192.168.16.144:6949 |  
TCP | 8488 | 192.168.16.99 | 8488 | BitComet TCP |  
UDP | 8488 | 192.168.16.99 | 8488 | BitComet UDP |  
UDP | 51893 | 192.168.16.99 | 51893 | Teredo 192.168.16.99:51893->51893 UDP |

pV-nRT, 2020-01-23:
> 这个是我旁路由的快照，我这次编译加入了luci的里面的UPnP的
> 这个是旁路由的的OpenWrt的服务菜单里面的的UPnP的菜单。还是会有比特彗星的UPnP的通过旁路由。
> 旁路由的LAN的网关立马指向ROS（主路由了）互连的计算机
> 网关是插入由。
> 按照你的说法，这个UPNP里面应该看不到任何连接。
> 因为上网的电脑其实每一个的ip和ROS都是在同一个网段。在255.255.255.0的同轴电缆下，其实是可以自由的和ROS这个主路由直接通讯的。
> 
> 协议	外部端口	客户端地址	客户端端口	描述	 
> TCP协议	34001	192.168.16.144	49681	DIRECT_REVERSE_SERVER	 
> TCP协议	34002	192.168.16.142	34491	DIRECT_REVERSE_SERVER	 
> UDP协议	35555	192.168.16.142	35555	微信voip	 
> TCP协议	39005	192.168.16.144	39005	BJSDK位于192.168.16.144:39005	 
> UDP协议	39005	192.168.16.144	39005	BJSDK位于192.168.16.144:39005	 
> UDP协议	6949	192.168.16.144	6949	BJSDK位于192.168.16.144:6949	 
> TCP协议	6949	192.168.16.144	6949	BJSDK位于192.168.16.144:6949	 
> TCP协议	8488	192.168.16.99	8488	比特彗星TCP	 
> UDP协议	8488	192.168.16.99	8488	比特彗星UDP	 
> UDP协议	51893	192.168.16.99	51893	Teredo 192.168.16.99:51893->51893 UDP

那它就是有用的，开着吧

pV-nRT, 2020-01-23:
> 这个是我旁路由的快照，我这次编译加入了luci里面的upnp
> 这个是旁路由的openwrt的服务菜单里面的upnp的菜单。还是会有比特彗星的upnp通过旁路由。
> 旁路由的LAN的网关立马指向ROS（主路由了）
> 局域网的电脑网关是旁路由。
> 按照你的说法，这个UPNP里面应该看不到任何连接。
> 因为上网的电脑其实每一个的ip和 [...]

有建立规则不一定代表能用。

端口映射

端口映射就是将外网主机的IP地址的一个端口映射到内网中一台机器，提供相应的服务。当用户访问该IP的这个端口时，服务器自动将请求映射到对应局域网内部的机器上。端口映射有动态和静态之分。

动态端口映射：内网中的一台电脑要访问网站，会向NAT网关发送数据包，包头中包括对方网站IP、端口和本机IP、端口，NAT网关会把本机IP、端口替换成自己的公网IP、一个未使用的端口，并且会记下这个映射关系，为以后转发数据包使用。

然后再把数据发给网站，网站收到数据后做出反应，发送数据到NAT网关的那个未使用的端口，然后NAT网关将数据转发给内网中的那台电脑，实现内网和公网的通讯.当连接关闭时，NAT网关会释放分配给这条连接的端口，以便以后的连接可以继续使用。

动态端口映射其实也就是NAT网关的工作方式。

静态端口映射：就是在NAT网关上开放一个固定的端口，然后设定此端口收到的数据要转发给内网哪个IP和端口，不管有没有连接，这个映射关系都会一直存在。就可以让公网主动访问内网的一台电脑。

作为旁路网关时，之所以你每一次访问网站都能打开页面，是因为打开网站这个动作是由你发起的，你发出数据包之后会产生一个映射关系用来接收网站发回来的数据包。
但是你使用p2p时，需要的是别人主动去连接你，别人从tracker服务器获取你映射出的端口数来连接你的电脑，你不是主动发起连接的那个人，别人发来的数据包首先到达你的主路网关，但是你的主路网关没有建立映射关系，它不知道这个数据包发给谁，它会自动丢包，如果你建立了端口映射，告诉路由器在某个端口下，只要接收到数据包，那么就发送到子网中的某一台电脑，这样别人才可以和你建立连接。
回到问题当中来，你的旁路网关在upnp下建立了各种映射规则，这个规则指定了旁路网关接收到相应端口的数据包，就要发送到特定设备，问题是，外网的人根本就访问不到你的旁路网关，所以旁路网关即使建立各种映射规则，也接收不到任何外网机器相应端口的数据包，所以规则失效，但并不影响其映射规则的建立。因为upnp建立映射规则并不判断规则是否可以。

当然，有一个简单的解决方法就是，在你的主路网关设置旁路网关的ip为dmz。

参考资料
http://www.wayos.com:8081/support/FAQ/jiaochengzhidao_baojiaohui/20171205/1955.html

norikohxb, 2020-02-01:
谢谢你

closed 2020-02-01 by norikohxb

kabuda2077, 2021-07-23:
> > 这个是我旁路由的快照，我这次编译加入了luci里面的upnp
> > 这个是旁路由的openwrt的服务菜单里面的upnp的菜单。还是会有比特彗星的upnp通过旁路由。
> > 旁路由的LAN的网关立马指向ROS（主路由了）
> > 局域网的电脑网关是旁路由。
> > 按照你的说法，这个UPNP里面应该看不到任何连接。
> > 因为上网的电脑其实每一个的ip和 ROS都是在同一个网段。在255.255.255.0的子网掩码下，其实是可以自由的和ROS这个主路由直接通讯的。
> > 协议	外部端口	客户端地址	客户端端口	描述	 
> > TCP	34001	192.168.16.144	49681	DIRECT_REVERSE_SERVER	 
> > TCP	34002	192.168.16.142	34491	DIRECT_REVERSE_SERVER	 
> > UDP	35555	192.168.16.142	35555	wechat voip	 
> > TCP	39005	192.168.16.144	39005	BJSDK at 192.168.16.144:39005	 
> > UDP	39005	192.168.16.144	39005	BJSDK at 192.168.16.144:39005	 
> > UDP	6949	192.168.16.144	6949	BJSDK at 192.168.16.144:6949	 
> > TCP	6949	192.168.16.144	6949	BJSDK at 192.168.16.144:6949	 
> > TCP	8488	192.168.16.99	8488	BitComet TCP	 
> > UDP	8488	192.168.16.99	8488	BitComet UDP	 
> > UDP	51893	192.168.16.99	51893	Teredo 192.168.16.99:51893->51893 UDP
> 
> 有建立规则不一定代表能用。
> 
> 端口映射
> 
> 端口映射就是将外网主机的IP地址的一个端口映射到内网中一台机器，提供相应的服务。当用户访问该IP的这个端口时，服务器自动将请求映射到对应局域网内部的机器上。端口映射有动态和静态之分。
> 
> 动态端口映射：内网中的一台电脑要访问网站，会向NAT网关发送数据包，包头中包括对方网站IP、端口和本机IP、端口，NAT网关会把本机IP、端口替换成自己的公网IP、一个未使用的端口，并且会记下这个映射关系，为以后转发数据包使用。
> 
> 然后再把数据发给网站，网站收到数据后做出反应，发送数据到NAT网关的那个未使用的端口，然后NAT网关将数据转发给内网中的那台电脑，实现内网和公网的通讯.当连接关闭时，NAT网关会释放分配给这条连接的端口，以便以后的连接可以继续使用。
> 
> 动态端口映射其实也就是NAT网关的工作方式。
> 
> 静态端口映射：就是在NAT网关上开放一个固定的端口，然后设定此端口收到的数据要转发给内网哪个IP和端口，不管有没有连接，这个映射关系都会一直存在。就可以让公网主动访问内网的一台电脑。
> 
> 作为旁路网关时，之所以你每一次访问网站都能打开页面，是因为打开网站这个动作是由你发起的，你发出数据包之后会产生一个映射关系用来接收网站发回来的数据包。
> 但是你使用p2p时，需要的是别人主动去连接你，别人通过扫描你的端口或者是从tracker服务器获取你映射出的端口数来连接你的电脑，你不是主动发起连接的那个人，别人发来的数据包首先到达你的主路网关，但是你的主路网关没有建立映射关系，它不知道这个数据包发给谁，它会自动丢包，如果你建立了端口映射，告诉路由器在某个端口下，只要接收到数据包，那么就发送到子网中的某一台电脑，这样别人才可以和你建立连接。
> 回到问题当中来，你的旁路网关在upnp下建立了各种映射规则，这个规则指定了旁路网关接收到相应端口的数据包，就要发送到特定设备，问题是，外网的人根本就访问不到你的旁路网关，所以旁路网关即使建立各种映射规则，也接收不到任何外网机器相应端口的数据包，所以规则失效，但并不影响其映射规则的建立。因为upnp建立映射规则并不判断规则是否可用。
> 
> 当然，有一个简单的解决方法就是，在你的主路网关设置旁路网关的ip为dmz。
> 
> 参考资料
> http://www.wayos.com:8081/support/FAQ/jiaochengzhidao_baojiaohui/20171205/1955.html

请问设置dmz后，应该开启主路由的upnp还是旁路由的？