coolsnowwolf/lede #2822: 有的ss订阅地址不提供#标记的alias描述，导致订阅直接异常退出

jesset, 2020-01-20:
有的ss订阅地址不提供#标记的alias描述，content:find("#") 返回 nil，导致订阅直接异常退出

```
2020-01-20 17:01:39 /usr/share/shadowsocksr/subscribe.lua:164: attempt to perform arithmetic on a nil value
2020-01-20 17:01:39 stack traceback:
	/usr/share/shadowsocksr/subscribe.lua:335: in function </usr/share/shadowsocksr/subscribe.lua:333>
	/usr/share/shadowsocksr/subscribe.lua:164: in function 'processData'
	/usr/share/shadowsocksr/subscribe.lua:252: in function </usr/share/shadowsocksr/subscribe.lua:204>
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:333: in main chunk
	[C]: ?
2020-01-20 17:01:39 发生错误, 正在恢复服务
```

luckyyyyy, 2020-01-20:
哈哈 这个只能靠你们了 很多格式太奇怪了

closed 2020-01-21 by coolsnowwolf

RealKiro, 2020-01-22:
[image]

还是报错

jesset, 2020-01-22:
你要更新代码重新编译吧，或者直接替换 subscribe.lua 这个脚本

`wget -O /usr/share/shadowsocksr/subscribe.lua https://raw.githubusercontent.com/coolsnowwolf/lede/46fc685ec42e327ffe8b4afd166321eacfd4290a/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua`

RealKiro, 2020-01-22:
> 你要更新代码重新编译吧，或者直接替换 subscribe.lua 这个脚本
> 
> `wget -O /usr/share/shadowsocksr/subscribe.lua https://raw.githubusercontent.com/coolsnowwolf/lede/46fc685ec42e327ffe8b4afd166321eacfd4290a/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua`

试过了，还是一样

jesset, 2020-01-22:
看了下，你这个格式又是一种不兼容的格式了

subscribe.lua 对 vmess 协议是假设后面是一段 json，但你这个的格式是接近于ss协议，类似于

> vmess://chacha20-poly1305:<密码>@<IP>:<端口>?<其他参数对>

```
141     elseif szType == 'vmess' then
142         local info = luci.jsonc.parse(content)
143         result.type = 'v2ray'
144         result.server = info.add
145         result.server_port = info.port

```

luckyyyyy, 2020-01-22:
这些各位自己改一下即可，我最近没有太多时间ss有两种，默认只兼容一种。

发自我的iPhone

 
  
vmess://chacha20-poly1305:<密码&gt;@:<端口&gt;?<其他参数对&gt;

jesset, 2020-01-22:
> > 你要更新代码重新编译吧，或者直接替换 subscribe.lua 这个脚本
> > `wget -O /usr/share/shadowsocksr/subscribe.lua https://raw.githubusercontent.com/coolsnowwolf/lede/46fc685ec42e327ffe8b4afd166321eacfd4290a/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua`
> 
> 试过了，还是一样 [URL地址测试](https://raw.githubusercontent.com/eycorsican/rule-sets/master/kitsunebi_sub)

我更新了脚本，你试试：

```
wget -O /usr/share/shadowsocksr/subscribe.lua https://raw.githubusercontent.com/jesset/lede-1/d207eefaf1abb416b1ee01d242ecabbd5cf3f671/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua
```

RealKiro, 2020-01-22:
> ```
> wget -O /usr/share/shadowsocksr/subscribe.lua https://raw.githubusercontent.com/jesset/lede-1/d207eefaf1abb416b1ee01d242ecabbd5cf3f671/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua
> ```

正常了，感谢🙏

Mackerly, 2020-01-30:
感谢这个脚本，比bash script快了不是一点点。说一个小问题，密码带了@字符的会出错的。比如下面这个

> c3M6Ly9ZV1Z6TFRJMU5pMWpabUk2VFY1VlVtWnljVTFJWTBkaVIxVXlRRzgxUmxCUE9VSkVORU5RU0ZWaloyVkFORFV1TVRRd0xqRTNNQzR4TlRRNk1qQXdOamc9I1JV

luckyyyyy, 2020-01-30:
看看有没有什么办法改一下 应该简单的 

发自我的iPhone