coolsnowwolf/lede #2828: > 我貌似发现了更深源的原因。如果是通过MIUI自己安装的框架，内置的Google服务配置文件，使用的是services.googleapis.cn。而通过apk mirror之类安装的框架，则是用的services.googleapis.com。这样导致了路由的GFW列表识别不出而不走梯子。我试了一下在SSP P+的强制走代理加入services.googleapis.cn，同时把加州google的216.58.197.195加入强制走代理WAN IP（216.58.197.0/24），手机上的google play、相册、油土鳖什么的app，都能顺利通过登录和连接了。

angelmengg, 2020-01-20:
> 我貌似发现了更深源的原因。如果是通过MIUI自己安装的框架，内置的Google服务配置文件，使用的是services.googleapis.cn。而通过apk mirror之类安装的框架，则是用的services.googleapis.com。这样导致了路由的GFW列表识别不出而不走梯子。我试了一下在SSP P+的强制走代理加入services.googleapis.cn，同时把加州google的216.58.197.195加入强制走代理WAN IP（216.58.197.0/24），手机上的google play、相册、油土鳖什么的app，都能顺利通过登录和连接了。
然而我也是小米手机，从miui9到11，从安卓6到10，从来没发现过这个问题，只是有些机场可以出国但是依然无法谷歌罢了

试过这样加入强制代理依然不行，安卓9类原生系统

AmadeusGhost, 2020-01-20:
有毒啊，哪有这么长的题目，描述应该放底下这块区域而不是题目上，手机打开issues一看半个页面都是你的issue...

rampageX, 2020-01-20:
菊花等手机必须把 services.googleapis.cn 拿出来单独处理，这个 .cn 后缀很害人。

angelmengg, 2020-01-21:
> 有毒啊，哪有这么长的题目，描述应该放底下这块区域而不是题目上，手机打开issues一看半个页面都是你的issue...

不好意思，已经改了

coolsnowwolf, 2020-01-21:
cn 是个挺特殊的问题

badcrazy, 2020-01-21:
> 试过这样加入强制代理依然不行，安卓9类原生系统

> 试过这样加入强制代理依然不行，安卓9类原生系统
然而我用miui一致正常用play啊，我用v2ray，不管手机上用v2ary还是路由器ssrp用v2ray，截图
https://s2.ax1x.com/2020/01/21/1kliYF.jpg

ly19811105, 2020-01-22:
我的米11也是不能正常使用油土鳖等App，能用chorme 上谷歌看油土鳖。很奇怪。

angelmengg, 2020-01-22:
> 我的米11也是不能正常使用油土鳖等App，能用chorme 上谷歌看油土鳖。很奇怪。

搞不懂，实在不行就不折腾了，直接用手机客户端

ly19811105, 2020-01-22:
> > 我的米11也是不能正常使用油土鳖等App，能用chorme 上谷歌看油土鳖。很奇怪。
> 
> 搞不懂，实在不行就不折腾了，直接用手机客户端

我的很奇怪，可以用手机VPN先连接一下，关闭VPN然后就能通过路由的SSR P+上了，但只能上一会，感觉VPN同步了一个DNS给手端，手机就会用这个DNS一段时间，时间到了就又用回手机的DNS，路由的DNS手机好像不理会一样的。还是手机DNS问题，

angelmengg, 2020-01-22:
> 的

我的手机这样操作连半分钟就翻不了，而且这个手机的DNS好像不能修改吧

closed 2020-01-23 by coolsnowwolf

qingxiwa, 2020-02-28:
> 试过这样加入强制代理依然不行，安卓9类原生系统

我发现了一个比较好的解决方法，miui下适用，把play市场，play服务，Google框架能卸载的都卸了，然后安装外区的谷歌框架，可以用go安装器装（可以在酷安下到最新的）不要装miui应用市场自动安装的那个国行框架，之后就没有Googleapis.cn这个问题