coolsnowwolf/lede #2829: cifsd安装后，Win10的系统无法共享指定的目录。

jy3344511, 2020-01-20:
试了下，安装samba4的话，win10可以直接共享指定的目录，但是用luci-app-cifsd的话，无法共享。

samba4可以正常使用，但是在播放视频的时候，鼠标拖动进度条播放时，会有卡顿，想试下cifsd的效果，但无法使用，怎么解决呢？

拖动进度条播放时卡顿，不是配置、网线问题。

keivenliao, 2020-01-21:
是USB？还是软路由？如果是软路由，那么黑群晖不香吗？

jy3344511, 2020-01-21:
> 是USB？还是软路由？如果是软路由，那么黑群晖不香吗？

USB接口，接的3.5 机械硬盘。

coolsnowwolf, 2020-01-21:
CIFSD 没有包含 NetBIOS 服务，有些版本的 Windows 10 确实依赖局域网里面的 Master 服务宣告

jy3344511, 2020-01-21:
> CIFSD 没有包含 NetBIOS 服务，有些版本的 Windows 10 确实依赖局域网里面的 Master 服务宣告

又试了下ftp的方法，用PotPlayer播放的话，会自动快速加载进度条，就像看网页视频缓冲的那种，会直接缓冲把进度条加满，然后就可以随意的拖动进度条播放，一点都不卡，但ftp的话，用不习惯，还是没有那种直接挂载硬盘，在硬盘里操作来的方便。

9l, 2020-01-21:
视频你需要流媒体服务器，直接挂载网络硬盘看，如果视频文件本身的码率高，受限于硬盘速度网络速度，播放时快进会卡也是正常现象

jy3344511, 2020-01-21:
> 流媒体服务器

哪个插件是流媒体服务器？

9l, 2020-01-22:
在你放视频的电脑上，装个 plex server

Aes64X, 2020-01-22:
nfs不香吗。。。。。改个文件而已

jy3344511, 2020-01-22:
> nfs

这个怎么弄，有详细教程吗？

closed 2020-02-19 by coolsnowwolf