coolsnowwolf/lede #2835: 网件 R7800 最新 4.14.162 4.19.93内核 160MHZ 不能使用

lqzhj, 2020-01-22:
网件 R7800 最新 4.14.162 4.19.93内核 160MHZ 不能使用 ， 强行选择160MHZ会导致5G WIFI信号丢失

AmadeusGhost, 2020-01-22:
最起码给个日志啊

lqzhj, 2020-01-23:
4.14.162内核日志

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.162 (lqzhj@lqzhj) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r1837-64fb649b)) #0 SMP Tue Jan 21 10:28:55 2020
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Netgear Nighthawk X4S R7800
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 122880
[    0.000000] free_area_init_node: node 0, pgdat c0b299c0, node_mem_map dd9c1000
[    0.000000]   Normal zone: 1080 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x42c with crng_init=0
[    0.000000] percpu: Embedded 15 pages/cpu s29452 r8192 d23796 u61440
[    0.000000] pcpu-alloc: s29452 r8192 d23796 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121800
[    0.000000] Kernel command line: 
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 475864K/491520K available (4986K kernel code, 168K rwdata, 1296K rodata, 1024K init, 234K bss, 15656K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xde800000 - 0xff800000   ( 528 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xde000000   ( 480 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0208000 - 0xc07dea88   (5979 kB)
[    0.000000]       .init : 0xc0a00000 - 0xc0b00000   (1024 kB)
[    0.000000]       .data : 0xc0b00000 - 0xc0b2a2c0   ( 169 kB)
[    0.000000]        .bss : 0xc0b2c000 - 0xc0b66918   ( 235 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000007] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000020] Switching to timer-based delay loop, resolution 160ns
[    0.000219] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000243] pid_max: default: 32768 minimum: 301
[    0.000401] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000420] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001249] CPU: Testing write buffer coherency: ok
[    0.002103] Setting up static identity map for 0x42300000 - 0x42300060
[    0.002273] Hierarchical SRCU implementation.
[    0.003065] smp: Bringing up secondary CPUs ...
[    0.004979] smp: Brought up 1 node, 2 CPUs
[    0.004996] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005008] CPU: All CPU(s) started in SVC mode.
[    0.016801] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.016973] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016998] futex hash table entries: 512 (order: 3, 32768 bytes)
[    0.017151] pinctrl core: initialized pinctrl subsystem
[    0.018187] NET: Registered protocol family 16
[    0.018446] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.019859] cpuidle: using governor ladder
[    0.019926] cpuidle: using governor menu
[    0.043059] msm_bus_fabric_init_driver
[    0.044497] usbcore: registered new interface driver usbfs
[    0.044581] usbcore: registered new interface driver hub
[    0.044664] usbcore: registered new device driver usb
[    0.044724] pps_core: LinuxPPS API ver. 1 registered
[    0.044737] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.044770] PTP clock support registered
[    0.046514] clocksource: Switched to clocksource dg_timer
[    0.049924] NET: Registered protocol family 2
[... 452 lines cut ...]
[ 1898.640198] br-lan: port 2(wlan0) entered disabled state
[ 1904.966794] ath10k_pci 0000:01:00.0: Unknown eventid: 36933
[ 1904.970626] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1959.645560] ath10k_pci 0000:01:00.0: Unknown eventid: 36933
[ 1959.647847] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1959.653566] br-lan: port 2(wlan0) entered blocking state
[ 1959.657040] br-lan: port 2(wlan0) entered disabled state
[ 1959.661661] device wlan0 entered promiscuous mode
[ 1959.666766] br-lan: port 2(wlan0) entered blocking state
[ 1959.671367] br-lan: port 2(wlan0) entered forwarding state
[ 1959.690340] br-lan: port 2(wlan0) entered disabled state
[ 1960.384025] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1960.384163] br-lan: port 2(wlan0) entered blocking state
[... cut ...]
[106216.790291] ath10k_pci 0000:01:00.0: Unknown eventid: 36933
[106216.794373] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[106216.808649] br-lan: port 2(wlan0) entered blocking state
[106216.808711] br-lan: port 2(wlan0) entered disabled state
[106216.813996] device wlan0 entered promiscuous mode
[106217.164412] device wlan0 left promiscuous mode
[106217.164526] br-lan: port 2(wlan0) entered disabled state
[106223.481256] ath10k_pci 0000:01:00.0: Unknown eventid: 36933
[106223.485011] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

AmadeusGhost, 2020-01-24:
这样吧，```make menuconfig```改用ath10k-firmware-qca9984-ct和kmod-ath10k-ct
```print_req_error: I/O error```这个把文件系统改成f2fs: mkfs.f2fs /dev/xxx

lqzhj, 2020-01-25:
5g wifi 160MHZ 掉信号可以用上面方法解决，谢谢。
print_req_error 不敢试😂😂

pk49800, 2020-01-26:
> 这样吧，`make menuconfig`改用ath10k-firmware-qca9984-ct和kmod-ath10k-ct
> `print_req_error: I/O error`这个把文件系统改成f2fs: `mkfs.f2fs /dev/xxx`

请教：云编译固件如何设置这几个编译参数，同样是R7800，没有编译机可用，一直用云编译

lqzhj, 2020-01-28:
CONFIG_PACKAGE_ath10k-firmware-qca9984-ct=y
CONFIG_PACKAGE_kmod-ath10k-ct=y

lqzhj, 2020-01-30:
ipq806x(R7800 4.19.98内核) ath10k-ct（160MHZ）下不兼容intel 9260，能连接WIFI但无网络

其他高通却手机连的好好的

pk49800, 2020-01-30:
160MHZ intel 的ax200也不行，连接无网络，只能用89mhz

dtgit006, 2020-02-01:
https://github.com/greearb/ath10k-ct

这个驱动不知道能不能修复160MHZ的问题

closed 2020-02-19 by coolsnowwolf

LeeMenHin, 2020-02-28:
> https://github.com/greearb/ath10k-ct
> 
> 这个驱动不知道能不能修复160MHZ的问题
建議屌大也看看

poweryb9802, 2020-03-27:
希望可以更新ATH10K-CT的驱动