coolsnowwolf/lede #2836: adbyby出现问题

lkde558907, 2020-01-22:
adbyby有问题了哦，开启之后不能更新GFW列表，某些域名也打不开，比如renyim.cm，开启过滤就会打不开，还有网易对战平台打开也很慢，对战平台下载，要等很久才会下载

sage-E3845, 2020-01-22:
卧槽，开错帐号了，上面这个是我发的，有人有这个问题吗，我新编译的固件

sage-E3845, 2020-01-22:
已找到解决方案，果然还是要花时间想办法，方案是删除adbyby的IP黑名单里119.188.13.0/24
和119.4.249.166/32，然后你就会发现神奇般的好了

iyuangang, 2020-01-22:
这次adbyby更新感觉被人夹带私货。小米很多正常服务都被屏蔽。无法更新系统，无法更新主题，小米视频被完全屏蔽... 对一个公司有意见，可以自己设置屏蔽。不应该这样操作。个人意见，不一定对。

hxlls, 2020-01-22:
实际测试中，确实不少的小米的服务被误杀，可能是去广告规则太激进了。

sage-E3845, 2020-01-23:
> 这次adbyby更新感觉被人夹带私货。小米很多正常服务都被屏蔽。无法更新系统，无法更新主题，小米视频被完全屏蔽... 对一个公司有意见，可以自己设置屏蔽。不应该这样操作。个人意见，不一定对。

你用的应该是全局过滤，换成列表过滤或许会正常了

iyuangang, 2020-01-23:
> > 这次adbyby更新感觉被人夹带私货。小米很多正常服务都被屏蔽。无法更新系统，无法更新主题，小米视频被完全屏蔽... 对一个公司有意见，可以自己设置屏蔽。不应该这样操作。个人意见，不一定对。
> 
> 你用的应该是全局过滤，换成列表过滤或许会正常了

是plus+ 模式，只能手动编辑 package/lean/luci-app-adbyby-plus/root/usr/share/adbyby 下的配置重新编译。

KFERMercer, 2020-01-23:
↑ 提交回滚了

closed 2020-01-23 by coolsnowwolf