coolsnowwolf/lede #2841: igmpproxy无法转发igmp流量

ostrichii, 2020-01-23:
用大雕的源码编译出来igmppoxy运行正常，防火墙策略也没问题，但是就是无法把IGMP流量成功转发。尝试了koolshare和openwrt官方固件，均无问题。主要是做IPTV和宽带融合，UDPXY倒是可以正常工作。

coolsnowwolf, 2020-01-23:
确实没做这个选项，因为没有测试环境

ostrichii, 2020-01-25:
> 确实没做这个选项，因为没有测试环境

可以给大雕提供远程环境，不过可能要等到年后开工返程之后（不知道还能不能开工，国内疫情严重）

flexyangjie, 2020-01-29:
有可能 是IGMP版本的问题，我之前弄IPTV遇见过这个问题。
下面的文字请参考一下，我是参考这个成功解决问题的。
【有些组播是IGMPv2的则需要改变一下版本
net.ipv4.conf.all.force_igmp_version=2
添加到 /etc/sysctl.conf里面就可以了】

ostrichii, 2020-01-30:
> 有可能 是IGMP版本的问题，我之前弄IPTV遇见过这个问题。
> 下面的文字请参考一下，我是参考这个成功解决问题的。
> 【有些组播是IGMPv2的则需要改变一下版本
> net.ipv4.conf.all.force_igmp_version=2
> 添加到 /etc/sysctl.conf里面就可以了】

非常感谢，我试试，能解决那就完美了。如果可以我这这个issue下反馈一下

luckyyyyy, 2020-01-31:
的确有问题  尝试官方的可以正常转发

[image]

luckyyyyy, 2020-02-01:
@ostrichii 有什么思路可以讨论下

ostrichii, 2020-02-10:
> 有可能 是IGMP版本的问题，我之前弄IPTV遇见过这个问题。
> 下面的文字请参考一下，我是参考这个成功解决问题的。
> 【有些组播是IGMPv2的则需要改变一下版本
> net.ipv4.conf.all.force_igmp_version=2
> 添加到 /etc/sysctl.conf里面就可以了】

尝试了不行，和这个没关系

ostrichii, 2020-02-10:
> @ostrichii 有什么思路可以讨论下

还没找到原因，这个得请教大雕这部分转发的功能在官方源码上做了些啥优化

luckyyyyy, 2020-02-10:
> > @ostrichii 有什么思路可以讨论下
> 
> 还没找到原因，这个得请教大雕这部分转发的功能在官方源码上做了些啥优化

我最近iptv没有在搞组播，所以暂时没用到了，因为我发现我家里盒子走的也是rtsp，所以我也用这个看了，目前来看挺稳的的。

closed 2020-02-19 by coolsnowwolf

kasen-zhang, 2023-10-16:
想知道如题这个问题目前解决了吗

kasen-zhang, 2023-10-16:
> > > @ostrichii 有什么思路可以讨论下
> > 
> > 
> > 还没找到原因，这个得请教大雕这部分转发的功能在官方源码上做了些啥优化
> 
> 我最近iptv没有在搞组播，所以暂时没用到了，因为我发现我家里盒子走的也是rtsp，所以我也用这个看了，目前来看挺稳的的。

大佬我的IPTV直播源地址也是rtsp地址类型的，目前我正在折腾IPTV,目前通过设置Vlan的方式在OP中能取得20.X.X.X的IPTV地址，但同时看同IPTV和上网业务就，只能上网，无法在笔记本上通过VLC播放M3U格式文件看IPTV，相知道这个是应为OP系统版本或内核的原因还是我的设置有问题